pyocd/pyOCD #1400: setuptools_scm ValueError: time data 'gpg: Signature made Sun May 29 19:42:43 2022 EES' does not match format '%Y-%m-%d'

eugene-bright, 2022-05-30:
```
$ pip3 install -U git+https://github.com/pyocd/pyOCD@develop
Defaulting to user installation because normal site-packages is not writeable
Collecting git+https://github.com/pyocd/pyOCD@develop
  Cloning https://github.com/pyocd/pyOCD (to revision develop) to /tmp/pip-req-build-kfacbd_7
  Running command git clone --filter=blob:none --quiet https://github.com/pyocd/pyOCD /tmp/pip-req-build-kfacbd_7
  Running command git checkout -b develop --track origin/develop
  Switched to a new branch 'develop'
  Branch 'develop' set up to track remote branch 'develop' from 'origin'.
  Resolved https://github.com/pyocd/pyOCD to commit bd445ca814df130810b2a18354bcd0c8132bfe33
  Installing build dependencies ... done
  Getting requirements to build wheel ... error
  error: subprocess-exited-with-error
  
  × Getting requirements to build wheel did not run successfully.
  │ exit code: 1
  ╰─> [48 lines of output]
      Traceback (most recent call last):
        File "/home/hexum/.local/lib/python3.8/site-packages/pip/_vendor/pep517/in_process/_in_process.py", line 363, in <module>
          main()
        File "/home/hexum/.local/lib/python3.8/site-packages/pip/_vendor/pep517/in_process/_in_process.py", line 345, in main
          json_out['return_val'] = hook(**hook_input['kwargs'])
        File "/home/hexum/.local/lib/python3.8/site-packages/pip/_vendor/pep517/in_process/_in_process.py", line 130, in get_requires_for_build_wheel
          return hook(config_settings)
        File "/tmp/pip-build-env-je0dgnae/overlay/lib/python3.8/site-packages/setuptools/build_meta.py", line 177, in get_requires_for_build_wheel
          return self._get_build_requires(
        File "/tmp/pip-build-env-je0dgnae/overlay/lib/python3.8/site-packages/setuptools/build_meta.py", line 159, in _get_build_requires
          self.run_setup()
        File "/tmp/pip-build-env-je0dgnae/overlay/lib/python3.8/site-packages/setuptools/build_meta.py", line 174, in run_setup
          exec(compile(code, __file__, 'exec'), locals())
        File "setup.py", line 43, in <module>
          setup()
        File "/tmp/pip-build-env-je0dgnae/overlay/lib/python3.8/site-packages/setuptools/__init__.py", line 87, in setup
          return distutils.core.setup(**attrs)
        File "/tmp/pip-build-env-je0dgnae/overlay/lib/python3.8/site-packages/setuptools/_distutils/core.py", line 109, in setup
          _setup_distribution = dist = klass(attrs)
        File "/tmp/pip-build-env-je0dgnae/overlay/lib/python3.8/site-packages/setuptools/dist.py", line 477, in __init__
          _Distribution.__init__(
        File "/tmp/pip-build-env-je0dgnae/overlay/lib/python3.8/site-packages/setuptools/_distutils/dist.py", line 293, in __init__
          self.finalize_options()
        File "/tmp/pip-build-env-je0dgnae/overlay/lib/python3.8/site-packages/setuptools/dist.py", line 901, in finalize_options
          ep(self)
        File "/tmp/pip-build-env-je0dgnae/overlay/lib/python3.8/site-packages/setuptools_scm/integration.py", line 104, in infer_version
          _assign_version(dist, config)
        File "/tmp/pip-build-env-je0dgnae/overlay/lib/python3.8/site-packages/setuptools_scm/integration.py", line 48, in _assign_version
          maybe_version = _get_version(config)
        File "/tmp/pip-build-env-je0dgnae/overlay/lib/python3.8/site-packages/setuptools_scm/__init__.py", line 151, in _get_version
          parsed_version = _do_parse(config)
        File "/tmp/pip-build-env-je0dgnae/overlay/lib/python3.8/site-packages/setuptools_scm/__init__.py", line 98, in _do_parse
          version = _version_from_entrypoints(config) or _version_from_entrypoints(
        File "/tmp/pip-build-env-je0dgnae/overlay/lib/python3.8/site-packages/setuptools_scm/_entrypoints.py", line 37, in _version_from_entrypoints
          version: Optional[ScmVersion] = _call_entrypoint_fn(root, config, ep.load())
        File "/tmp/pip-build-env-je0dgnae/overlay/lib/python3.8/site-packages/setuptools_scm/_entrypoints.py", line 13, in _call_entrypoint_fn
          return fn(root, config=config)
        File "/tmp/pip-build-env-je0dgnae/overlay/lib/python3.8/site-packages/setuptools_scm/git.py", line 146, in parse
          return _git_parse_inner(
        File "/tmp/pip-build-env-je0dgnae/overlay/lib/python3.8/site-packages/setuptools_scm/git.py", line 179, in _git_parse_inner
          node_date = wd.get_head_date() or date.today()
        File "/tmp/pip-build-env-je0dgnae/overlay/lib/python3.8/site-packages/setuptools_scm/git.py", line 81, in get_head_date
          return datetime.strptime(date_part, r"%Y-%m-%d").date()
        File "/usr/local/lib/python3.8/_strptime.py", line 568, in _strptime_datetime
          tt, fraction, gmtoff_fraction = _strptime(data_string, format)
        File "/usr/local/lib/python3.8/_strptime.py", line 349, in _strptime
          raise ValueError("time data %r does not match format %r" %
      ValueError: time data 'gpg: Signature made Sun May 29 19:42:43 2022 EES' does not match format '%Y-%m-%d'
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× Getting requirements to build wheel did not run successfully.
│ exit code: 1
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.

$ git version
git version 2.34.1

$ git config log.showSignature
true
```

Relates to: https://github.com/pypa/setuptools_scm/issues/548

eugene-bright, 2022-05-30:
Changing `log.showSignature` to `false` did not help

```
$ git config log.showSignature
false
```

flit, 2022-05-30:
What a bother. 😄 Thanks for reporting this!

flit, 2022-05-30:
Hrmm. For some reason I'm not able to reproduce it on macOS. I'll try Linux in a little while.

It seems pypa/setuptools_scm#548 has been fixed, but doesn't look like it's made it into a release yet. pypa/setuptools_scm@7950bdaa3e5 shows it will be in 6.4.3.

eugene-bright, 2022-05-30:
```
$ pip3 show setuptools_scm
Name: setuptools-scm
Version: 6.4.2
```

eugene-bright, 2022-05-30:
You can freek out how much I have to share 😉

eugene-bright, 2022-05-30:
For now I use this fd2994411061 as workaround

eugene-bright, 2022-06-13:
Showsignature should be disabled globally to work `setuptools_scm` bug around.

```bash
git config --global log.showSignature false
pip3.10 install git+ssh://git@github.com/pyocd/pyOCD.git@develop
```

PR #1431 (2022-07-06): packaging: setuptools_scm: fix GPG parse issue and .gitarchival template
This patch applies two fixes related to setuptools_scm, the package used to automatically set the pyocd package version at build or install time based on git tags.

1. Set the minimum setuptools_scm version  7.0 for building pyocd on Python 3.7 to fix a GPG output parsing issue. See pypa/setuptools_scm#548 for details. For Python 3.6, the minimum setuptools_scm version remains 6.0 since version doesn't support Python 3.6
2. Migrate the template in `.gitarchival.txt` to the one required by the built-in git archive support. [... cut]
commit dde4e8d2 (2022-07-06): packaging: setuptools_scm: fix GPG parse issue and .gitarchival template (#1431)

* pyproject.toml: bump required setuptools_scm version to >=7.0 to fix GPG parsing issue

Fixes #1400

See https://github.com/pypa/setuptools_scm/issues/548 for the
setuptools_scm bug.

The setuptools_scm version 7.0 that includes the fix only supports
Python 3.7 and above. Pyocd currently still supports Python 3.6,
making this fix unavailable for pyocd running on that version.

* .git_archival: migrate to builtin setuptools_scm git archive support
closed 2022-07-06 by flit