pyocd/pyOCD #1402: st-info --chipid and --descr like  functionality

eugene-bright, 2022-05-31:
`st-info` can rougtly identify the chip:

```
$ st-info --chipid 
0x0466

$ st-info --descr
G030/G031/G041
```

I wonder if such functionality can be added to the pyOCD.

eugene-bright, 2022-05-31:
It looks like the `chipid` is stored in magic register `0xE000ED00` in case of STM32 chips
https://github.com/stlink-org/stlink/blob/924e1ec93b9080593a10c02c25e483a0ceb60a04/src/common.c#L187
https://github.com/stlink-org/stlink/blob/cb551d50f879ec21e3d54c75041bffe506e830e5/doc/dev/developer.txt#L333

elfmimi, 2022-05-31:
pyOCD's '[User script](https://pyocd.io/docs/user_scripts.html)' feature is very handy to do things like this.

And also there is a valuable accumulation of knowledge in pystlink's repo nicely licensed under MIT lincese.
https://github.com/pavelrevak/pystlink/blob/master/lib/stm32devices.py