pyocd/pyOCD #1404: STM32H745ZIT6 "Error attempting to probe CoreSight component referenced by ROM table entry #5: Memory transfer fault (read)"

Nekspire, 2022-06-01:
Hello,
I've got an inssue with flashing STM32H745ZIT6. This happens only when i've got some code inside the flash memory. For example if the flash is erased with tool like STM32CubeProgrammer the problem does not exist and the flashing  process is done without any problem.

The pycod gdb output says that: 

```
[2022-06-01T16:44:48.859Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session connected. You can switch to "DEBUG CONSOLE" to see GDB interactions.
pyocd gdbserver --port 50000 --telnet-port 50002 --target stm32h745zitx
0001031 I Target type is stm32h745zitx [board]
0001220 I DP IDR = 0x6ba02477 (v2 rev6) [dap]
0001289 I AHB-AP#0 IDR = 0x84770001 (AHB-AP var0 rev8) [ap]
0001291 I AHB-AP#1 IDR = 0x84770001 (AHB-AP var0 rev8) [ap]
0001292 I APB-AP#2 IDR = 0x54770002 (APB-AP var0 rev5) [ap]
0001294 I AHB-AP#3 IDR = 0x24770011 (AHB-AP var1 rev2) [ap]
0001297 I AHB-AP#0 Class 0x1 ROM table #0 @ 0xe00fe000 (designer=020 part=450) [rom_table]
0001298 I [0]<e00ff000:ROM class=1 designer=43b part=4c7> [rom_table]
0001299 I   AHB-AP#0 Class 0x1 ROM table #1 @ 0xe00ff000 (designer=43b part=4c7) [rom_table]
0001300 I   [0]<e000e000:SCS v7-M class=14 designer=43b part=00c> [rom_table]
0001301 I   [1]<e0001000:DWT v7-M class=14 designer=43b part=002> [rom_table]
0001302 I   [2]<e0002000:FPB v7-M class=14 designer=43b part=00e> [rom_table]
0001302 I   [3]<e0000000:ITM v7-M class=14 designer=43b part=001> [rom_table]
0001304 I [1]<e0041000:ETM M7 class=9 designer=43b part=975 devtype=13 archid=4a13 devid=0:0:0> [rom_table]
0001305 I [2]<e0043000:CTI class=9 designer=43b part=906 devtype=14 archid=0000 devid=40800:0:0> [rom_table]
0001306 I APB-AP#2 Class 0x1 ROM table #0 @ 0xe00e0000 (designer=020 part=450) [rom_table]
0001308 W Invalid coresight component, cidr=0x0 [rom_table]
0001308 I [2]<e00e3000: cidr=0, pidr=0, component invalid> [rom_table]
0001308 W Invalid coresight component, cidr=0x0 [rom_table]
0001308 I [3]<e00e4000: cidr=0, pidr=0, component invalid> [rom_table]
0001309 I [4]<e00e5000:TSGEN class=15 designer=43b part=101> [rom_table]
0001310 E Error attempting to probe CoreSight component referenced by ROM table entry #5: Memory transfer fault (read) @ 0xe00f0fd0-0xe00f102f [rom_table]
Traceback (most recent call last):
  File "C:\Users\jwoli\AppData\Local\Programs\Python\Python310\lib\site-packages\pyocd\coresight\rom_table.py", line 372, in _read_table
    self._handle_table_entry(entry, entryNumber)
  File "C:\Users\jwoli\AppData\Local\Programs\Python\Python310\lib\site-packages\pyocd\coresight\rom_table.py", line 427, in _handle_table_entry
    cmpid.read_id_registers()
  File "C:\Users\jwoli\AppData\Local\Programs\Python\Python310\lib\site-packages\pyocd\coresight\rom_table.py", line 123, in read_id_registers
    regs = self.ap.read_memory_block32(self.top_address + self.IDR_READ_START, self.IDR_READ_COUNT)
  File "C:\Users\jwoli\AppData\Local\Programs\Python\Python310\lib\site-packages\pyocd\probe\stlink_probe.py", line 292, in read_memory_block32
    return conversion.byte_list_to_u32le_list(self._link.read_mem32(addr, size * 4, self._apsel))
  File "C:\Users\jwoli\AppData\Local\Programs\Python\Python310\lib\site-packages\pyocd\probe\stlink\stlink.py", line 440, in read_mem32
    return self._read_mem(addr, size, Commands.JTAG_READMEM_32BIT, self.MAXIMUM_TRANSFER_SIZE, apsel)
  File "C:\Users\jwoli\AppData\Local\Programs\Python\Python310\lib\site-packages\pyocd\probe\stlink\stlink.py", line 398, in _read_mem
    raise exc
pyocd.core.exceptions.TransferFaultError: Memory transfer fault (read) @ 0xe00f0fd0-0xe00f102f
0001313 I AHB-AP#3 Class 0x1 ROM table #0 @ 0xe00ff000 (designer=020 part=450) [rom_table]
0001314 I [0]<e000e000:SCS v7-M class=14 designer=43b part=00c> [rom_table]
0001315 I [1]<e0001000:DWT v7-M class=14 designer=43b part=002> [rom_table]
0001316 I [2]<e0002000:FPB v7-M class=14 designer=43b part=003> [rom_table]
0001316 I [3]<e0000000:ITM v7-M class=14 designer=43b part=001> [rom_table]
0001318 I [5]<e0041000:ETM M4 class=9 designer=43b part=925 devtype=13 archid=0000 devid=0:0:0> [rom_table]
0001319 I [6]<e0043000:CTI class=9 designer=43b part=906 devtype=14 archid=0000 devid=40800:0:0> [rom_table]
0001320 I CPU core #0 is Cortex-M7 r1p1 [cortex_m]
0001322 I FPU present: FPv5-D16-M [cortex_m]
0001323 I CPU core #1 is Cortex-M4 r0p1 [cortex_m]
0001326 I FPU present: FPv4-SP-D16-M [cortex_m]
0001327 I 4 hardware watchpoints [dwt]
0001329 I 8 hardware breakpoints, 1 literal comparators [fpb]
0001335 I 4 hardware watchpoints [dwt]
0001337 I 6 hardware breakpoints, 4 literal comparators [fpb]
0001350 I Semihost server started on port 50002 (core 0) [server]
0001388 I GDB server started on port 50000 (core 0) [gdbserver]
0001391 I Semihost server started on port 50003 (core 1) [server]
0001391 I GDB server started on port 50001 (core 1) [gdbserver]
0003461 I Client connected to port 50000! [gdbserver]
0003481 I Attempting to load RTOS plugins [gdbserver]
0008049 E Unhandled exception in handle_message (b'v'): cannot read register r0 because core #0 is not halted [gdbserver]
Traceback (most recent call last):
  File "C:\Users\jwoli\AppData\Local\Programs\Python\Python310\lib\site-packages\pyocd\flash\builder.py", line 259, in _enable_read_access
    self.flash.init(self.flash.Operation.VERIFY)
  File "C:\Users\jwoli\AppData\Local\Programs\Python\Python310\lib\site-packages\pyocd\flash\flash.py", line 252, in init
[... 26 lines cut ...]
  File "C:\Users\jwoli\AppData\Local\Programs\Python\Python310\lib\site-packages\pyocd\flash\flash.py", line 616, in wait_for_completion
    return self.target.read_core_register('r0')
  File "C:\Users\jwoli\AppData\Local\Programs\Python\Python310\lib\site-packages\pyocd\core\soc_target.py", line 218, in read_core_register
    return self.selected_core_or_raise.read_core_register(id)
  File "C:\Users\jwoli\AppData\Local\Programs\Python\Python310\lib\site-packages\pyocd\coresight\cortex_m.py", line 971, in read_core_register
    regValue = self.read_core_register_raw(reg_info.index)
  File "C:\Users\jwoli\AppData\Local\Programs\Python\Python310\lib\site-packages\pyocd\coresight\cortex_m.py", line 988, in read_core_register_raw
    vals = self.read_core_registers_raw([reg])
  File "C:\Users\jwoli\AppData\Local\Programs\Python\Python310\lib\site-packages\pyocd\coresight\cortex_m.py", line 1009, in read_core_registers_raw
    return self._base_read_core_registers_raw(reg_list)
  File "C:\Users\jwoli\AppData\Local\Programs\Python\Python310\lib\site-packages\pyocd\coresight\cortex_m.py", line 1023, in _base_read_core_registers_raw
    raise exceptions.CoreRegisterAccessError(
pyocd.core.exceptions.CoreRegisterAccessError: cannot read register r0 because core #0 is not halted
0008051 E Unexpected exception: a bytes-like object is required, not 'tuple' [gdbserver]
Traceback (most recent call last):
  File "C:\Users\jwoli\AppData\Local\Programs\Python\Python310\lib\site-packages\pyocd\gdbserver\gdbserver.py", line 390, in _run_connection
    self.packet_io.send(resp)
  File "C:\Users\jwoli\AppData\Local\Programs\Python\Python310\lib\site-packages\pyocd\gdbserver\packet_io.py", line 83, in send
    self._write_packet(packet)
  File "C:\Users\jwoli\AppData\Local\Programs\Python\Python310\lib\site-packages\pyocd\gdbserver\packet_io.py", line 146, in _write_packet
    written = self._abstract_socket.write(packet)
  File "C:\Users\jwoli\AppData\Local\Programs\Python\Python310\lib\site-packages\pyocd\utility\sockets.py", line 58, in write
    return self.conn.send(data)
TypeError: a bytes-like object is required, not 'tuple'
```

elfmimi, 2022-06-02:
I guess this issue shares common root cause with this one https://github.com/pyocd/pyOCD/issues/1368 .

Nekspire, 2022-06-02:
@elfmimi In my case the solution presented in #1368 does not work.

Here is the output:

```
[2022-06-02T17:12:17.428Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session connected. You can switch to "DEBUG CONSOLE" to see GDB interactions.
pyocd gdbserver --port 50000 --telnet-port 50002 --target stm32h745zitx
0001133 I Target type is stm32h745zitx [board]
0001362 I DP IDR = 0x6ba02477 (v2 rev6) [dap]
0001438 I AHB-AP#0 IDR = 0x84770001 (AHB-AP var0 rev8) [discovery]
0001440 I AHB-AP#1 IDR = 0x84770001 (AHB-AP var0 rev8) [discovery]
0001442 I APB-AP#2 IDR = 0x54770002 (APB-AP var0 rev5) [discovery]
0001443 I AHB-AP#3 IDR = 0x24770011 (AHB-AP var1 rev2) [discovery]
0001445 I AHB-AP#0 Class 0x1 ROM table #0 @ 0xe00fe000 (designer=020:ST part=450) [rom_table]
0001447 I [0]<e00ff000:ROM class=1 designer=43b:Arm part=4c7> [rom_table]
0001447 I   AHB-AP#0 Class 0x1 ROM table #1 @ 0xe00ff000 (designer=43b:Arm part=4c7) [rom_table]
0001448 I   [0]<e000e000:SCS v7-M class=14 designer=43b:Arm part=00c> [rom_table]
0001449 I   [1]<e0001000:DWT v7-M class=14 designer=43b:Arm part=002> [rom_table]
0001450 I   [2]<e0002000:FPB v7-M class=14 designer=43b:Arm part=00e> [rom_table]
0001451 I   [3]<e0000000:ITM v7-M class=14 designer=43b:Arm part=001> [rom_table]
0001452 I [1]<e0041000:ETM M7 class=9 designer=43b:Arm part=975 devtype=13 archid=4a13 devid=0:0:0> [rom_table]
0001453 I [2]<e0043000:CTI class=9 designer=43b:Arm part=906 devtype=14 archid=0000 devid=40800:0:0> [rom_table]
0001454 I APB-AP#2 Class 0x1 ROM table #0 @ 0xe00e0000 (designer=020:ST part=450) [rom_table]
0001456 I [2]<e00e3000:SWO CS-400 class=9 designer=43b:Arm part=914 devtype=11 archid=0000 devid=ea0:0:0> [rom_table]
0001457 I [3]<e00e4000:CSTF class=9 designer=43b:Arm part=908 devtype=12 archid=0000 devid=32:0:0> [rom_table]
0001458 I [4]<e00e5000:TSGEN class=15 designer=43b:Arm part=101> [rom_table]
0001459 I [5]<e00f0000:ROM class=1 designer=020:ST part=001> [rom_table]
0001459 I   APB-AP#2 Class 0x1 ROM table #1 @ 0xe00f0000 (designer=020:ST part=001) [rom_table]
0001461 I   [0]<e00f1000:CTI class=9 designer=43b:Arm part=906 devtype=14 archid=0000 devid=40800:0:0> [rom_table]
0001462 I   [2]<e00f3000:CSTF class=9 designer=43b:Arm part=908 devtype=12 archid=0000 devid=34:0:0> [rom_table]
0001463 I   [3]<e00f4000:ETF class=9 designer=43b:Arm part=961 devtype=32 archid=0000 devid=380:0:0> [rom_table]
0001465 I   [4]<e00f5000:TPIU class=9 designer=43b:Arm part=912 devtype=11 archid=0000 devid=a0:0:0> [rom_table]
0001467 I AHB-AP#3 Class 0x1 ROM table #0 @ 0xe00ff000 (designer=020:ST part=450) [rom_table]
0001468 I [0]<e000e000:SCS v7-M class=14 designer=43b:Arm part=00c> [rom_table]
0001469 I [1]<e0001000:DWT v7-M class=14 designer=43b:Arm part=002> [rom_table]
0001470 I [2]<e0002000:FPB v7-M class=14 designer=43b:Arm part=003> [rom_table]
0001470 I [3]<e0000000:ITM v7-M class=14 designer=43b:Arm part=001> [rom_table]
0001472 I [5]<e0041000:ETM M4 class=9 designer=43b:Arm part=925 devtype=13 archid=0000 devid=0:0:0> [rom_table]
0001473 I [6]<e0043000:CTI class=9 designer=43b:Arm part=906 devtype=14 archid=0000 devid=40800:0:0> [rom_table]
0001475 I CPU core #0 is Cortex-M7 r1p1 [cortex_m]
0001475 I FPU present: FPv5-D16-M [cortex_m]
0001477 I CPU core #1 is Cortex-M4 r0p1 [cortex_m]
0001478 I FPU present: FPv4-SP-D16-M [cortex_m]
0001480 I 4 hardware watchpoints [dwt]
0001482 I 8 hardware breakpoints, 1 literal comparators [fpb]
0001488 I 4 hardware watchpoints [dwt]
0001491 I 6 hardware breakpoints, 4 literal comparators [fpb]
0001500 I Semihost server started on port 50002 (core 0) [server]
0001536 I GDB server started on port 50000 (core 0) [gdbserver]
0001539 I Semihost server started on port 50003 (core 1) [server]
0001539 I GDB server started on port 50001 (core 1) [gdbserver]
0003602 I Client connected to port 50000! [gdbserver]
0003624 I Attempting to load RTOS plugins [gdbserver]
0008204 E Unhandled exception in handle_message (b'v'): target was not halted as expected after calling flash algorithm routine (IPSR=3) [gdbserver]
Traceback (most recent call last):
  File "C:\Users\jwoli\AppData\Local\Programs\Python\Python310\lib\site-packages\pyocd\flash\builder.py", line 259, in _enable_read_access
    self.flash.init(self.flash.Operation.VERIFY)
  File "C:\Users\jwoli\AppData\Local\Programs\Python\Python310\lib\site-packages\pyocd\flash\flash.py", line 252, in init
    result = self._call_function_and_wait(self.flash_algo['pc_init'],
  File "C:\Users\jwoli\AppData\Local\Programs\Python\Python310\lib\site-packages\pyocd\flash\flash.py", line 654, in _call_function_and_wait
    return self.wait_for_completion(timeout=timeout)
  File "C:\Users\jwoli\AppData\Local\Programs\Python\Python310\lib\site-packages\pyocd\flash\flash.py", line 648, in wait_for_completion
    raise exceptions.FlashFailure(f"flash algorithm overflowed stack ({self.begin_stack - self.end_stack} bytes)")
pyocd.core.exceptions.FlashFailure: flash algorithm overflowed stack (8192 bytes)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\jwoli\AppData\Local\Programs\Python\Python310\lib\site-packages\pyocd\gdbserver\gdbserver.py", line 423, in handle_message
    reply = handler(msg[msgStart:])
[... 8 lines cut ...]
  File "C:\Users\jwoli\AppData\Local\Programs\Python\Python310\lib\site-packages\pyocd\flash\builder.py", line 669, in _compute_sector_erase_pages_and_weight
[... cut ...]
  File "C:\Users\jwoli\AppData\Local\Programs\Python\Python310\lib\site-packages\pyocd\gdbserver\packet_io.py", line 83, in send
    self._write_packet(packet)
  File "C:\Users\jwoli\AppData\Local\Programs\Python\Python310\lib\site-packages\pyocd\gdbserver\packet_io.py", line 146, in _write_packet
    written = self._abstract_socket.write(packet)
  File "C:\Users\jwoli\AppData\Local\Programs\Python\Python310\lib\site-packages\pyocd\utility\sockets.py", line 58, in write
    return self.conn.send(data)
TypeError: a bytes-like object is required, not 'tuple'
```