pyocd/pyOCD #1420: CMSIS-DAP has typo in product string

wolfmanjm, 2022-06-15:
I have a popular DAP link (purchased in the UK) that has a typo in the product string which makes it unrecognized by default (even with the current develop branch).

By adding `DAP_LINK` to the KNOWN_DEVICE_STRINGS (the typo is _ instead of -), it is recognized.

I attach the results of `lsusb -v -d c251:f001` to show the issue.

[lsusb.txt](https://github.com/pyocd/pyOCD/files/8911701/lsusb.txt)

With this fix the results are now:

```
#   Probe/Board                  Unique ID      Target  
----------------------------------------------------------
  0   CMSIS-DAP by ARM CMSIS_DAP   202108120001   n/a     

```

Thanks

flit, 2022-06-15:
Interesting, thanks. And thanks for providing the lsusb output! That's a big help.

Do you have more information about the device? Like a link to where you purchased it? Just curious. We haven't seen this one before.

Also, did you have to add `DAP_LINK` or `CMSIS_DAP` to the known device strings list? In the issue description you said you added `DAP_LINK`, but the lsusb output shows `CMSIS_DAP` for the iProduct string, so I want to make sure I get it right since I can't directly test.

flit, 2022-06-16:
Closing now the fix is merged to `develop`.

letshin, 2022-11-09:
I bought the same device from eBay as well.

In case anyone has an issue getting this to work, you will need to install the prerelease version from the HEAD of the develop branch. 

@wolfmanjm Thanks - saves a lot of trouble troubleshooting why it doesn't work.

The stable version (0.34.2) does not work (eg device c251:f001 is not detected) while 0.34.3.dev18 detects the debugger.

PR #1421 (2022-06-16): probe: cmsis-dap: add "CMSIS_DAP" to known device strings
Fixes #1420
commit 94dd8fcf (2022-06-16): probe: cmsis-dap: add "CMSIS_DAP" to known device strings (#1421)

Fixes #1420