pyocd/pyOCD #1422: Waiting for a debug probe to be connected...

xinkunlong, 2022-06-20:
```pyinstaller info
55 INFO: PyInstaller: 5.1
55 INFO: Python: 3.9.6
73 INFO: Platform: Windows-10-10.0.19042-SP0
75 INFO: UPX is not available.
76 INFO: Extending PYTHONPATH with paths    
```
```Python
from pyocd.core.helpers import ConnectHelper

import libusb_package
import usb.core
import usb.backend.libusb1

libusb1_backend = usb.backend.libusb1.get_backend(find_library=libusb_package.find_library)
print(list(usb.core.find(find_all=True, backend=libusb1_backend)))

def main():
    debuggers = ConnectHelper.get_all_connected_probes(blocking=True)
    print(debuggers[0].description)

if __name__ == '__main__':
    main()
```

```spec
# -*- mode: python ; coding: utf-8 -*-

block_cipher = None

binaries = [("C:\\Windows\\SysWOW64\\libusb-1.0.dll", ".")]
a = Analysis(
    ['testPyocd.py'],
    pathex=[],
    binaries=binaries,
    datas=[],
    hiddenimports=[],
    hookspath=[],
    hooksconfig={},
    runtime_hooks=[],
    excludes=[],
    win_no_prefer_redirects=False,
    win_private_assemblies=False,
    cipher=block_cipher,
    noarchive=False,
)
pyz = PYZ(a.pure, a.zipped_data, cipher=block_cipher)

exe = EXE(
    pyz,
    a.scripts,
    a.binaries,
    a.zipfiles,
    a.datas,
    [],
    name='testPyocd',
    debug=False,
    bootloader_ignore_signals=False,
    strip=False,
    upx=True,
    upx_exclude=[],
    runtime_tmpdir=None,
    console=True,
    disable_windowed_traceback=False,
    argv_emulation=False,
    target_arch=None,
    codesign_identity=None,
    entitlements_file=None,
)

```
An exception occurs with the pyinstaller packer.

xinkunlong, 2022-06-20:
[image]

xinkunlong, 2022-06-20:
[image]
```
python testPyocd.py
```
In this way, it works fine

flit, 2022-06-24:
Closing as duplicate of #1358.

closed 2022-06-24 by flit