pyocd/pyOCD #1423: Can not hit the breakpoint.

swordligit, 2022-06-23:
My board has the cortex-m55 cpu, and I use Segger Jlink as the probe.
When I use pyocd gdbserver, after setting the breakpoint in gdb(
version is gcc-arm-none-eabi-10-2020-q4-major), the breakpoint can not  be hitted,
But I can debug step by step, and even I can use info threads, 
Seems all other functions are right.

With Segger's Jlinkgdbserver, the breakpoint can be hitted.

more info:
```
pyocd -V
0.34.1.dev3

pyocd list
  #   Probe/Board     Unique ID   Target  
------------------------------------------
  0   Segger J-Link   50120677    n/a  

pyocd gdbserver
0001370 I Target type is cortex_m [board]
0001508 I DP IDR = 0x4c013477 (v3 MINDP rev4) [dap]
0001518 I AHB5-AP@0x0 IDR = 0x34770005 (AHB5-AP var0 rev3) [ap]
0001527 I AHB5-AP@0x0 Class 0x1 ROM table #0 @ 0xe00ff000 (designer=43b:Arm part=4d2) [rom_table]
0001535 I [0]<e000e000:SCS M55 class=9 designer=43b:Arm part=d22 devtype=00 archid=2a04 devid=0:0:0> [rom_table]
0001540 I [1]<e0001000:DWT M55 class=9 designer=43b:Arm part=d22 devtype=00 archid=1a02 devid=0:0:0> [rom_table]
0001546 I [2]<e0002000:BPU M55 class=9 designer=43b:Arm part=d22 devtype=00 archid=1a03 devid=0:0:0> [rom_table]
0001551 I [6]<e0003000:PMU M55 class=9 designer=43b:Arm part=d22 devtype=16 archid=0a06 devid=0:0:0> [rom_table]
0001556 I [7]<e0042000:CTI M55 class=9 designer=43b:Arm part=d22 devtype=14 archid=1a14 devid=40800:0:0> [rom_table]
0001561 I CPU core #0 is Cortex-M55 r0p1 (security ext present) [cortex_m_v8m]
0001562 I FPU present: FPv5-D16-M [cortex_m]
0001563 I 4 hardware watchpoints [dwt]
0001567 I 8 hardware breakpoints, 1 literal comparators [fpb]
0001575 I Semihost server started on port 4444 (core 0) [server]
0001600 I GDB server started on port 3333 (core 0) [gdbserver]
```
Please give some advice, thanks.

swordligit, 2022-06-25:
- What board and MCU are you using?

The board and MCU are built by ourself(we are chip maker), but the cpu
is stardard cortex-m55.

- What is the memory layout of the MCU?

```
arm-none-eabi-readelf -S myprogram.elf
 [ 0]                   NULL            00000000 000000 000000 00      0   0  0
  [ 1] .boot_struct      PROGBITS        2c000000 010000 000010 00   A  0   0  4
  [ 2] .boot_text_flash  PROGBITS        2c000010 010010 002bfc 00  AX  0   0  8
  [ 3] .got_info         PROGBITS        2c002c0c 1f2fcc 000000 00   W  0   0  1
  [ 4] .vector_table     NOBITS          200c0000 010000 000168 00  WA  0   0  4
....
....
  [40] .text             PROGBITS        2c007c20 1d7c20 009830 00  AX  0   0 16
  [41] .ARM.exidx        ARM_EXIDX       2c011450 1e1450 000008 00  AL 40   0  4
  [42] .rodata           PROGBITS        2c011458 1e1458 0019b0 00  WA  0   0  4
  [43] .zInit            PROGBITS        2c012e08 1f2fcc 000000 00   W  0   0  1
  [44] .data             PROGBITS        200c0fe4 1f0fe4 000078 00  WA  0   0  4
  [45] .bss              NOBITS          200c105c 010000 00b0c0 00  WA  0   0  4
...
```
I remove some unused sections. the ram is from 0x2000 0000--0x2020 0000
and the flash is 0x2c00 0000-0x2c20 0000

- Where is your code placed in memory?

my code I want to debug is placed in flash.

- What address is the breakpoint you're trying to place?

in the gdb, 
```
(gdb) info breakpoints 
Num     Type           Disp Enb Address    What
1       breakpoint     keep y   0x2c00828c in main() at ../../platform/main/noapp_main.cpp:287
````

- And more info:

when I use JLinkExe to start the probe, it show:
```
J-Link>connect
Device "CORTEX-M55" selected.

Connecting to target via SWD
Found SW-DP with ID 0x4C013477
DPIDR: 0x4C013477
DPv3 detected
Detecting available APs
APSpace base (BASEPTR0): 0x00000000
APSpace size (DPIDR1.ASIZE): 12-bit (4 KiB)
AP[0]: Stopped AP scan as end of AP map has been reached
AP[0] (APAddr 0x00000000): AHB-AP (IDR: 0x34770005)
Iterating through AP map to find AHB-AP to use
AP[0]: Core found
AP[0]: AHB-AP ROM base: 0xE00FF000
CPUID register: 0x410FD221. Implementer code: 0x41 (ARM)
Feature set: Mainline
Found Cortex-M55 r0p1, Little endian.
FPUnit: 8 code (BP) slots and 0 literal slots
Security extension: implemented
Secure debug: enabled
CoreSight components:
ROMTbl[0] @ E00FF000
ROMTbl[0][0]: E000E000, CID: B105900D, PID: 000BBD22 ???
ROMTbl[0][1]: E0001000, CID: B105900D, PID: 000BBD22 DWT
ROMTbl[0][2]: E0002000, CID: B105900D, PID: 000BBD22 FPB
ROMTbl[0][6]: E0003000, CID: B105900D, PID: 000BBD22 ???
ROMTbl[0][7]: E0042000, CID: B105900D, PID: 000BBD22 CSS600-CTI
Cortex-M55 identified.
```

flit, 2022-06-25:
Thanks for the info! The flash being located in the Data (aka SRAM) space of the M-profile architectural memory map is unusual. However, it shouldn't cause performance problems on the Cortex-M55 like it would on the Cortex-M33 and others. Otherwise I don't see anything that stands out.

Cortex-M55 silicon is still very rare. Although I've tested on an FPGA, that's never quite the same as a real MCU, especially with flash. So I'm not entirely surprised there are some issues.

Guesses for the root cause:
- Pyocd (or gdb) is trying to use a software breakpoint. Since the generic `cortex-m` target type is being used, it doesn't know that flash is non-writable. The flash being located in the Data space might contribute to this (on either the pyocd or gdb side).
- Security level issue. Do you have any security controls configured (SAU, IDAU, bus responder security checkers, etc)?

Try telling gdb to use a hardware breakpoint by using the `hbreak` command.

If that doesn't work, let's get some logs from pyocd. Could you please provide a debug log from pyocd? Just run `pyocd gdb -vv 2> pyocd_debug.log`. (That's a unix shell redirection of stderr, not sure if it will work the same for Windows.)
Also, let's go ahead and get a low-level trace log. For this, please run `pyocd gdb -vv -L "pyocd.coresight.*ap.trace=debug" 2> pyocd_trace.log`. This will be a quite large log file.

These logs should provide everything necessary to understand the problem. You can attach the logs to a comment here. Thanks!

swordligit, 2022-06-27:
- Try telling gdb to use a hardware breakpoint by using the hbreak command.
This works.
```
gdb) hbreak 287
Hardware assisted breakpoint 1 at 0x2c00828c: file ../../platform/main/noapp_main.cpp, line 287.
(gdb) set abc=0
(gdb) c
Continuing.
[New Thread 537666960]
[New Thread 537700256]
[New Thread 537700168]
[Switching to Thread 537666960]

Thread 2 "main" hit Breakpoint 1, main () at ../../platform/main/noapp_main.cpp:287
287	    analog_open();
(gdb) info breakpoints 
Num     Type           Disp Enb Address    What
1       hw breakpoint  keep y   0x2c00828c in main() at ../../platform/main/noapp_main.cpp:287
	breakpoint already hit 1 time

```
I still attach the log file, please verify it.
[pyocd_trace.log](https://github.com/pyocd/pyOCD/files/8988536/pyocd_trace.log)
Thanks for your kindly help.

closed 2022-06-27 by swordligit

flit, 2022-06-27:
Great! I'm happy that worked for you.

Btw, are your Cortex-M55 devices going to be publicly available soon? I'd like to have some M55 silicon to test with. If you don't want to talk in this open forum, you can email me (see [my profile](https://github.com/flit) for my Arm email address). Presumably Arm has an NDA with you.