pyocd/pyOCD #1439: `monitor rtt start` not supported

werediver, 2022-08-11:
RTT support is super-useful, but without support in GDB server it's not available for integration, e.g. in https://github.com/Marus/cortex-debug

As I understand, `pyocd rtt` command cannot be used when a debugging session is active, which limits the use cases further.

Am I missing something? Is there any ongoing work on this topic or, perhaps, some plans? 👀

flit, 2022-10-22:
(Sorry for late reply; I was sick for a long time.)

You're right that it's not possible to use the RTT support with cortex-debug (which I use myself). Right now the RTT support is entirely contained within the `pyocd rtt` subcommand. Fortunately, there is ongoing work (and some pull requests) to improve RTT support, and integration with the gdb server and output in a form consumable by cortex-debug is on the roadmap.

baneageorge, 2022-11-09:
@flit hopefully nothing serious

It would also be very cool to have the possibility to flash while rtt is open. I am able to do this with jlink.
I normally keep rtt open and repeatedly make code changes and flash.

werediver, 2022-11-09:
> It would also be very cool to have the possibility to flash while rtt is open. I am able to do this with jlink.
I normally keep rtt open and repeatedly make code changes and flash.

Same is possible with the Black Magic Probe (BMP).

karlp, 2026-04-14:
The monitor RTT interface is still basically undocumented, see https://pyocd.io/docs/command_reference.html#rtt  perhaps this ticket could be used for expanding that?