pyocd/pyOCD #1444: stlink_probe.read_memory_block8 returning extra bytes on un-aligned 1 byte read

pdietl, 2022-08-19:
It seems to me that commit 46f1ac66c26169270b6c6e3d7992f3b106a8077b introduced a bug. The function stlink_probe.read_memory_block8 will return an array with two bytes in it if the addressed is aligned on a 2-byte boundary but a read of 1 byte (size=1) is requested. This ends up causing an assertion at `/memory.py(263)read_memory_block8()`: `0107462 E Unhandled exception in handle_message (b'v'): result size (2) != requested size (1) [gdbserver]`

I assume we need to add something along the lines of `array[0]` somewhere, though I am not sure about the appropriate place. 

@flit

flit, 2022-10-22:
Thanks for the detailed report! (And very sorry it took a long time.) Fortunately, it looks like this was fixed in #1428 and released in v0.34.1. If you still have a problem, please reopen this issue.

closed 2022-10-22 by flit

DariusBabrauskas, 2022-11-09:
v0.34.2 still have same problem. #1428  not fix this problem.  But error in same place.  Error occurs if (address & 3) >1   and read size 1 or 2.