pyocd/pyOCD #1450: Got flash algorithm overflowed stack when using flash loader converted from KEIL pack FLM

majorlin, 2022-09-18:
I got following errors when using flash loader converted from Keil pack FLM, the FLM working well under KEIL and Segger JLINK, but failed when using in pyocd, seems pyocd could only download one flash sector.

```
[=================================                 ]  68%0003571 C flash algorithm overflowed stack (4096 bytes) [__main__]
Traceback (most recent call last):
  File "/opt/homebrew/lib/python3.10/site-packages/pyocd/__main__.py", line 161, in run
    status = cmd.invoke()
  File "/opt/homebrew/lib/python3.10/site-packages/pyocd/subcommands/load_cmd.py", line 125, in invoke
    programmer.program(filename,
  File "/opt/homebrew/lib/python3.10/site-packages/pyocd/flash/file_programmer.py", line 170, in program
    self._loader.commit()
  File "/opt/homebrew/lib/python3.10/site-packages/pyocd/flash/loader.py", line 289, in commit
    perf = builder.program(chip_erase=chipErase,
  File "/opt/homebrew/lib/python3.10/site-packages/pyocd/flash/builder.py", line 509, in program
    flash_operation = self._sector_erase_program_double_buffer(progress_cb)
  File "/opt/homebrew/lib/python3.10/site-packages/pyocd/flash/builder.py", line 949, in _sector_erase_program_double_buffer
    result = self.flash.wait_for_completion(timeout=program_timeout)
  File "/opt/homebrew/lib/python3.10/site-packages/pyocd/flash/flash.py", line 648, in wait_for_completion
    raise exceptions.FlashFailure(f"flash algorithm overflowed stack ({self.begin_stack - self.end_stack} bytes)")
pyocd.core.exceptions.FlashFailure: flash algorithm overflowed stack (4096 bytes)
```

majorlin, 2022-09-18:
I tried to add offset 0x1000 to ``begin_stack`` and ``end_stack``, Now I can program flash successfully.

hagibr, 2025-05-26:
Similar problem here, the `FLASH_ALGO` generated by `generate_flash_algo.py` for stm32c031 didn't work at first. The problem was that pyocd was considering the `page_size` == 0x800, although the generated value was 0x400. 0x800 is the sector size.
So I've had to change the page_buffers to accomodate these 0x800 byte sized pages and also change `end_stack `and `begin_stack `to avoid overlapping with page_buffers.
The strange part is that by using cmsis pack directly, it considers `sector_size` == 0x800 and `page_size` == 0x400.
Output using built in:
```Erased 28672 bytes (14 sectors), programmed 28672 bytes (14 pages), skipped 0 bytes (0 pages) at 19.21 kB/s```
Output using cmsis pack:
```Erased 28672 bytes (14 sectors), programmed 27648 bytes (27 pages), skipped 0 bytes (0 pages) at 17.14 kB/s```

I'm using `pyocd==0.36.0`