pyocd/pyOCD #1455: Unable to start SWV/SWO trace export on nrf52840-dk (JLink OB)

bbain, 2022-10-08:
When I try to start swv for the  nrf52840-dk  I get the following error:

```
pyocd gdbserver --target nrf52840 -S -Oswv_clock=4000000 -O swv_system_clock=32000000 -Oenable_swv=true -vvvv
<snip>
0001039 I Target type is nrf52840 [board]
0001148 D Running task load_svd [sequencer]
0001148 D Running task pre_connect [sequencer]
0001148 D Running task dp_init [sequencer]
0001148 D Running task lock_probe [sequencer]
0001148 D Running task get_probe_capabilities [sequencer]
0001149 D Running task connect [sequencer]
0001153 D Default wire protocol selected; using SWD [dap]
0001155 I DP IDR = 0x2ba01477 (v1 rev2) [dap]
0001155 D Running task clear_sticky_err [sequencer]
0001155 D Running task power_up_debug [sequencer]
0001158 D Running task check_version [sequencer]
0001158 D Running task unlock_probe [sequencer]
0001158 D Running task create_discoverer [sequencer]
0001158 D Running task discovery [sequencer]
0001158 D Running task find_aps [sequencer]
0001168 D Running task create_aps [sequencer]
0001168 D Running task create_ap.0 [sequencer]
0001171 I AHB-AP#0 IDR = 0x24770011 (AHB-AP var1 rev2) [ap]
0001188 D AHB-AP#0 default HPROT=3 HNONSEC=0 [ap]
0001205 D AHB-AP#0 implemented HPROT=3 HNONSEC=0 [ap]
0001237 D Running task create_ap.1 [sequencer]
0001238 I AP#1 IDR = 0x02880000 (AP var0 rev0) [ap]
0001238 D Running task check_ctrl_ap_idr [sequencer]
0001238 D CTRL-AP version 0 [target_nRF52]
0001238 D Running task check_flash_security [sequencer]
0001239 I NRF52840 not in secure state [target_nRF52]
0001239 D Running task find_components [sequencer]
0001239 D Running task init_ap.0 [sequencer]
0001244 I AHB-AP#0 Class 0x1 ROM table #0 @ 0xe00ff000 (designer=244 part=008) [rom_table]
0001249 I [0]<e000e000:SCS v7-M class=14 designer=43b:Arm part=00c> [rom_table]
0001253 I [1]<e0001000:DWT v7-M class=14 designer=43b:Arm part=002> [rom_table]
0001256 I [2]<e0002000:FPB v7-M class=14 designer=43b:Arm part=003> [rom_table]
0001259 I [3]<e0000000:ITM v7-M class=14 designer=43b:Arm part=001> [rom_table]
0001263 I [4]<e0040000:TPIU M4 class=9 designer=43b:Arm part=9a1 devtype=11 archid=0000 devid=ca1:0:0> [rom_table]
0001268 I [5]<e0041000:ETM M4 class=9 designer=43b:Arm part=925 devtype=13 archid=0000 devid=0:0:0> [rom_table]
0001268 D Running task create_cores [sequencer]
0001268 D Creating SCS component [discovery]
0001268 D selected core #0 [soc_target]
0001269 I CPU core #0 is Cortex-M4 r0p1 [cortex_m]
0001270 I FPU present: FPv4-SP-D16-M [cortex_m]
0001270 D Running task create_components [sequencer]
0001270 D Creating DWT component [discovery]
0001271 I 4 hardware watchpoints [dwt]
0001273 D Creating FPB component [discovery]
0001274 I 6 hardware breakpoints, 4 literal comparators [fpb]
0001274 D fpb has been disabled [fpb]
0001277 D Creating ITM component [discovery]
0001278 D Creating TPIU component [discovery]
0001279 D Running task check_for_cores [sequencer]
0001279 D Running task halt_on_connect [sequencer]
0001279 D halting core 0 [cortex_m]
0001279 D Running task post_connect [sequencer]
0001279 D Running task post_connect_hook [sequencer]
0001279 D Running task create_flash [sequencer]
0001280 D Running task notify [sequencer]
0001280 D Setting vector catch to 0x00000001 [cortex_m]
0001282 I Semihost server started on port 4444 (core 0) [server]
0001287 I Set SWO clock to 4000000 [swv]
0001288 I SWV raw server started on port 3443 [server]
Exception in thread SWVReader:
Traceback (most recent call last):
  File "/home/guest/.pyenv/versions/3.10.7/envs/pyocd/lib/python3.10/site-packages/pyocd/probe/jlink_probe.py", line 341, in swo_start
    self._link.swo_start(baudrate)
  File "/home/guest/.pyenv/versions/3.10.7/envs/pyocd/lib/python3.10/site-packages/pylink/jlink.py", line 172, in wrapper
    raise errors.JLinkException('Target is not connected.')
pylink.errors.JLinkException: Target is not connected.

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/guest/.pyenv/versions/3.10.7/lib/python3.10/threading.py", line 1016, in _bootstrap_inner
    self.run()
  File "/home/guest/.pyenv/versions/3.10.7/envs/pyocd/lib/python3.10/site-packages/pyocd/trace/swv.py", line 187, in run
    self._session.probe.swo_start(self._swo_clock)
  File "/home/guest/.pyenv/versions/3.10.7/envs/pyocd/lib/python3.10/site-packages/pyocd/probe/jlink_probe.py", line 343, in swo_start
    raise self._convert_exception(exc) from exc
pyocd.core.exceptions.ProbeError: Target is not connected.
0001320 I GDB server started on port 3333 (core 0) [gdbserver]
```

bbain, 2022-10-08:
```
pyocd --version
0.34.1
```

elfmimi, 2023-12-28:
Direct cause of above exception message is that jlink itself requires to know what the target is.
passing `-O jlink.device=nRF52840_xxAA` will suffice that requirement.

but then immediately another problem comes up.
modifying,
https://github.com/pyocd/pyOCD/blob/5ab5c327987c4c3ba5a37928cac25263d4db7586/pyocd/probe/jlink_probe.py#L483-L487
to something like,
```python
    def swo_read(self):
        num_bytes = self._link.swo_num_bytes()
        if num_bytes == 0:
            return []
        try:
            return self._link.swo_read(0, num_bytes, True)
        except JLinkException as exc:
            raise self._convert_exception(exc) from exc
```
solve that next problem.

note: I've never tested swo/swv in action.