pyocd/pyOCD #1458: pyOCD does not erase/flash nRF9160

beethoven62, 2022-10-20:
I'm using a DAPlink adapter over SWD 10-pin connecter to a Nordic Semi development board (nRF9160DK) containing an nRF9160 which has an ARM-Cortex M33 core. I downloaded and installed the Nordic Pack that supports this chip. I also have a J-Link adapter which has no problem erasing and flashing this board. I've heard that pyOCD supports other Nordic MCUs (nRF52 series), but I don't think this has been tested with the nRF9160. I've included a verbose log of an attempt to erase the nRF9160 using the development board.
[pyOCD_debug_log_20221017.txt](https://github.com/pyocd/pyOCD/files/9830516/pyOCD_debug_log_20221017.txt)

flit, 2022-10-23:
Thanks for creating the issue to track this.

(Copy of my reply from Slack:)
It’s not really clear what’s going on here. The CoreSight structure of the device is discovered correctly, except that AP#1 is disabled (this is probably the second core, disabled by default like the nRF5340). (It has a surprisingly complex ETM trace subsystem!)
Pyocd starts the erase process by resetting the chip, then loading and calling the flash algo init. That appears to go well. But when it calls the flash algo erase all routine, the chip just hangs and doesn’t respond to an SWD transaction (it times out). Without having the code to the flash algo, it’ll be rather difficult to figure out, since we don’t know what Nordic’s flash algo implementation does…

Reading the nRF9160 reference manual and looking for any special requirements for chip erase and/or flash programming is a good starting point. Sorry there’s not an easy answer—this is pretty common with unusually complex MCUs like the nRF9160.

Anyway, I’ve ordered a SparkFun nRF9160 Thing Plus so I can take a look at it myself.

maz3max, 2024-01-01:
not sure if this is still relevant - but it looks like it's related to the nrf91 stalling the interface while erasing. should be fixed in the most recent version though