pyocd/pyOCD #1463: Target type stm32g070cb not recognized

USTHzhanglu, 2022-10-26:
When I use PYOCD for stm32g070, it tell me that 
```
 C Target type stm32g070cb not recognized. Use 'pyocd list --targets' to see currently available target types. See <https://pyocd.io/docs/target_support.html> for how to install additional target support. [__main__]
```

Then i use " pyocd pack find stm32g070cb",it show:
```
pyocd pack find stm32g070cb
  Part            Vendor               Pack                 Version   Installed
---------------------------------------------------------------------------------
  STM32G070CBTx   STMicroelectronics   Keil.STM32G0xx_DFP   1.4.0     True
```

How can i used it？thanks

mbrossard, 2023-02-17:
Have you tried passing `-t STM32G070CBTx`?

USTHzhanglu, 2023-02-21:
> Have you tried passing `-t STM32G070CBTx`?

it works. very thinks. pyocd is good

USTHzhanglu, 2023-02-21:
passing -t STM32G070CBTx

closed 2023-02-21 by USTHzhanglu