pyocd/pyOCD #1465: Cannot flash/debug STM32WL55JC

HelloHungryImDad, 2022-11-01:
Hello,

When trying to flash a Nucleo WL55JC1 (stm32wl55jci7u) using the command `pyocd flash --target STM32WL55JCIx --frequency 1800000 -O connect_mode=under-reset mbed-os-example-blinky.bin` I receive transfer errors. I am running the latest ST Link firmware (3.10.3) and have installed the `Keil.STM32WLxx_DFP.1.1.0-small.pack` CMSIS pack for this chip using `pyocd pack install` but it is the same if I use the one bundled with Mbed Studio or one downloaded directly from CMSIS.

On first attempt I get:

```
0000574 E Transfer error while reading AHB-AP#1 ROM table: Memory transfer fault (read) @ 0xf0000fd0-0xf000102f [ap]
Traceback (most recent call last):
  File "/home/rikki/.local/lib/python3.10/site-packages/pyocd/coresight/ap.py", line 797, in find_components
    cmpid.read_id_registers()
  File "/home/rikki/.local/lib/python3.10/site-packages/pyocd/coresight/rom_table.py", line 124, in read_id_registers
    regs = self.ap.read_memory_block32(self.top_address + self.IDR_READ_START, self.IDR_READ_COUNT)
  File "/home/rikki/.local/lib/python3.10/site-packages/pyocd/utility/concurrency.py", line 29, in _locking
    return func(self, *args, **kwargs)
  File "/home/rikki/.local/lib/python3.10/site-packages/pyocd/coresight/ap.py", line 1216, in _accelerated_read_memory_block32
    return self._accelerated_memory_interface.read_memory_block32(addr, size,
  File "/home/rikki/.local/lib/python3.10/site-packages/pyocd/probe/stlink_probe.py", line 301, in read_memory_block32
    return conversion.byte_list_to_u32le_list(self._link.read_mem32(addr, size * 4, self._apsel, csw))
  File "/home/rikki/.local/lib/python3.10/site-packages/pyocd/probe/stlink/stlink.py", line 473, in read_mem32
    return self._read_mem(addr, size, Commands.JTAG_READMEM_32BIT, self.MAXIMUM_TRANSFER_SIZE, apsel, csw)
  File "/home/rikki/.local/lib/python3.10/site-packages/pyocd/probe/stlink/stlink.py", line 423, in _read_mem
    raise exc
pyocd.core.exceptions.TransferFaultError: Memory transfer fault (read) @ 0xf0000fd0-0xf000102f
0000591 I Loading /home/rikki/Mbed Programs/mbed-os-example-blinky-baremetal/BUILD/NUCLEO_WL55JC/ARMC6/mbed-os-example-blinky-baremetal.bin [load_cmd]
[==================================================] 100%
0001317 I Erased chip, programmed 23552 bytes (23 pages), skipped 0 bytes (0 pages) at 31.69 kB/s [loader]
```

and on subsequent attempts I get:

```
0000632 E Error reading AP#1 IDR: Memory transfer fault (STLink error (17): AP fault) [discovery]
Traceback (most recent call last):
  File "/home/rikki/.local/lib/python3.10/site-packages/pyocd/coresight/discovery.py", line 162, in _create_1_ap
    ap = AccessPort.create(self.dp, ap_address)
  File "/home/rikki/.local/lib/python3.10/site-packages/pyocd/coresight/ap.py", line 367, in create
    ap.init()
  File "/home/rikki/.local/lib/python3.10/site-packages/pyocd/utility/concurrency.py", line 29, in _locking
    return func(self, *args, **kwargs)
  File "/home/rikki/.local/lib/python3.10/site-packages/pyocd/coresight/ap.py", line 1256, in init
    super().init()
  File "/home/rikki/.local/lib/python3.10/site-packages/pyocd/utility/concurrency.py", line 29, in _locking
    return func(self, *args, **kwargs)
  File "/home/rikki/.local/lib/python3.10/site-packages/pyocd/coresight/ap.py", line 778, in init
    _init_hprot()
  File "/home/rikki/.local/lib/python3.10/site-packages/pyocd/coresight/ap.py", line 746, in _init_hprot
    csw = AccessPort.read_reg(self, self._reg_offset + MEM_AP_CSW)
  File "/home/rikki/.local/lib/python3.10/site-packages/pyocd/utility/concurrency.py", line 29, in _locking
    return func(self, *args, **kwargs)
  File "/home/rikki/.local/lib/python3.10/site-packages/pyocd/coresight/ap.py", line 453, in read_reg
    return self.dp.read_ap(self.address.address + addr, now)
  File "/home/rikki/.local/lib/python3.10/site-packages/pyocd/coresight/dap.py", line 875, in read_ap
    result_cb = self.probe.read_ap(addr, now=False)
  File "/home/rikki/.local/lib/python3.10/site-packages/pyocd/probe/stlink_probe.py", line 207, in read_ap
    result = self._link.read_dap_register(apsel, addr & 0xffff)
  File "/home/rikki/.local/lib/python3.10/site-packages/pyocd/probe/stlink/stlink.py", line 521, in read_dap_register
    self._check_status(response[:2])
  File "/home/rikki/.local/lib/python3.10/site-packages/pyocd/probe/stlink/stlink.py", line 363, in _check_status
    raise self._ERROR_CLASSES[status](error_message)
pyocd.core.exceptions.TransferFaultError: Memory transfer fault (STLink error (17): AP fault)
0000657 I Loading /home/rikki/Mbed Programs/mbed-os-example-blinky-baremetal/BUILD/NUCLEO_WL55JC/ARMC6/mbed-os-example-blinky-baremetal.bin [load_cmd]
[==================================================] 100%
0001769 I Erased 24576 bytes (12 sectors), programmed 23552 bytes (23 pages), skipped 0 bytes (0 pages) at 20.69 kB/s [loader]
```

Is this indicating that the pack is malformed? How do I fix this?

HelloHungryImDad, 2022-11-01:
Running with `-vv` gives the following additional information:

```
0000573 D [Errno 13] Access denied (insufficient permissions) while trying to interrogate a USB device (VID=0bda PID=8153). This can probably be remedied with a udev rule. See <https://github.com/pyocd/pyOCD/tree/master/udev> for help. [pyusb_backend]
0000576 D [Errno 13] Access denied (insufficient permissions) while trying to interrogate a USB device (VID=04ca PID=3802). This can probably be remedied with a udev rule. See <https://github.com/pyocd/pyOCD/tree/master/udev> for help. [pyusb_backend]
0000576 D [Errno 13] Access denied (insufficient permissions) while trying to interrogate a USB device (VID=046a PID=0076). This can probably be remedied with a udev rule. See <https://github.com/pyocd/pyOCD/tree/master/udev> for help. [pyusb_backend]
0000576 D [Errno 13] Access denied (insufficient permissions) while trying to interrogate a USB device (VID=0461 PID=4e22). This can probably be remedied with a udev rule. See <https://github.com/pyocd/pyOCD/tree/master/udev> for help. [pyusb_backend]
0000576 D [Errno 13] Access denied (insufficient permissions) while trying to interrogate a USB device (VID=0408 PID=a094). This can probably be remedied with a udev rule. See <https://github.com/pyocd/pyOCD/tree/master/udev> for help. [pyusb_backend]
0000577 D [Errno 13] Access denied (insufficient permissions) while trying to interrogate a USB device (VID=0bda PID=8153). This can probably be remedied with a udev rule. See <https://github.com/pyocd/pyOCD/tree/master/udev> for help. [pyusb_v2_backend]
0000580 D [Errno 13] Access denied (insufficient permissions) while trying to interrogate a USB device (VID=04ca PID=3802). This can probably be remedied with a udev rule. See <https://github.com/pyocd/pyOCD/tree/master/udev> for help. [pyusb_v2_backend]
0000580 D [Errno 13] Access denied (insufficient permissions) while trying to interrogate a USB device (VID=046a PID=0076). This can probably be remedied with a udev rule. See <https://github.com/pyocd/pyOCD/tree/master/udev> for help. [pyusb_v2_backend]
0000581 D [Errno 13] Access denied (insufficient permissions) while trying to interrogate a USB device (VID=0461 PID=4e22). This can probably be remedied with a udev rule. See <https://github.com/pyocd/pyOCD/tree/master/udev> for help. [pyusb_v2_backend]
0000581 D [Errno 13] Access denied (insufficient permissions) while trying to interrogate a USB device (VID=0408 PID=a094). This can probably be remedied with a udev rule. See <https://github.com/pyocd/pyOCD/tree/master/udev> for help. [pyusb_v2_backend]
0000581 D Project directory: /home/rikki/Downloads/Keil.STM32WLxx_DFP.1.1.0 [session]
0000587 D STLink probe 0037001A3756501620303658 firmware version: V3J10M3 [stlink]
0000588 D Project directory: /home/rikki/Downloads/Keil.STM32WLxx_DFP.1.1.0 [session]
0000828 D flash algo: [code=0x254] [b1=0x260,0x660] [b2=0x660,0xa60] [stack=0x2a60; 0x2000 b] (ram=0x20000000, 0x8000 b) [flash_algo]
0000831 I Target type is stm32wl55jcix [board]
0000834 D STLink probe 0037001A3756501620303658 firmware version: V3J10M3 [stlink]
0000834 D actual SWD frequency is 1000 kHz [stlink]
0000834 D Running task load_svd [sequencer]
0000891 D Running task pre_connect [sequencer]
0000897 I Asserting reset prior to connect [coresight_target]
0000908 D Running task dp_init [sequencer]
0000913 D Running task lock_probe [sequencer]
0000919 D Running task get_probe_capabilities [sequencer]
0000925 D Running task connect [sequencer]
0000947 D Default wire protocol selected; using SWD [dap]
0000953 I DP IDR = 0x6ba02477 (v2 rev6) [dap]
0000953 D Running task clear_sticky_err [sequencer]
0000953 D Running task power_up_debug [sequencer]
0000954 D Running task check_version [sequencer]
0000954 D Running task unlock_probe [sequencer]
0000954 D Running task create_discoverer [sequencer]
0000954 D Running task discovery [sequencer]
0000954 D Running task find_aps [sequencer]
0000956 D Running task create_aps [sequencer]
0000956 D Running task create_ap.0 [sequencer]
0000957 D Using accelerated memory access interface for AHB-AP#0 [ap]
0000957 I AHB-AP#0 IDR = 0x24770011 (AHB-AP var1 rev2) [ap]
0000958 D AHB-AP#0 default HPROT=3 HNONSEC=0 [ap]
0000959 D AHB-AP#0 implemented HPROT=3 HNONSEC=0 [ap]
0000960 D Running task create_ap.1 [sequencer]
0000960 D Using accelerated memory access interface for AHB-AP#1 [ap]
0000960 I AHB-AP#1 IDR = 0x84770001 (AHB-AP var0 rev8) [ap]
0000961 D AHB-AP#1 default HPROT=3 HNONSEC=1 [ap]
0000962 D AHB-AP#1 implemented HPROT=f HNONSEC=1 [ap]
0000963 D Running task find_components [sequencer]
0000963 D Running task init_ap.0 [sequencer]
0000967 I AHB-AP#0 Class 0x1 ROM table #0 @ 0xe00ff000 (designer=020:ST part=497) [rom_table]
0000970 I [0]<e000e000:SCS v7-M class=14 designer=43b:Arm part=000> [rom_table]
0000971 I [1]<e0001000:DWT v7-M class=14 designer=43b:Arm part=002> [rom_table]
0000972 I [2]<e0002000:FPB v7-M class=14 designer=43b:Arm part=003> [rom_table]
0000973 I [3]<e0000000:ITM v7-M class=14 designer=43b:Arm part=001> [rom_table]
0000975 I [4]<e0040000:TPIU M4 class=9 designer=43b:Arm part=9a1 devtype=11 archid=0000 devid=ca0:0:0> [rom_table]
0000975 D [5]<fff42002 not present> [rom_table]
0000977 I [6]<e0043000:CTI class=9 designer=43b:Arm part=906 devtype=14 archid=0000 devid=40800:0:0> [rom_table]
0000977 D Running task init_ap.1 [sequencer]
[... cut ...]
0002042 D reset, core 0, type=SW_SYSRESETREQ [cortex_m]
0002160 D clear reset catch, core 0 [cortex_m]
0002160 I Erased 24576 bytes (12 sectors), programmed 23552 bytes (23 pages), skipped 0 bytes (0 pages) at 19.91 kB/s [loader]
0002160 D uninit session <pyocd.core.session.Session object at 0x7f838d2e3310> [session]
0002161 D uninit board <pyocd.board.mbed_board.MbedBoard object at 0x7f838d2e2830> [board]
0002162 D resuming core 0 [cortex_m]
0002162 D added=[] removed=[] [manager]
0002162 D bps after flush={} [manager]
```

DariusBabrauskas, 2022-11-15:
Try my bug fix at #1428.

niliha, 2022-11-16:
Hi @HelloHungryImDad,

I'm having the exact same issue with a NUCLEO-WL55JC1 board.
I already updated the ST-Link firmware to the latest version (V3J10M3) using [STLinkUpgrade](https://www.st.com/en/development-tools/stsw-link007.html), but this didn't change anything.

Also trying the latest pyocd release (v0.34.2) didn't help.

As a workaround, I'm currently using the mass-storage flash method, i.e. I copy the compiled binary to the NUCLEO-WL55JC1 mass-storage device with a file manager. Afterwards, I have to unplug the board to make it reset and run the flashed program.

@DariusBabrauskas what do exactly mean with "your bug fix" ?
The mentioned PR (https://github.com/pyocd/pyOCD/pull/1428) contains a fix from @flit only, which is already included in the pyocd version I'm using.
Do you refer to your [comment](https://github.com/pyocd/pyOCD/pull/1428#pullrequestreview-1175229501) ?

DariusBabrauskas, 2022-11-16:
> 
Yes. See my comments. I add this changes directly into pyocd install site in python dir.  Example.: C:\Python38\Lib\site-packages\pyocd\probe\stlink_probe.py
If you using stlink adapter, then this fix  need to do. 
BR.

niliha, 2022-11-16:
@DariusBabrauskas
Do you mind sharing your diff for `stlink_probe.py` or the whole file?
That makes it easier for me to reproduce your results.
Thank you!

DariusBabrauskas, 2022-11-16:
[stlink_probe.py.log](https://github.com/pyocd/pyOCD/files/10021082/stlink_probe.py.log)
This from   (v0.34.2) with my fix.  Rename file to py, and replace with original.
BR.

niliha, 2022-11-16:
@DariusBabrauskas thanks for sharing the file.
Unfortunately, it didn't change anything.

niliha, 2022-11-21:
@DariusBabrauskas Yes I can do it with STM32CubeProgrammer.
I think the problem occurs before flashing, since even `pyocd reset` does not work.

Can you elaborate on the stop mode issue, i.e. how to avoid it.

DariusBabrauskas, 2022-11-21:
//insert this code anywhere and must be executed:
//#if CONFIG_DEBUG_SWD == 1
    LL_DBGMCU_EnableDBGSleepMode();//
//#endif
More: https://github.com/zephyrproject-rtos/zephyr/issues/35055