pyocd/pyOCD #1467: Musca-B1 gdb connection problem

danielRep, 2022-11-01:
Hi all, I'm getting back to using one of Arm's Musca boards, B1 specifically, after having used A1 on the last couple of weeks without any issue. 

The DAPLink is configured to QSPI Flash v3.4 and I'm using the last version of pyocd with xPack GNU Arm Embedded GCC  v12.1.90.
When I try to connect to the gdb server, the pyocd gets closed since the connection was closed by gdb.

Output:
```
0000396 D [Errno 13] Access denied (insufficient permissions) while trying to interrogate a USB device (VID=1ea7 PID=0064). This can probably be remedied with a udev rule. See <https://github.com/pyocd/pyOCD/tree/master/udev> for help. [pyusb_backend]
0000399 D [Errno 13] Access denied (insufficient permissions) while trying to interrogate a USB device (VID=1770 PID=ff00). This can probably be remedied with a udev rule. See <https://github.com/pyocd/pyOCD/tree/master/udev> for help. [pyusb_backend]
0000402 D [Errno 13] Access denied (insufficient permissions) while trying to interrogate a USB device (VID=1ea7 PID=0064). This can probably be remedied with a udev rule. See <https://github.com/pyocd/pyOCD/tree/master/udev> for help. [pyusb_v2_backend]
0000405 D [Errno 13] Access denied (insufficient permissions) while trying to interrogate a USB device (VID=1770 PID=ff00). This can probably be remedied with a udev rule. See <https://github.com/pyocd/pyOCD/tree/master/udev> for help. [pyusb_v2_backend]
0000405 D Project directory: /home/daniel [session]
0000406 D Project directory: /home/daniel [session]
0000414 D CMSIS-DAP v1 probe 500700001d14dad300000000000000000000000097969902: protocol version 1.0.0 [dap_access_cmsis_dap]
0000418 D closing interface [pyusb_backend]
0000420 I Target type is musca_b1 [board]
0000426 D Running task load_svd [sequencer]
0000427 D Running task pre_connect [sequencer]
0000427 D Running task dp_init [sequencer]
0000427 D Running task lock_probe [sequencer]
0000427 D Running task get_probe_capabilities [sequencer]
0000428 D Running task connect [sequencer]
0000453 D Default wire protocol selected; using SWD [dap]
0000455 D Sending deprecated SWJ sequence to select SWD [swj]
0000463 I DP IDR = 0x6ba02477 (v2 rev6) [dap]
0000463 D Running task clear_sticky_err [sequencer]
0000463 D Running task power_up_debug [sequencer]
0000465 D Running task check_version [sequencer]
0000465 D Running task unlock_probe [sequencer]
0000466 D Running task create_discoverer [sequencer]
0000466 D Running task discovery [sequencer]
0000466 D Running task find_aps [sequencer]
0000478 D Running task create_aps [sequencer]
0000478 D Running task create_ap.0 [sequencer]
0000480 I APB-AP#0 IDR = 0x54770002 (APB-AP var0 rev5) [ap]
0000486 D APB-AP#0 default HPROT=0 HNONSEC=0 [ap]
0000488 D APB-AP#0 implemented HPROT=0 HNONSEC=0 [ap]
0000490 D Running task create_ap.1 [sequencer]
0000492 I AHB-AP#1 IDR = 0x84770001 (AHB-AP var0 rev8) [ap]
0000496 D AHB-AP#1 default HPROT=3 HNONSEC=0 [ap]
0000499 E Error reading AP#1 IDR: Memory transfer fault [discovery]
Traceback (most recent call last):
  File "/home/daniel/.local/lib/python3.8/site-packages/pyocd/probe/cmsis_dap_probe.py", line 502, in read_ap_result_callback
    value = result()
  File "/home/daniel/.local/lib/python3.8/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 1060, in read_reg_cb
    res = transfer.get_result()
  File "/home/daniel/.local/lib/python3.8/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 157, in get_result
    self.daplink.flush()
  File "/home/daniel/.local/lib/python3.8/site-packages/pyocd/utility/concurrency.py", line 29, in _locking
    return func(self, *args, **kwargs)
  File "/home/daniel/.local/lib/python3.8/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 870, in flush
    self._read_packet()
  File "/home/daniel/.local/lib/python3.8/site-packages/pyocd/utility/concurrency.py", line 29, in _locking
    return func(self, *args, **kwargs)
  File "/home/daniel/.local/lib/python3.8/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 1146, in _read_packet
    decoded_data = cmd.decode_data(raw_data)
  File "/home/daniel/.local/lib/python3.8/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 480, in decode_data
    data = self._decode_transfer_data(data)
  File "/home/daniel/.local/lib/python3.8/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 386, in _decode_transfer_data
    self._check_response(data[2])
  File "/home/daniel/.local/lib/python3.8/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 364, in _check_response
    raise DAPAccessIntf.TransferFaultError()
pyocd.probe.pydapaccess.dap_access_api.DAPAccessIntf.TransferFaultError
 
The above exception was the direct cause of the following exception:
 
Traceback (most recent call last):
  File "/home/daniel/.local/lib/python3.8/site-packages/pyocd/coresight/discovery.py", line 162, in _create_1_ap
    ap = AccessPort.create(self.dp, ap_address)
  File "/home/daniel/.local/lib/python3.8/site-packages/pyocd/coresight/ap.py", line 367, in create
[... 133 lines cut ...]
0011292 D GDB get_reg_context: d14 = None -> b'xxxxxxxxxxxxxxxx' [context_facade]
0011292 D GDB get_reg_context: d15 = None -> b'xxxxxxxxxxxxxxxx' [context_facade]
0011292 D GDB get_reg_context: basepri = None -> b'xxxxxxxx' [context_facade]
0011292 D GDB get_reg_context: faultmask = None -> b'xxxxxxxx' [context_facade]
0011293 D GDB get_reg_context: msp_ns = None -> b'xxxxxxxx' [context_facade]
0011293 D GDB get_reg_context: psp_ns = None -> b'xxxxxxxx' [context_facade]
0011293 D GDB get_reg_context: msp_s = None -> b'xxxxxxxx' [context_facade]
0011293 D GDB get_reg_context: psp_s = None -> b'xxxxxxxx' [context_facade]
0011293 D GDB get_reg_context: msplim_ns = None -> b'xxxxxxxx' [context_facade]
0011293 D GDB get_reg_context: psplim_ns = None -> b'xxxxxxxx' [context_facade]
0011293 D GDB get_reg_context: msplim_s = None -> b'xxxxxxxx' [context_facade]
0011293 D GDB get_reg_context: psplim_s = None -> b'xxxxxxxx' [context_facade]
0011293 D GDB packet thread: other side closed connection [packet_io]
0011293 D GDB packet thread stopping [packet_io]
0011393 D gdbserver connection loop exiting; client closed connection [gdbserver]
0011394 D gdbserver exiting connection loop [gdbserver]
0011394 I Client disconnected from port 3333! [gdbserver]
0011394 D gdbserver thread exiting [gdbserver]
0011394 D GDB server cleaning up [gdbserver]
0011709 I Semihost server stopped [server]
0011779 D uninit session <pyocd.core.session.Session object at 0x7f9834c93e50> [session]
0011780 D uninit board <pyocd.board.mbed_board.MbedBoard object at 0x7f9834c4a370> [board]
0011783 D cannot resume: target not halted [cortex_m]
0011791 D closing interface [pyusb_backend]
```

danielRep, 2022-11-01:
So, afaiu the problem can actually come from the gdb. I tried with other version of arm's toolchain. Although I started to try to connect with other versions of the toolchain, I noticed that the pyocd is being able to respond to gdb with the actual register values, since the core was stopped, right?

The output if it helps:
```
0000357 D [Errno 13] Access denied (insufficient permissions) while trying to interrogate a USB device (VID=1ea7 PID=0064). This can probably be remedied with a udev rule. See <https://github.com/pyocd/pyOCD/tree/master/udev> for help. [pyusb_backend]
0000360 D [Errno 13] Access denied (insufficient permissions) while trying to interrogate a USB device (VID=1770 PID=ff00). This can probably be remedied with a udev rule. See <https://github.com/pyocd/pyOCD/tree/master/udev> for help. [pyusb_backend]
0000362 D [Errno 13] Access denied (insufficient permissions) while trying to interrogate a USB device (VID=1ea7 PID=0064). This can probably be remedied with a udev rule. See <https://github.com/pyocd/pyOCD/tree/master/udev> for help. [pyusb_v2_backend]
0000365 D [Errno 13] Access denied (insufficient permissions) while trying to interrogate a USB device (VID=1770 PID=ff00). This can probably be remedied with a udev rule. See <https://github.com/pyocd/pyOCD/tree/master/udev> for help. [pyusb_v2_backend]
0000365 D Project directory: /home/daniel [session]
0000366 D Project directory: /home/daniel [session]
0000374 D CMSIS-DAP v1 probe 500700001d14dad300000000000000000000000097969902: protocol version 1.0.0 [dap_access_cmsis_dap]
0000378 D closing interface [pyusb_backend]
0000380 I Target type is musca_b1 [board]
0000386 D Running task load_svd [sequencer]
0000387 D Running task pre_connect [sequencer]
0000387 D Running task dp_init [sequencer]
0000387 D Running task lock_probe [sequencer]
0000389 D Running task get_probe_capabilities [sequencer]
0000390 D Running task connect [sequencer]
0000407 D Default wire protocol selected; using SWD [dap]
0000409 D Sending deprecated SWJ sequence to select SWD [swj]
0000417 I DP IDR = 0x6ba02477 (v2 rev6) [dap]
0000417 D Running task clear_sticky_err [sequencer]
0000417 D Running task power_up_debug [sequencer]
0000419 D Running task check_version [sequencer]
0000419 D Running task unlock_probe [sequencer]
0000419 D Running task create_discoverer [sequencer]
0000419 D Running task discovery [sequencer]
0000419 D Running task find_aps [sequencer]
0000431 D Running task create_aps [sequencer]
0000431 D Running task create_ap.0 [sequencer]
0000433 I APB-AP#0 IDR = 0x54770002 (APB-AP var0 rev5) [ap]
0000439 D APB-AP#0 default HPROT=0 HNONSEC=0 [ap]
0000441 D APB-AP#0 implemented HPROT=0 HNONSEC=0 [ap]
0000443 D Running task create_ap.1 [sequencer]
0000445 I AHB-AP#1 IDR = 0x84770001 (AHB-AP var0 rev8) [ap]
0000449 D AHB-AP#1 default HPROT=3 HNONSEC=0 [ap]
0000451 D AHB-AP#1 implemented HPROT=f HNONSEC=1 [ap]
0000453 D Running task create_ap.2 [sequencer]
0000455 I AHB-AP#2 IDR = 0x84770001 (AHB-AP var0 rev8) [ap]
0000459 D AHB-AP#2 default HPROT=3 HNONSEC=0 [ap]
0000461 D AHB-AP#2 implemented HPROT=f HNONSEC=1 [ap]
0000463 D Running task find_components [sequencer]
0000463 D Running task init_ap.0 [sequencer]
0000469 I APB-AP#0 Class 0x1 ROM table #0 @ 0xf0000000 (designer=43b:Arm part=7d1) [rom_table]
0000481 I [0]<f0001000:CSTF class=9 designer=43b:Arm part=908 devtype=12 archid=0000 devid=34:0:0> [rom_table]
0000490 I [1]<f0002000:CTI class=9 designer=43b:Arm part=906 devtype=14 archid=0000 devid=40800:0:0> [rom_table]
0000495 I [2]<f0080000:ROM class=1 designer=43b:Arm part=7d4> [rom_table]
0000496 I   APB-AP#0 Class 0x1 ROM table #1 @ 0xf0080000 (designer=43b:Arm part=7d4) [rom_table]
0000507 I   [0]<f0081000:TPIU class=9 designer=43b:Arm part=912 devtype=11 archid=0000 devid=a0:0:0> [rom_table]
0000516 I   [1]<f0082000:CTI class=9 designer=43b:Arm part=906 devtype=14 archid=0000 devid=40800:0:0> [rom_table]
0000521 I   [2]<f0083000:TSGEN class=15 designer=43b:Arm part=101> [rom_table]
0000530 I   [3]<f0084000:CSTF class=9 designer=43b:Arm part=908 devtype=12 archid=0000 devid=33:0:0> [rom_table]
0000531 D Running task init_ap.1 [sequencer]
0000542 I AHB-AP#1 Class 0x1 ROM table #0 @ 0xf0008000 (designer=43b:Arm part=744) [rom_table]
0000553 I [0]<f0009000:GPR class=9 designer=43b:Arm part=9a4 devtype=34 archid=0a34 devid=1:0:0> [rom_table]
0000556 I Enabled power to power domain #0 [rom_table]
0000561 I [1]<e00ff000:ROM class=1 designer=43b:Arm part=4c9 pwrid=0> [rom_table]
0000562 I   AHB-AP#1 Class 0x1 ROM table #1 @ 0xe00ff000 (designer=43b:Arm part=4c9) [rom_table]
0000573 I   [0]<e000e000:SCS M33 class=9 designer=43b:Arm part=d21 devtype=00 archid=2a04 devid=0:0:0> [rom_table]
0000582 I   [1]<e0001000:DWT M33 class=9 designer=43b:Arm part=d21 devtype=00 archid=1a02 devid=0:0:0> [rom_table]
0000591 I   [2]<e0002000:BPU M33 class=9 designer=43b:Arm part=d21 devtype=00 archid=1a03 devid=0:0:0> [rom_table]
0000600 I   [3]<e0000000:ITM M33 class=9 designer=43b:Arm part=d21 devtype=43 archid=1a01 devid=0:0:0> [rom_table]
0000600 D   [4]<fff41002 not present> [rom_table]
0000609 I   [5]<e0041000:ETM M33 class=9 designer=43b:Arm part=d21 devtype=13 archid=4a13 devid=0:0:0> [rom_table]
0000618 I   [6]<e0042000:CTI M33 class=9 designer=43b:Arm part=d21 devtype=14 archid=1a14 devid=40800:0:0> [rom_table]
0000619 D   [7]<fff44002 not present> [rom_table]
0000619 D Running task init_ap.2 [sequencer]
[... cut ...]
0015966 D GDB get_reg_context: msp_s = 0x30027FE0 -> b'e07f0230' [context_facade]
0015966 D GDB get_reg_context: psp_s = 0x00000000 -> b'00000000' [context_facade]
0015966 D GDB get_reg_context: msplim_ns = 0x00000000 -> b'00000000' [context_facade]
0015966 D GDB get_reg_context: psplim_ns = 0x00000000 -> b'00000000' [context_facade]
0015966 D GDB get_reg_context: msplim_s = 0x00000000 -> b'00000000' [context_facade]
0015966 D GDB get_reg_context: psplim_s = 0x00000000 -> b'00000000' [context_facade]
```

danielRep, 2022-11-01:
Any trick to halt the target @flit ? Tried with the below commands, still can do the trick. 
```
pyocd> reset halt
Resetting target with halt
0229928 W Timed out waiting for core to halt after reset (state is RUNNING) [cortex_m]
Failed to halt device on reset (state is RUNNING)
pyocd> halt
Failed to halt device; target state is Running
pyocd> 
```

Hoohaha, 2022-11-02:
@danielRep You should check reason from gdbserver's output. 
And I can see there are permission errors while pyocd accessing devices.
Maybe it is the cause