pyocd/pyOCD #1471: RTT: control block not found

flit, 2022-11-09:
Reports of RTT control block not being found by @letshin and @purpl3F0x.

[image]

[image]

flit, 2022-11-09:
@letshin @purpl3F0x Do either of you have example firmware you could attach to this issue?  I can try building a Zephyr app, but it'd be nice to be able to exactly reproduce your issues. Thanks.

letshin, 2022-11-09:
Hi. I'd preface this by saying that I know very little about working with OCD. 

I've been trying to build the b-parasite firmware:
https://github.com/rbaron/b-parasite/tree/main/code/b-parasite

I've attached the hex file I used here (I assume you meant this by firmware?)
[nrf52840_xxaa.hex.txt](https://github.com/pyocd/pyOCD/files/9975438/nrf52840_xxaa.hex.txt)

letshin, 2022-11-09:
I initially attached the firmware here but actually writing out a response here made me realise that I had compiled for the wrong chip version (RF52840 vs NRF52833). Compiling the correct firmware and loading it resolved this issue (see screenshot below). Apologies for the mix-up on my end - this is my first experience with dealing with OCD stuff.

[image]

flit, 2022-11-10:
@letshin No worries, I'm just glad it ended up working for you! 😄 

(I'll keep this issue open for a little while in case @purpl3F0x replies.)

purpl3F0x, 2022-11-11:
if you could give me few days, im busy this weekend.
also if @letshin wants' to upload its hex for cross reference just in case I'm compiling something wrong :P

letshin, 2022-11-14:
Firmware that didn't work (probably this, it was in my documents folder, didn't bother recompiling).

[nrf52840_xxaa.hex.txt](https://github.com/pyocd/pyOCD/files/9998844/nrf52840_xxaa.hex.txt)

Version that works:
[nrf52833_xxaa.hex.txt](https://github.com/pyocd/pyOCD/files/9998851/nrf52833_xxaa.hex.txt)

purpl3F0x, 2023-01-09:
Hello, sorry for the extra long delay. logging to rtt is working!
You can close this issue

closed 2023-01-10 by flit