pyocd/pyOCD #1472: Using pyinstaller to build executable but cannot load plugins

randychung75, 2022-11-14:
Hello,

I want to create a daplink loader using pyocd, but when I use pyinstaller to pack it, it seems cannot load plugins (pyocd/probe/aggregator.py line 83 **_load_plugin_classes_of_type('pyocd.probe', PROBE_CLASSES, DebugProbe)_**), so executing the .exe file cannot link to any probe, but it is work on IDE mode.

Does anyone knows how to fix this issue?

Thanks.

Regards
Randy

elfmimi, 2022-11-25:
Here are some hints.
https://github.com/pyocd/pyOCD/discussions/1274

randychung75, 2023-02-08:
This hint is worked!
Thanks for your help!

closed 2023-02-08 by randychung75

USTHzhanglu, 2023-03-18:
I also encountered similar problems，and finally i add this in .spec
```
datas=[('./venv/Lib/site-packages/libusb_package/libusb-1.0.dll', '.'),('./venv/Lib/site-packages/cmsis_pack_manager/cmsis_pack_manager/native.so','./cmsis_pack_manager/cmsis_pack_manager/.')],
```