pyocd/pyOCD #1473: RTT: Address and size must be word aligned

Krensi, 2022-11-14:
Hi, I get the following error in the RTT terminal when using `pyocd rtt --target=stm32wb55vgqx --uid stlink:0`

---

`0001552 I RTT control block search range [0x20000000, 0x030000] [rtt_cmd]`
`0003272 I _SEGGER_RTT @ 0x2002ad6c with 3 aUp and 3 aDown [rtt_cmd]`

SOME RTT MESSAGES IN BETWEEN

`0003308 C Error: address and size must be word aligned [__main__]`

---

However, the detected address is word aligned, and the RTT source code is unchanged (so there should not be a change in the control block size). Using Segger RTT viewer or OpenOCD + Cortex Debug works, so RTT works basically.

What am I doing wrong? 

Thank you, 
Christian

Krensi, 2022-11-14:
Using a cmsisdap probe with command `pyocd rtt --target=stm32wb55vgqx --uid cmsisdap:0` everything works just fine.
The RTT block is detected at the same location and there is no error message.

DariusBabrauskas, 2022-11-16:
What show -vv option?

Krensi, 2022-11-29:
```
 *  Executing task: pyocd rtt -vv --target=stm32wb55vgqx --uid stlink:0 

0001143 D STLink probe 003400243137510339383538 firmware version: V3J10M3 [stlink]
0001146 D Fetching mounted devices from disk service registry entry [windows]
0001147 D Fetching usb storage devices from USBSTOR service registry entry [windows]
0001147 D target_id_usb_id -> mount_point mapping: {}  [windows]
0001148 D Project directory: C:\Users\ChristianKrenslehner\Documents\Software\firmware-stm32 [session]
0001374 D flash algo: [code=0x288] [b1=0x290,0x690] [b2=0x690,0xa90] [stack=0x2a90; 0x2000 b] (ram=0x20000000, 0x30000 b) [flash_algo]
0001377 I Target type is stm32wb55vgqx [board]
0001395 D STLink probe 003400243137510339383538 firmware version: V3J10M3 [stlink]
0001396 D actual SWD frequency is 1000 kHz [stlink]
0001397 D Running task load_svd [sequencer]
0001443 D Running task pre_connect [sequencer]
0001457 D Running task dp_init [sequencer]
0001473 D Running task lock_probe [sequencer]
0001489 D Running task get_probe_capabilities [sequencer]
0001505 D Running task connect [sequencer]
0001520 D Default wire protocol selected; using SWD [dap]
0001521 I DP IDR = 0x6ba02477 (v2 rev6) [dap]
0001521 D Running task clear_sticky_err [sequencer]
0001522 D Running task power_up_debug [sequencer]
0001523 D Running task check_version [sequencer]
0001523 D Running task unlock_probe [sequencer]
0001523 D Running task create_discoverer [sequencer]
0001523 D Running task discovery [sequencer]
0001523 D Running task find_aps [sequencer]
0001525 D Running task create_aps [sequencer]
0001525 D Running task create_ap.0 [sequencer]
0001526 D Using accelerated memory access interface for AHB-AP#0 [ap]
0001526 I AHB-AP#0 IDR = 0x24770011 (AHB-AP var1 rev2) [ap]
0001526 D AHB-AP#0 default HPROT=3 HNONSEC=0 [ap]
0001527 D AHB-AP#0 implemented HPROT=3 HNONSEC=0 [ap]
0001528 D Running task create_ap.1 [sequencer]
0001528 D Using accelerated memory access interface for AHB-AP#1 [ap]
0001529 I AHB-AP#1 IDR = 0x84770001 (AHB-AP var0 rev8) [ap]
0001530 D AHB-AP#1 default HPROT=3 HNONSEC=1 [ap]
0001531 D AHB-AP#1 implemented HPROT=f HNONSEC=1 [ap]
0001532 D Running task find_components [sequencer]
0001532 D Running task init_ap.0 [sequencer]
0001535 I AHB-AP#0 Class 0x1 ROM table #0 @ 0xe00ff000 (designer=020:ST part=495) [rom_table]
0001537 I [0]<e000e000:SCS v7-M class=14 designer=43b:Arm part=00c> [rom_table]
0001538 I [1]<e0001000:DWT v7-M class=14 designer=43b:Arm part=002> [rom_table]
0001540 I [2]<e0002000:FPB v7-M class=14 designer=43b:Arm part=003> [rom_table]
0001541 I [3]<e0000000:ITM v7-M class=14 designer=43b:Arm part=001> [rom_table]
0001542 I [4]<e0040000:TPIU M4 class=9 designer=43b:Arm part=9a1 devtype=11 archid=0000 devid=ca1:0:0> [rom_table]
0001544 I [5]<e0041000:ETM M4 class=9 designer=43b:Arm part=925 devtype=13 archid=0000 devid=0:0:0> [rom_table]
0001545 I [6]<e0043000:CTI class=9 designer=43b:Arm part=906 devtype=14 archid=0000 devid=40800:0:0> [rom_table]
0001546 D Running task create_cores [sequencer]
0001546 D Creating SCS component [discovery]
0001546 D selected core #0 [soc_target]
0001548 I CPU core #0 is Cortex-M4 r0p1 [cortex_m]
0001549 I FPU present: FPv4-SP-D16-M [cortex_m]
0001550 D Running task set_default_reset_type [sequencer]
0001550 D Running task create_components [sequencer]
0001550 D Creating DWT component [discovery]
0001551 I 4 hardware watchpoints [dwt]
0001554 D Creating FPB component [discovery]
0001554 I 6 hardware breakpoints, 4 literal comparators [fpb]
0001555 D fpb has been disabled [fpb]
0001558 D Creating ITM component [discovery]
0001560 D Creating TPIU component [discovery]
0001560 D Running task check_for_cores [sequencer]
0001561 D Running task halt_on_connect [sequencer]
0001561 D halting core 0 [cortex_m]
0001562 D Running task post_connect [sequencer]
0001562 D Running task post_connect_hook [sequencer]
0001562 D Running task create_flash [sequencer]
0001562 D Running task notify [sequencer]
0001563 I RTT control block search range [0x20000000, 0x030000] [rtt_cmd]
0001583 I _SEGGER_RTT @ 0x200005a0 with 3 aUp and 3 aDown [rtt_cmd]
0001584 D resuming core 0 [cortex_m]
0001585 D added=[] removed=[] [manager]
0001585 D bps after flush={} [manager]
0001589 D uninit session <pyocd.core.session.Session object at 0x00000141451EDF40> [session]
0001589 D uninit board <pyocd.board.board.Board object at 0x000001414521FA30> [board]
0001591 D cannot resume: target not halted [cortex_m]
0001594 C Error: address and size must be word aligned [__main__]
```

Krensi, 2022-11-29:
```
Executing task: pyocd rtt -vv --target=stm32wb55vgqx --uid cmsisdap:0 

0001126 D Project directory: C:\Users\ChristianKrenslehner\Documents\Software\firmware-stm32 [session]
0001230 D Project directory: C:\Users\ChristianKrenslehner\Documents\Software\firmware-stm32 [session]
0001231 D Project directory: C:\Users\ChristianKrenslehner\Documents\Software\firmware-stm32 [session]
0001233 D CMSIS-DAP v1 probe 000601006224: protocol version 1.2.0 [dap_access_cmsis_dap]
0001235 D CMSIS-DAP probe 000601006224 reported invalid SWO_BUFFER_SIZE (0) [dap_access_cmsis_dap]
0001235 D closing interface [hidapi_backend]
0001470 D flash algo: [code=0x288] [b1=0x290,0x690] [b2=0x690,0xa90] [stack=0x2a90; 0x2000 b] (ram=0x20000000, 0x30000 b) [flash_algo]
0001474 I Target type is stm32wb55vgqx [board]
0001476 D Running task load_svd [sequencer]
0001523 D Running task pre_connect [sequencer]
0001533 D Running task dp_init [sequencer]
0001549 D Running task lock_probe [sequencer]
0001564 D Running task get_probe_capabilities [sequencer]
0001580 D Running task connect [sequencer]
0001599 D Default wire protocol selected; using SWD [dap]
0001600 D Sending deprecated SWJ sequence to select SWD [swj]
0001602 I DP IDR = 0x6ba02477 (v2 rev6) [dap]
0001602 D Running task clear_sticky_err [sequencer]
0001603 D Running task power_up_debug [sequencer]
0001603 D Running task check_version [sequencer]
0001604 D Running task unlock_probe [sequencer]
0001604 D Running task create_discoverer [sequencer]
0001604 D Running task discovery [sequencer]
0001604 D Running task find_aps [sequencer]
0001608 D Running task create_aps [sequencer]
0001608 D Running task create_ap.0 [sequencer]
0001609 I AHB-AP#0 IDR = 0x24770011 (AHB-AP var1 rev2) [ap]
0001610 D AHB-AP#0 default HPROT=3 HNONSEC=0 [ap]
0001612 D AHB-AP#0 implemented HPROT=3 HNONSEC=0 [ap]
0001613 D Running task create_ap.1 [sequencer]
0001615 I AHB-AP#1 IDR = 0x84770001 (AHB-AP var0 rev8) [ap]
0001615 D AHB-AP#1 default HPROT=3 HNONSEC=1 [ap]
0001616 D AHB-AP#1 implemented HPROT=f HNONSEC=1 [ap]
0001618 D Running task find_components [sequencer]
0001618 D Running task init_ap.0 [sequencer]
0001623 I AHB-AP#0 Class 0x1 ROM table #0 @ 0xe00ff000 (designer=020:ST part=495) [rom_table]
0001627 I [0]<e000e000:SCS v7-M class=14 designer=43b:Arm part=00c> [rom_table]
0001629 I [1]<e0001000:DWT v7-M class=14 designer=43b:Arm part=002> [rom_table]
0001631 I [2]<e0002000:FPB v7-M class=14 designer=43b:Arm part=003> [rom_table]
0001633 I [3]<e0000000:ITM v7-M class=14 designer=43b:Arm part=001> [rom_table]
0001636 I [4]<e0040000:TPIU M4 class=9 designer=43b:Arm part=9a1 devtype=11 archid=0000 devid=ca1:0:0> [rom_table]
0001639 I [5]<e0041000:ETM M4 class=9 designer=43b:Arm part=925 devtype=13 archid=0000 devid=0:0:0> [rom_table]
0001642 I [6]<e0043000:CTI class=9 designer=43b:Arm part=906 devtype=14 archid=0000 devid=40800:0:0> [rom_table]
0001643 D Running task create_cores [sequencer]
0001643 D Creating SCS component [discovery]
0001643 D selected core #0 [soc_target]
0001646 I CPU core #0 is Cortex-M4 r0p1 [cortex_m]
0001647 I FPU present: FPv4-SP-D16-M [cortex_m]
0001648 D Running task set_default_reset_type [sequencer]
0001648 D Running task create_components [sequencer]
0001648 D Creating DWT component [discovery]
0001650 I 4 hardware watchpoints [dwt]
0001650 D Creating FPB component [discovery]
0001652 I 6 hardware breakpoints, 4 literal comparators [fpb]
0001652 D fpb has been disabled [fpb]
0001653 D Creating ITM component [discovery]
0001656 D Creating TPIU component [discovery]
0001657 D Running task check_for_cores [sequencer]
0001657 D Running task halt_on_connect [sequencer]
0001658 D halting core 0 [cortex_m]
0001659 D Running task post_connect [sequencer]
0001659 D Running task post_connect_hook [sequencer]
0001660 D Running task create_flash [sequencer]
0001660 D Running task notify [sequencer]
0001661 I RTT control block search range [0x20000000, 0x030000] [rtt_cmd]
0001722 I _SEGGER_RTT @ 0x200005a0 with 3 aUp and 3 aDown [rtt_cmd]
0001723 D resuming core 0 [cortex_m]
0001724 D added=[] removed=[] [manager]
0001724 D bps after flush={} [manager]
```

Krensi, 2022-12-01:
Debug works as expected: Started debugging and paused execution

```
0005226 D Value for symbol xSchedulerRunning = 0x2000260c [provider]
0005227 D Value for symbol xSuspendedTaskList = 0x20002610 [provider]
0005227 D Value for symbol xTasksWaitingTermination = 0x20002624 [provider]
0005228 D Value for symbol vPortEnableVFP = 0x8016940 [provider]
0005228 W FreeRTOS: pxReadyTasksLists size is unexpected, maybe an unsupported version of FreeRTOS. Try using the --elf option. [freertos]
0005228 D Attempting to load rtx5 [gdbserver]
0005228 D Attempting to load threadx [gdbserver]
0005228 D Attempting to load zephyr [gdbserver]
0005229 D Value for symbol _kernel = <none> [provider]
0005248 D GDB received query: [b'Rcmd,72657365742068616c74#72'] [gdbserver]
0005248 D Remote command: reset halt [gdbserver]
0005248 D set reset catch, core 0 [cortex_m]
0005248 D halting core 0 [cortex_m]
0005251 D reset, core 0, type=SW_SYSRESETREQ [cortex_m]
0005371 D Received POST_RESET event [gdbserver]
0005376 D clear reset catch, core 0 [cortex_m]
0005403 D flash op: b'FlashErase' [gdbserver]
0005403 D flash op: b'FlashErase' [gdbserver]
0005403 D flash op: b'FlashWrite' [gdbserver]
0005403 D flash write addr: 0x8000000 [gdbserver]
0005404 D flash op: b'FlashWrite' [gdbserver]
0005404 D flash write addr: 0x8009000 [gdbserver]
0005404 D flash op: b'FlashWrite' [gdbserver]
0005404 D flash write addr: 0x8009140 [gdbserver]
0005405 D flash op: b'FlashWrite' [gdbserver]
0005405 D flash write addr: 0x8009160 [gdbserver]
0005406 D flash op: b'FlashWrite' [gdbserver]
0005407 D flash write addr: 0x8009910 [gdbserver]
0005408 D flash op: b'FlashWrite' [gdbserver]
0005408 D flash write addr: 0x800a0d0 [gdbserver]
0005409 D flash op: b'FlashWrite' [gdbserver]
0005409 D flash write addr: 0x800a890 [gdbserver]
0005410 D flash op: b'FlashWrite' [gdbserver]
0005410 D flash write addr: 0x800b050 [gdbserver]
0005412 D flash op: b'FlashWrite' [gdbserver]
0005412 D flash write addr: 0x800b800 [gdbserver]
0005415 D flash op: b'FlashWrite' [gdbserver]
0005415 D flash write addr: 0x800bfc0 [gdbserver]
0005416 D flash op: b'FlashWrite' [gdbserver]
0005416 D flash write addr: 0x800c780 [gdbserver]
0005418 D flash op: b'FlashWrite' [gdbserver]
0005418 D flash write addr: 0x800cf40 [gdbserver]
0005420 D flash op: b'FlashWrite' [gdbserver]
0005420 D flash write addr: 0x800d700 [gdbserver]
0005421 D flash op: b'FlashWrite' [gdbserver]
0005421 D flash write addr: 0x800deb0 [gdbserver]
0005422 D flash op: b'FlashWrite' [gdbserver]
0005422 D flash write addr: 0x800e670 [gdbserver]
0005424 D flash op: b'FlashWrite' [gdbserver]
0005424 D flash write addr: 0x800ee10 [gdbserver]
0005425 D flash op: b'FlashWrite' [gdbserver]
0005425 D flash write addr: 0x800f5b0 [gdbserver]
0005426 D flash op: b'FlashWrite' [gdbserver]
0005427 D flash write addr: 0x800fd60 [gdbserver]
0005429 D flash op: b'FlashWrite' [gdbserver]
0005429 D flash write addr: 0x8010500 [gdbserver]
0005431 D flash op: b'FlashWrite' [gdbserver]
0005431 D flash write addr: 0x8010c90 [gdbserver]
0005433 D flash op: b'FlashWrite' [gdbserver]
0005433 D flash write addr: 0x8011420 [gdbserver]
0005434 D flash op: b'FlashWrite' [gdbserver]
0005434 D flash write addr: 0x8011bc0 [gdbserver]
0005435 D flash op: b'FlashWrite' [gdbserver]
0005435 D flash write addr: 0x8012330 [gdbserver]
0005437 D flash op: b'FlashWrite' [gdbserver]
0005437 D flash write addr: 0x8012ad0 [gdbserver]
0005438 D flash op: b'FlashWrite' [gdbserver]
[... 889 lines cut ...]
0020225 D GDB get_reg_context: fpscr = 0x00000000 -> b'00000000' [context_facade]
0020225 D GDB get_reg_context: d0 = 0x00000000 -> b'0000000000000000' [context_facade]
0020225 D GDB get_reg_context: d1 = 0x00000000 -> b'0000000000000000' [context_facade]
0020225 D GDB get_reg_context: d2 = 0x00000000 -> b'0000000000000000' [context_facade]
0020225 D GDB get_reg_context: d3 = 0x00000000 -> b'0000000000000000' [context_facade]
0020225 D GDB get_reg_context: d4 = 0x00000000 -> b'0000000000000000' [context_facade]
0020225 D GDB get_reg_context: d5 = 0x00000000 -> b'0000000000000000' [context_facade]
0020225 D GDB get_reg_context: d6 = 0x00000000 -> b'0000000000000000' [context_facade]
0020225 D GDB get_reg_context: d7 = 0x00000000 -> b'0000000000000000' [context_facade]
0020225 D GDB get_reg_context: d8 = 0x00000000 -> b'0000000000000000' [context_facade]
0020225 D GDB get_reg_context: d9 = 0x00000000 -> b'0000000000000000' [context_facade]
0020225 D GDB get_reg_context: d10 = 0x00000000 -> b'0000000000000000' [context_facade]
0020225 D GDB get_reg_context: d11 = 0x00000000 -> b'0000000000000000' [context_facade]
0020225 D GDB get_reg_context: d12 = 0x00000000 -> b'0000000000000000' [context_facade]
0020226 D GDB get_reg_context: d13 = 0x00000000 -> b'0000000000000000' [context_facade]
0020226 D GDB get_reg_context: d14 = 0x00000000 -> b'0000000000000000' [context_facade]
0020226 D GDB get_reg_context: d15 = 0x00000000 -> b'0000000000000000' [context_facade]
0020226 D GDB get_reg_context: basepri = 0x00000000 -> b'00000000' [context_facade]
0020226 D GDB get_reg_context: faultmask = 0x00000000 -> b'00000000' [context_facade]
0020247 D GDB received query: [b'Xfer', b'threads', b'read', b'', b'0,7fb#d0'] [gdbserver]
0020247 D GDB query b'threads': annex: b'', offset: 0, size: 2043 [gdbserver]
0028544 D thread_actions={1: None}; default_action=b'c' [gdbserver]
0028545 D resuming core 0 [cortex_m]

```

DariusBabrauskas, 2022-12-03:
Maybe try update pyocd to v0.34.3 .