pyocd/pyOCD #1486: reset_and_halt() fails to halt

maxgerhardt, 2022-12-16:
Hardware: Nucleo L476RG (STM32L476RG chip)
pyOCD version: latest stable 0.34.3
Python version: 3.9.7
System info: Ubuntu Budgie 22.04, x64

When I try to programmatically flash a firmware, reset and halt, then resume, like the [API example](https://pyocd.io/docs/api_examples.html) shows, the firmware gets flashed but the `.resume()` fails with a "cannot resume: target not halted" error message although `.reset_and_halt()` has been executed beforehand.

```py
from pyocd.core.helpers import ConnectHelper
from pyocd.flash.file_programmer import FileProgrammer
import time
import logging
logging.basicConfig(level=logging.DEBUG)

firmware_path = "/home/max/XYZ/.pio/build/nucleo_l476rg/firmware.bin"
with ConnectHelper.session_with_chosen_probe() as session:
    board = session.board
    target = board.target
    # Load firmware into device.
    print("HAAAALT")
    target.halt()
    print("FLASHINGGGG")
    FileProgrammer(session).program(firmware_path, base_address=0x8000000)
    print("FLASHING DONE")
    # Reset, run.
    time.sleep(0.5)
    print("RESET AND HAAALT")
    target.reset_and_halt()
    print("RESUMEEEE")
    target.resume()
```

Gives output
```
(py) max@PC-LT-23-VM:~/kamstrup-wave2$ python3 minimal.py 
DEBUG:pyocd.probe.pydapaccess.interface.pyusb_backend:[Errno 13] Access denied (insufficient permissions) while trying to interrogate a USB device (VID=0e0f PID=0003). This can probably be remedied with a udev rule. See <https://github.com/pyocd/pyOCD/tree/master/udev> for help.
DEBUG:pyocd.probe.pydapaccess.interface.pyusb_v2_backend:[Errno 13] Access denied (insufficient permissions) while trying to interrogate a USB device (VID=0e0f PID=0003). This can probably be remedied with a udev rule. See <https://github.com/pyocd/pyOCD/tree/master/udev> for help.
DEBUG:pyocd.core.session:Project directory: /home/max/XYZ
DEBUG:pyocd.probe.stlink.stlink:STLink probe 066FFF323532543457095314 firmware version: V2J28M17
DEBUG:pyocd.core.session:Project directory: /home/max/XYZ
WARNING:pyocd.board.board:Generic 'cortex_m' target type is selected by default; is this intentional? You will be able to debug most devices, but not program  flash. To set the target type use the '--target' argument or 'target_override' option. Use 'pyocd list --targets' to see available targets types.
DEBUG:pyocd.coresight.coresight_target:Using default Cortex-M memory map (no memory map supplied)
INFO:pyocd.board.board:Target type is cortex_m
DEBUG:pyocd.probe.stlink.stlink:STLink probe 066FFF323532543457095314 firmware version: V2J28M17
DEBUG:pyocd.utility.sequencer:Running task load_svd
DEBUG:pyocd.utility.sequencer:Running task pre_connect
DEBUG:pyocd.utility.sequencer:Running task dp_init
DEBUG:pyocd.utility.sequencer:Running task lock_probe
DEBUG:pyocd.utility.sequencer:Running task get_probe_capabilities
DEBUG:pyocd.utility.sequencer:Running task connect
DEBUG:pyocd.coresight.dap:Default wire protocol selected; using SWD
INFO:pyocd.coresight.dap:DP IDR = 0x2ba01477 (v1 rev2)
DEBUG:pyocd.utility.sequencer:Running task clear_sticky_err
DEBUG:pyocd.utility.sequencer:Running task power_up_debug
DEBUG:pyocd.utility.sequencer:Running task check_version
DEBUG:pyocd.utility.sequencer:Running task unlock_probe
DEBUG:pyocd.utility.sequencer:Running task create_discoverer
DEBUG:pyocd.utility.sequencer:Running task discovery
DEBUG:pyocd.utility.sequencer:Running task find_aps
DEBUG:pyocd.utility.sequencer:Running task create_aps
DEBUG:pyocd.utility.sequencer:Running task create_ap.0
DEBUG:pyocd.coresight.ap:Using accelerated memory access interface for AHB-AP#0
INFO:pyocd.coresight.ap:AHB-AP#0 IDR = 0x24770011 (AHB-AP var1 rev2)
DEBUG:pyocd.coresight.ap:AHB-AP#0 default HPROT=3 HNONSEC=0
DEBUG:pyocd.coresight.ap:AHB-AP#0 implemented HPROT=3 HNONSEC=0
DEBUG:pyocd.utility.sequencer:Running task find_components
DEBUG:pyocd.utility.sequencer:Running task init_ap.0
INFO:pyocd.coresight.rom_table:AHB-AP#0 Class 0x1 ROM table #0 @ 0xe00ff000 (designer=020:ST part=415)
INFO:pyocd.coresight.rom_table:[0]<e000e000:SCS v7-M class=14 designer=43b:Arm part=00c>
INFO:pyocd.coresight.rom_table:[1]<e0001000:DWT v7-M class=14 designer=43b:Arm part=002>
INFO:pyocd.coresight.rom_table:[2]<e0002000:FPB v7-M class=14 designer=43b:Arm part=003>
INFO:pyocd.coresight.rom_table:[3]<e0000000:ITM v7-M class=14 designer=43b:Arm part=001>
INFO:pyocd.coresight.rom_table:[4]<e0040000:TPIU M4 class=9 designer=43b:Arm part=9a1 devtype=11 archid=0000 devid=ca1:0:0>
INFO:pyocd.coresight.rom_table:[5]<e0041000:ETM M4 class=9 designer=43b:Arm part=925 devtype=13 archid=0000 devid=0:0:0>
DEBUG:pyocd.utility.sequencer:Running task create_cores
DEBUG:pyocd.coresight.discovery:Creating SCS component
DEBUG:pyocd.core.soc_target:selected core #0
INFO:pyocd.coresight.cortex_m:CPU core #0 is Cortex-M4 r0p1
INFO:pyocd.coresight.cortex_m:FPU present: FPv4-SP-D16-M
DEBUG:pyocd.utility.sequencer:Running task create_components
DEBUG:pyocd.coresight.discovery:Creating DWT component
INFO:pyocd.coresight.dwt:4 hardware watchpoints
DEBUG:pyocd.coresight.discovery:Creating FPB component
INFO:pyocd.coresight.fpb:6 hardware breakpoints, 4 literal comparators
DEBUG:pyocd.coresight.fpb:fpb has been disabled
DEBUG:pyocd.coresight.discovery:Creating ITM component
DEBUG:pyocd.coresight.discovery:Creating TPIU component
DEBUG:pyocd.utility.sequencer:Running task check_for_cores
DEBUG:pyocd.utility.sequencer:Running task halt_on_connect
DEBUG:pyocd.coresight.cortex_m:halting core 0
DEBUG:pyocd.utility.sequencer:Running task post_connect
DEBUG:pyocd.utility.sequencer:Running task post_connect_hook
DEBUG:pyocd.utility.sequencer:Running task create_flash
DEBUG:pyocd.utility.sequencer:Running task notify
HAAAALT
DEBUG:pyocd.coresight.cortex_m:halting core 0
FLASHINGGGG
[==================================================] 100%
INFO:pyocd.flash.loader:Erased 0 bytes (0 sectors), programmed 37440 bytes (0 pages), skipped 0 bytes (0 pages) at 59.85 kB/s
FLASHING DONE
RESET AND HAAALT
DEBUG:pyocd.coresight.cortex_m:set reset catch, core 0
DEBUG:pyocd.coresight.cortex_m:halting core 0
DEBUG:pyocd.coresight.cortex_m:reset, core 0, type=SW_SYSRESETREQ
DEBUG:pyocd.coresight.cortex_m:clear reset catch, core 0
RESUMEEEE
DEBUG:pyocd.coresight.cortex_m:resuming core 0
DEBUG:pyocd.debug.breakpoints.manager:added=[] removed=[]
DEBUG:pyocd.debug.breakpoints.manager:bps after flush={}
DEBUG:pyocd.core.session:uninit session <pyocd.core.session.Session object at 0x7f48c4125cd0>
DEBUG:pyocd.board.board:uninit board <pyocd.board.board.Board object at 0x7f48c205da30>
DEBUG:pyocd.coresight.cortex_m:cannot resume: target not halted
```

And yes I have tried using:
* `target.reset_and_halt(Target.ResetType.HW)`
* `target.reset_and_halt(Target.ResetType.SW)`
* `target.reset_and_halt(Target.ResetType.SW_CORE)`
They did not make a difference at all.

maxgerhardt, 2022-12-16:
And: Doing a `pyocd pack install stm32l476rg` and using 
```py
with ConnectHelper.session_with_chosen_probe(target_override="stm32l476rgtx") as session:
```
also makes no difference with the reset problem.

flit, 2022-12-16:
I'm able to reproduce this using your same script (though with an F413, different firmware, and using the `develop` branch).

It's actually not the `.resume()` call that is logging the "cannot resume" message. That happens when pyocd tries to resume when it disconnects, due to the `resume_on_disconnect` option being set (it's enabled by default). Passing `resume_on_disconnect=False` into `.session_with_chosen_probe()` will remove this log message.

You can add a `print('cpu state =', target.get_state().name)` line after the halting reset and resume to verify the target's state for yourself.

Clearly, that log message could be improved… 😅

flit, 2022-12-16:
Fyi, the confusing log message has been reworded in #1487.

flit, 2022-12-19:
@maxgerhardt Is it ok to close this issue?

closed 2022-12-19 by maxgerhardt