pyocd/pyOCD #1490: Performance regression with CMSIS-DAP v2 between release 0.32.3 and 0.33.0

jaydcarlson, 2023-01-03:
I'm experiencing extremely poor performance with my CMSIS-DAP v2 probe, which is an NXP MCU-Link running a current build of DAPLink. I've tested both the latest release and Develop branches and they both have the same issue.

If I force pyOCD to use CMSIS-DAP v1 by adding `cmsis_dap.prefer_v1: True` to `pyocd.yaml`, the performance issue is resolved. 

Reverting to old versions of pyOCD, I noticed the performance regression occurred between release 0.32.3 and 0.33.0. It looks like pyOCD has had CMSIS-DAP v2 support going back to 0.17.0, so I assumed the 0.32.3 build is talking to my probe over the bulk interface — but I'm not seeing any performance improvement in 0.32.3 (which is presumably using the v2 protocol) versus a newer build handicapped with `prefer_v1` set, so I'm not so sure.

To test, I'm starting with an erased ATSAMG55J19 chip and a 512KB test.bin file and I run
```
C:\> pyocd load -t atsamg55j19 -f20m test.bin
```
The performance drops from 45-50 KB/s to less than 3 KB/s.

I investigated a bit with my logic analyzer and found that with pyOCD version 0.32.3, SWD packets are tightly packed together with only a few hundred microseconds between each transfer. It takes 14 ms to perform these 37 transfers:
[image]

But with version 0.33.0 or newer, the same number of transactions takes several seconds. Each transaction has a 15ms delay between it. It takes more than 400 ms to perform 27 transfers:
[image]

I haven't looked at pyOCD's source code, but I've had similar issues in USB-based projects for a variety of reasons, but the main one is if those I/O calls aren't executed with overlapping transfers. 

While I have many other CMSIS-DAP-compatible probes laying around, none of them support CMSIS-DAP v2, so I can't eliminate the DAPLink firmware running on the MCU-Link as a possible suspect, either.

In case you want to examine the Saleae captures directly, here they are:
[PyOCD 0.32.3 vs 0.33.0 Saleae Captures.zip](https://github.com/pyocd/pyOCD/files/10334562/PyOCD.0.32.3.vs.0.33.0.Saleae.Captures.zip)

DariusBabrauskas, 2023-01-03:
How 0.34.3 version?

jaydcarlson, 2023-01-03:
I've tested both the latest release (0.34.3) and the Develop branch and they both have the same issue.

flit, 2023-01-05:
Thanks for the report, and apologies for issues this has caused.

It looks like you are using Windows?

Regarding overlapping transfers: the pyusb package used by pyocd (and most other python based tools using USB) doesn't support asynchronous transfers (required for overlapping transfers) even though the underlying libusb does provide an asynchronous API. This is supported by another Python wrapper around libusb, but would obviously require a pretty big change to pyocd.

Still, pyocd does use an independent receive thread that always keeps a receive request open (unless the request times out due to no IN packets, in which case it's immediately re-issued). While this isn't really asynchronous in the same way, it does, or at least is intended to, achieve much the same result.

In addition, pyocd uses a feature of the CMSIS-DAP protocol where there can be multiple outstanding requests, although they are always issued and responses received in requested order (otherwise you'd have a mess). The actual utility of is limited based on the operations being performed; often the result of one is required before the next can be issued. But it does help with things like flash programming. However(!), at one point in the past (a few years ago, can't remember the version) DAPLink lost its ability to support multiple outstanding CMSIS-DAP requests in an attempt to achieve much greater stability by serialising everything through its main thread (that worked… it's very stable now, but not as fast as it could be).

flit, 2023-01-07:
Naturally (😞), I'm not seeing any difference in performance between these versions on either Windows 10 or macOS (tested just for a sanity check) yet.

Using the `test/speed_test.py` functional test with a micro:bit v2 (with on-board DAPLink, but only full speed USB), the r/w performance for RAM is ~50 kB/s for both versions. Same results on macOS 12.6.2 and a Windows 10 VM.

Also tested using a Keil ULINKplus with an STM32H750, programming flash. In all cases it gets about 17 kB/s. This really isn't a good test, though, since flash performance seems to be the limiting factor here. (Which is why I also tested with RAM on the micro:bit.)

Clearly something's going on since the SWD traces show delayed packets. I'll review commits to see if anything appears suspicious. (One notable USB related change between 0.32.3 and 0.33.0 was a switch to using `libusb-package` to automatically install the libusb shared library. It's unlikely that this is the root cause, though.) I'll also try using an MCU-Link running DAPLink, although I don't have an ATSAMG55J19 to fully reproduce your setup.

What OS version and Python version are you using? Thanks!

Guozhanxin, 2023-06-13:
I think I have the same problem😣

Guozhanxin, 2023-06-14:
@flit Hello, I tested the latest develop version and it seems that there is no improvement. Is there any progress on this issue? Looking forward to your reply！

flit, 2023-06-14:
@Guozhanxin Well, so far I have been unable to reproduce this issue. Obviously, something is going wrong, though!

Just for reference, what debug probe(s) are you using?

Can you check pyocd v0.32.3 and v0.33.0 to verify that v0.33.0 was the version to introduce the issue?

Am I correct in understanding that, in your case, the problem only affects writes to flash memory? If this is true, it might not be a USB-related issue at all, although in that case it doesn't make sense that CMSIS-DAP v1 would not also be affected.

Thanks.

Guozhanxin, 2023-06-15:
@flit this is 0.32.2 version, Because I encountered problems installing libusb.lib, I merged this modification(https://github.com/pyocd/pyOCD/pull/1240) on the basis of 0.32.2. And all the following tests are done on this basis.

**It can be found that the download speed of this version v2 is much faster than that of v1**

 ### connect dap v1, speed is 4.64 kB/s
```
PS D:\RT-ThreadStudio\repo\Extract\Debugger_Support_Packages\RealThread\PyOCD\0.2.0> pyocd.exe flash --target=R7FA6M3AH --erase=auto --frequency=1000000 D:\RT-ThreadStudio\workspace\ttt\Debug\rtthread.hex
0000950:INFO:load_cmd:Loading D:\RT-ThreadStudio\workspace\ttt\Debug\rtthread.hex
0000988:WARNING:file_programmer:Failed to add data chunk: no memory region defined for address 0x0100a150
[==================================================] 100%
0021664:INFO:loader:Erased chip, programmed 97664 bytes (763 pages), skipped 640 bytes (5 pages) at 4.64 kB/s
```
 ### connect dap v2, speed is 9.07 kB/s

```
PS D:\RT-ThreadStudio\repo\Extract\Debugger_Support_Packages\RealThread\PyOCD\0.2.0> pyocd.exe flash --target=R7FA6M3AH --erase=auto --frequency=1000000 D:\RT-ThreadStudio\workspace\ttt\Debug\rtthread.hex
0000891:INFO:load_cmd:Loading D:\RT-ThreadStudio\workspace\ttt\Debug\rtthread.hex
0000927:WARNING:file_programmer:Failed to add data chunk: no memory region defined for address 0x0100a150
[==================================================] 100%
0011508:INFO:loader:Erased chip, programmed 97664 bytes (763 pages), skipped 640 bytes (5 pages) at 9.07 kB/s
```
## When flash to ram
### connect dap v1, speed is 52.08 kB/s
```
D:\Progrem\Python310\Scripts>pyocd.exe flash --target=R7FA6M3AH --erase=auto -a=0x20000000 --frequency=1000000 D:\RT-ThreadStudio\workspace\project\Debug\rtthread.bin 
0000978:INFO:load_cmd:Loading D:\RT-ThreadStudio\workspace\project\Debug\rtthread.bin at 0x20000000
[---|---|---|---|---|---|---|---|---|----]
[========================================]
0008651:INFO:loader:Erased 0 bytes (0 sectors), programmed 408652 bytes (0 pages), skipped 0 bytes (0 pages) at 52.08 kB/s
```
 ### connect dap v2, speed is 62.02 kB/s
```
D:\Progrem\Python310\Scripts>pyocd.exe flash --target=R7FA6M3AH --erase=auto -a=0x20000000 --frequency=1000000 D:\RT-ThreadStudio\workspace\project\Debug\rtthread.bin 
0000889:INFO:load_cmd:Loading D:\RT-ThreadStudio\workspace\project\Debug\rtthread.bin at 0x20000000
[---|---|---|---|---|---|---|---|---|----]
[========================================]
0007336:INFO:loader:Erased 0 bytes (0 sectors), programmed 408652 bytes (0 pages), skipped 0 bytes (0 pages) at 62.02 kB/s
```

Guozhanxin, 2023-06-15:
@flit this is 0.32.3 version, I merged this modification(https://github.com/pyocd/pyOCD/pull/1240) on the basis too. 

**Good news, the download speed of daplink-v2 under this version has increased to 33.17KB/s**
```
PS D:\RT-ThreadStudio\repo\Extract\Debugger_Support_Packages\RealThread\PyOCD\0.2.0> pyocd.exe flash --target=R7FA6M3AH --erase=auto --frequency=1000000 D:\RT-ThreadStudio\workspace\ttt\Debug\rtthread.hex
0000882:INFO:load_cmd:Loading D:\RT-ThreadStudio\workspace\ttt\Debug\rtthread.hex
0000918:WARNING:file_programmer:Failed to add data chunk: no memory region defined for address 0x0100a150
[==================================================] 100%
0003814:INFO:loader:Erased 0 bytes (0 sectors), programmed 0 bytes (0 pages), skipped 98304 bytes (768 pages) at 33.17 kB/s
```
Similarly, the download speed of daplink-v1 under this version has also improved, returning to the speed under the latest development version
```
PS D:\RT-ThreadStudio\repo\Extract\Debugger_Support_Packages\RealThread\PyOCD\0.2.0> pyocd.exe flash --target=R7FA6M3AH --erase=auto --frequency=1000000 D:\RT-ThreadStudio\workspace\ttt\Debug\rtthread.hex
0000977:INFO:load_cmd:Loading D:\RT-ThreadStudio\workspace\ttt\Debug\rtthread.hex
0001013:WARNING:file_programmer:Failed to add data chunk: no memory region defined for address 0x0100a150
[==================================================] 100%
0007162:INFO:loader:Erased 0 bytes (0 sectors), programmed 0 bytes (0 pages), skipped 98304 bytes (768 pages) at 15.62 kB/s
```

Guozhanxin, 2023-06-15:
@flit This is pyocd 0.33.0 version. 
Unfortunately, with this version of daplink-v2, the download speed drops to 2.46 kB/s. It can be seen that the bug that caused the performance degradation of daplink-v2 should be between the two versions of pyocd v0.32.3 and v0.33.0

```
PS D:\RT-ThreadStudio\repo\Extract\Debugger_Support_Packages\RealThread\PyOCD\0.2.0> pyocd.exe flash --target=R7FA6M3AH --erase=auto --frequency=1000000 D:\RT-ThreadStudio\workspace\ttt\Debug\rtthread.hex
0001829 I Loading D:\RT-ThreadStudio\workspace\ttt\Debug\rtthread.hex [load_cmd]
0001870 W Failed to add data chunk: no memory region defined for address 0x0100a150 [file_programmer]
[==================================================] 100%
0029887 I Erased 0 bytes (0 sectors), programmed 0 bytes (0 pages), skipped 70528 bytes (551 pages) at 2.46 kB/s
```
Moreover, the download speed of daplink-v1 under this version has not changed.

```
PS D:\RT-ThreadStudio\repo\Extract\Debugger_Support_Packages\RealThread\PyOCD\0.2.0> pyocd.exe flash --target=R7FA6M3AH --erase=auto --frequency=1000000 D:\RT-ThreadStudio\workspace\ttt\Debug\rtthread.hex
0000931 I Loading D:\RT-ThreadStudio\workspace\ttt\Debug\rtthread.hex [load_cmd]
0000972 W Failed to add data chunk: no memory region defined for address 0x0100a150 [file_programmer]
[==================================================] 100%
0005747 I Erased 0 bytes (0 sectors), programmed 0 bytes (0 pages), skipped 70528 bytes (551 pages) at 14.42 kB/s
```

flit, 2023-06-15:
@Guozhanxin Thanks for testing! That pretty much confirms that it's something related to libusb usage or the CMSIS-DAPv2 pyusb backend. CMSIS-DAPv1 isn't affected because it uses hidapi for HID class on Windows.

What's really interesting is that you used libusb-package for v0.32.3. That actually really helps, because it shows that the problem is not caused by an issue with the version of libusb in libusb-package.

flit, 2023-06-18:
@Guozhanxin Could you please test with this patch applied to commit 376d3f4 (current HEAD of `main`)? The patch reverts some CMSIS-DAP related changes that are in commit 269610f from v0.33.0.

[pyOCD-flit-16-58-28.patch](https://github.com/pyocd/pyOCD/files/11782687/pyOCD-flit-16-58-28.patch)

I really hope this shows something! Otherwise, there aren't any other changes that could have an obvious impact. And you've already pretty much proven that the change to libusb-package didn't introduce the issue.

If that doesn't change anything, would it be possible for you to perform a git bisect on the `develop` branch between commits ce05b14 (when libusb-package was added) and bade0ae (last v0.33.0 commit on `develop`)?

I tried testing again but still cannot reproduce the slowdown. ☹️ Tested under Windows 10 with both DAPLink running on an NXP MCU-Link Pro and an Arm ULINKplus, both CMSIS-DAPv2. Got about 14-16 kB/s programming an nRF9160.

Thanks for your help! 🙏🏽

Guozhanxin, 2023-06-21:
@flit I'm glad to hear your feedback, I will test it soon. If there is no time to test today, it may be after the Dragon Boat Festival!

Guozhanxin, 2023-06-21:
@flit I have tested by this.
Good news, the download speed of daplink-v2 under this version has increased to 39.04 kB/s
```
PS E:\workspace\HMI-Board\pyocd-test> pyocd flash --target=R7FA6M3AH --erase=auto --frequency=1000000 E:\workspace\HMI-Board\ra6m3_project\project\rtthread.hex
0000929 I Loading E:\workspace\HMI-Board\ra6m3_project\project\rtthread.hex [load_cmd]
[==================================================] 100%
0017239 I Erased 0 bytes (0 sectors), programmed 0 bytes (0 pages), skipped 638864 bytes (4992 pages) at 39.04 kB/s [loader]
```
And, the download speed of daplink-v1 under this version has not changed.
```
PS E:\workspace\HMI-Board\pyocd-test> pyocd flash --target=R7FA6M3AH --erase=auto --frequency=1000000 E:\workspace\HMI-Board\ra6m3_project\project\rtthread.hex
0000958 I Loading E:\workspace\HMI-Board\ra6m3_project\project\rtthread.hex [load_cmd]
[==================================================] 100%
0038204 I Erased 0 bytes (0 sectors), programmed 0 bytes (0 pages), skipped 638864 bytes (4992 pages) at 16.90 kB/s [loader]
```
And, when I change pyocd frequency to 10000000,  the download speed of daplink-v2 under this version has increased to 80.02 kB/s
```
PS E:\workspace\HMI-Board\pyocd-test> pyocd flash --target=R7FA6M3AH --erase=auto --frequency=10000000 E:\workspace\HMI-Board\ra6m3_project\project\rtthread.hex
0000909 I Loading E:\workspace\HMI-Board\ra6m3_project\project\rtthread.hex [load_cmd]
[==================================================] 100%
0009042 I Erased 0 bytes (0 sectors), programmed 0 bytes (0 pages), skipped 638864 bytes (4992 pages) at 80.02 kB/s [loader]
```

flit, 2023-06-21:
Thanks for testing. Glad to hear it had an effect! 😄 

If you don't mind, I'll send you some more patches to try to narrow down the cause and a workable solution. It might be this section: https://github.com/pyocd/pyOCD/blob/376d3f49db411a152d1ab5998967862707ae46fd/pyocd/probe/pydapaccess/interface/pyusb_v2_backend.py#L226-L231

That was added to ensure there aren't any race conditions due to thread scheduling. It's unlikely to occur, but really should be present in some form. My guess is that the 100 µs delay is turned into a 1 ms or larger delay on some versions of Windows, though I don't know why I don't see it on my Windows 10 VM.

If you want to just remove this section and try it out, that would be nice to confirm that this code is the problem. I'll also try to rewrite it using something like `queue.SimpleQueue` that supports a timeout on `.get()`.

flit, 2023-06-21:
Ah, quick note: since you are programming the same firmware image that is already in flash, pyocd is only reading flash to verify that it doesn't need to reprogram flash. You can force it to always erase and program by passing `-Osmart-flash=no`.

flit, 2023-06-25:
@Guozhanxin @jaydcarlson 

If you have a chance, could you please try out the attached patch against commit 376d3f4? Thanks!

[pyOCD-bcc744a-cmsis-dap_v2_queue.patch](https://github.com/pyocd/pyOCD/files/11860968/pyOCD-bcc744a-cmsis-dap_v2_queue.patch)

Guozhanxin, 2023-06-28:
@flit Good news, it looks like the patch is working.
The download speed of daplink-v2 under this version is 33.43 kB/s
```
PS E:\workspace\HMI-Board\pyocd-test> pyocd flash --target=R7FA6M3AH --erase=auto --frequency=1000000 -Osmart-flash=no E:\workspace\HMI-Board\ra6m3_project\project\rtthread.hex
0000380 W unknown session option 'smart-flash' [cmdline]
0000947 I Loading E:\workspace\HMI-Board\ra6m3_project\project\rtthread.hex [load_cmd]
[==================================================] 100%
0019945 I Erased 0 bytes (0 sectors), programmed 0 bytes (0 pages), skipped 638864 bytes (4992 pages) at 33.43 kB/s
```
And, the download speed of daplink-v1 under this version has not changed.

## Other
I noticed that the smart-flash option did not take effect, so I tested a few more downloads after erasing the flash.
### erase && freq=10M && dap-v2
The download speed of daplink-v2 under this version is 24.86 kB/s
```
PS E:\workspace\HMI-Board\pyocd-test> pyocd erase --chip --target=R7FA6M3AH
0000926 I Erasing chip... [eraser]
0009187 I Chip erase complete [eraser]
PS E:\workspace\HMI-Board\pyocd-test> pyocd flash --target=R7FA6M3AH --erase=auto --frequency=10000000 -Osmart-flash=no E:\workspace\HMI-Board\ra6m3_project\project\rtthread.hex
0000364 W unknown session option 'smart-flash' [cmdline]
0000892 I Loading E:\workspace\HMI-Board\ra6m3_project\project\rtthread.hex [load_cmd]
[==================================================] 100%
0026316 I Erased chip, programmed 638208 bytes (4986 pages), skipped 656 bytes (6 pages) at 24.86 kB/s
```
### erase && freq=1M && dap-v2
The download speed of daplink-v2 under this version is 11.27 kB/s
```
PS E:\workspace\HMI-Board\pyocd-test> pyocd erase --chip --target=R7FA6M3AH
0000933 I Erasing chip... [eraser]
0007172 I Chip erase complete [eraser]
PS E:\workspace\HMI-Board\pyocd-test> pyocd flash --target=R7FA6M3AH --erase=auto --frequency=1000000 -Osmart-flash=no E:\workspace\HMI-Board\ra6m3_project\project\rtthread.hex
0000368 W unknown session option 'smart-flash' [cmdline]
0000910 I Loading E:\workspace\HMI-Board\ra6m3_project\project\rtthread.hex [load_cmd]
[==================================================] 100%
0056583 I Erased chip, programmed 638208 bytes (4986 pages), skipped 656 bytes (6 pages) at 11.27 kB/s
```
### erase && freq=10M && dap-v1
The download speed of daplink-v1 under this version is 4.91 kB/s
```
PS E:\workspace\HMI-Board\pyocd-test> pyocd erase --chip --target=R7FA6M3AH
0000976 I Erasing chip... [eraser]
0007513 I Chip erase complete [eraser]
PS E:\workspace\HMI-Board\pyocd-test> pyocd flash --target=R7FA6M3AH --erase=auto --frequency=10000000 -Osmart-flash=no E:\workspace\HMI-Board\ra6m3_project\project\rtthread.hex
0000372 W unknown session option 'smart-flash' [cmdline]
0000973 I Loading E:\workspace\HMI-Board\ra6m3_project\project\rtthread.hex [load_cmd]
[==================================================] 100%
0128349 I Erased chip, programmed 638208 bytes (4986 pages), skipped 656 bytes (6 pages) at 4.91 kB/s
```
### erase && freq=1M && dap-v1
The download speed of daplink-v1 under this version is 4.77 kB/s
```
PS E:\workspace\HMI-Board\pyocd-test> pyocd erase --chip --target=R7FA6M3AH
0000979 I Erasing chip... [eraser]
0007053 I Chip erase complete [eraser]
PS E:\workspace\HMI-Board\pyocd-test> pyocd flash --target=R7FA6M3AH --erase=auto --frequency=1000000 -Osmart-flash=no E:\workspace\HMI-Board\ra6m3_project\project\rtthread.hex
0000364 W unknown session option 'smart-flash' [cmdline]
0000954 I Loading E:\workspace\HMI-Board\ra6m3_project\project\rtthread.hex [load_cmd]
[==================================================] 100%
0132212 I Erased chip, programmed 638208 bytes (4986 pages), skipped 656 bytes (6 pages) at 4.77 kB/s [loader]
```

Guozhanxin, 2023-06-28:
@flit Overall, it seems that this patch has taken effect, but the speed of usb communication seems to be slightly lower than the last patch.  39.04 kB/s VS 33.43 kB/s

closed 2023-07-30 by flit