pyocd/pyOCD #1491: pyOCD pack find issue

carlVirtualincision, 2023-01-03:
"pyocd pack find stm32l" does not seem to work on across a few versions of pyOCD as well across linux and windows. In particular this did not work for when I was trying to do it for the stm32l011, but it does not seem to find any packs with the name stm32l. 
[image]

mathias-arm, 2023-01-03:
Have you tried to run `pyocd pack update` first?

I just tried the following:
```
$ pip install -U pyocd cmsis-pack-manager
[...]
$ pyocd pack update
0000411 I Updating pack index... [pack_cmd]
20:44:52 [INFO] Downloaded http://www.keil.com/pack/index.pidx
20:44:52 [INFO] Found 832 Pdsc entries
[...]
$ pyocd pack find stm32l011
  Part            Vendor               Pack                 Version   Installed
---------------------------------------------------------------------------------
  STM32L011D3Px   STMicroelectronics   Keil.STM32L0xx_DFP   2.2.0     False
  STM32L011D4Px   STMicroelectronics   Keil.STM32L0xx_DFP   2.2.0     False
  STM32L011E3Yx   STMicroelectronics   Keil.STM32L0xx_DFP   2.2.0     False
  STM32L011E4Yx   STMicroelectronics   Keil.STM32L0xx_DFP   2.2.0     False
  STM32L011F3Px   STMicroelectronics   Keil.STM32L0xx_DFP   2.2.0     False
  STM32L011F3Ux   STMicroelectronics   Keil.STM32L0xx_DFP   2.2.0     False
  STM32L011F4Px   STMicroelectronics   Keil.STM32L0xx_DFP   2.2.0     False
  STM32L011F4Ux   STMicroelectronics   Keil.STM32L0xx_DFP   2.2.0     False
  STM32L011G3Ux   STMicroelectronics   Keil.STM32L0xx_DFP   2.2.0     False
  STM32L011G4Ux   STMicroelectronics   Keil.STM32L0xx_DFP   2.2.0     False
  STM32L011K3Tx   STMicroelectronics   Keil.STM32L0xx_DFP   2.2.0     False
  STM32L011K3Ux   STMicroelectronics   Keil.STM32L0xx_DFP   2.2.0     False
  STM32L011K4Tx   STMicroelectronics   Keil.STM32L0xx_DFP   2.2.0     False
  STM32L011K4Ux   STMicroelectronics   Keil.STM32L0xx_DFP   2.2.0     False
```

tonika31, 2023-01-04:
I have the same issue when I'm trying to install a pack 
$ pyocd pack install stm32l475xg
0001747 W No matching devices. Please make sure the pack index is up to date. [pack_cmd]
$ pyocd pack update 
11:53:53 [INFO] Downloaded http://www.keil.com/pack/index.pidx
11:53:53 [INFO] Found 832 Pdsc entries
11:53:53 [WARN] No Core found!32)
11:53:53 [WARN] No Core found!
11:53:53 [WARN] No Core found!
11:53:53 [WARN] No Core found!
11:53:53 [WARN] No Core found!
11:53:53 [WARN] No Core found!
11:53:53 [WARN] No Core found!
11:53:53 [WARN] No Core found!
11:53:53 [WARN] No Core found!
11:53:53 [WARN] No Core found!
11:53:53 [WARN] No Core found!

carlVirtualincision, 2023-01-04:
I did try that pyocd pack update, but I get a similar issue to tonika31. It does not seem to update properly as the pyocd pack find stm32l command still does not work after that.

mathias-arm, 2023-01-04:
@carlVirtualincision / @tonika31: what versions of `pyocd` and `cmsis-pack-manager` packages are you using?

tonika31, 2023-01-04:
> @carlVirtualincision / @tonika31: what versions of `pyocd` and `cmsis-pack-manager` packages are you using?

```
 $ pyocd --version
0.34.3
```
which command i should type to see the cmsis-pack-manager

mathias-arm, 2023-01-04:
```% pip show cmsis-pack-manager
Name: cmsis-pack-manager
Version: 0.5.1
Summary: Python manager for CMSIS-Pack index and cache with fast Rust backend
[...]```

tonika31, 2023-01-04:
> 

`pip show cmsis-pack-manager
Name: cmsis-pack-manager
Version: 0.5.1
Summary: Python manager for CMSIS-Pack index and cache with fast Rust backend
`

tonika31, 2023-01-04:
> ```
> % pip show cmsis-pack-manager
> Name: cmsis-pack-manager
> Version: 0.5.1
> Summary: Python manager for CMSIS-Pack index and cache with fast Rust backend
> [...]
> ```

> 

```
$ pip show cmsis-pack-manager
Name: cmsis-pack-manager
Version: 0.5.1
Summary: Python manager for CMSIS-Pack index and cache with fast Rust backend
```

carlVirtualincision, 2023-01-06:
[image]

tonika31, 2023-01-06:
> [image]

Please try again 
`$ pyocd pack find stm32l`

I test it today and its working on setup.( without any changes)

elliott-col, 2023-01-27:
I am encountering the same issue while setting up a new PC to debug an STM32 MCU.
```
$ pyocd --version
0.34.3
```
```
$ pip show cmsis-pack-manager
Name: cmsis-pack-manager
Version: 0.5.1
Summary: Python manager for CMSIS-Pack index and cache with fast Rust backend
Home-page:
Author: Jimmy Brisson <theotherjimmy@gmail.com>, Chris Reed <flit@me.com>, Mathias Brossard <mathias.brossard@arm.com>
Author-email: Jimmy Brisson <theotherjimmy@gmail.com>, Chris Reed <flit@me.com>, Mathias Brossard <mathias.brossard@arm.com>
License: Apache-2.0
Location: /home/dev/.local/lib/python3.10/site-packages
Requires: appdirs, cffi, pyyaml
Required-by: pyocd
```
```
$ pyocd pack install stm32u*
01:05:01 [INFO] Downloaded http://www.keil.com/pack/index.pidx
01:05:01 [INFO] Found 833 Pdsc entries
01:05:25 [WARN] No Core found!33)
01:05:25 [WARN] No Core found!
01:05:25 [WARN] No Core found!
01:05:25 [WARN] No Core found!
... (more of the same)
01:05:25 [WARN] XML error: Expecting </link> found </head>
.... (more of the same)
01:05:26 [WARN] cannot parse integer from empty string
... (more of the same)
01:05:26 [WARN] Support for two parent processors not implemented!
... (more of the same)
01:05:26 [WARN] XML error: Expecting </P> found </BODY>
01:05:26 [WARN] XML error: Expecting </P> found </BODY>
01:05:26 [WARN] XML error: Expecting </P> found </BODY>
01:05:26 [WARN] XML error: Expecting </P> found </BODY>
01:05:26 [WARN] XML error: Expecting </P> found </BODY>
01:05:26 [WARN] XML error: Expecting </P> found </BODY>
01:05:26 [WARN] XML error: Expecting </P> found </BODY>
01:05:26 [WARN] XML error: Expecting </P> found </BODY>
01:05:26 [WARN] XML error: Expecting </P> found </BODY>
01:05:26 [WARN] No Core found!
01:05:26 [WARN] XML error: Expecting </P> found </BODY>
0089853 W No matching devices. Please make sure the pack index is up to date. [pack_cmd]
```

mbrossard, 2023-01-28:
A new version of `cmsis-pack-manager` (0.5.2) is being published that addresses an issue where PDSC file downloads might fail because too many downloads are started concurrently. This might explain the different observations in this issue.

larsfagerbakke, 2023-04-21:
I have pyocd .0.34.3 and cmsis-pack-manager 0.5.2 and I cant update. If I try I get the same XML error and more as @elliott-col  posted. 

Any fix?

flit, 2023-04-21:
@larsfagerbakke We haven't been able to reliably reproduce any issues since cmsis-pack-manager v0.5.2. I have seen errors downloading _only_ NXP packs on very rare occasions, but they immediately went away next time I tried. @mathias-arm hasn't seen the errors at all, afaik.

Could you please provide the log output you see? The "XML error: Expecting \</P\>…" errors were caused by mishandling of errors in earlier versions that resulting in downloading of HTML responses to 404 errors (returned by the server because of too many simultaneous requests). CPM now correctly handles HTTP error responses (not taking into account possible bugs), so it won't try to parse HTML pages and therefore should not report similar XML parsing errors.

The errors from 404 responses now look like this (clearly not the most user-friendly):
```
20:12:31 [ERROR] download of Url { scheme: "http", cannot_be_a_base: false, username: "", password: None, host: Some(Domain("ziweixing.workspacep5.webfoss.com")), port: None, path: "/uploads/XMC.CX32L003_DFP.pdsc", query: None, fragment: None } failed: Response code in invalid range: 404
```
(This is from a pack hosted on a server that apparently isn't running anymore, or the vendor provided an invalid path.)

So, if you are still seeing XML parsing errors, there's something new and different happening, or a variant of a previous problem.

Note that you can always download CMSIS-Packs manually from https://www.keil.com/dd2/pack/ and use the `--pack` argument with pyocd (or the `pack` session option). See [the docs](https://pyocd.io/docs/target_support.html#manual-pack-usage) for more.

@mathias-arm I think the error @larsfagerbakke reports is the one I saw but couldn't reproduce.

AirChandler, 2025-02-28:
Maybe digging up an old issue, but this has started happening again for me with 

[image]

[image]

[image]

pyocd = v0.36.0
cmsis-pack-manager = v0.5.3
python venv = v3.12.9

mbrossard, 2025-03-02:
I tried to reproduce (on MacOS) on a clean virtualenv:
```
$ virtualenv venv
$ source venv/bin/activate
$ pip install pyocd
$ pip list
Package             Version
------------------- ------------
[...]
cmsis-pack-manager  0.5.3
[...]
pyocd               0.36.0
[...]
$ pyocd pack clean
0000126 I Removing all pack data... [pack_cmd]
$ pyocd pack find -v stm32g0
0000130 I No pack index present, downloading now... [pack_cmd]
  Part             Vendor               Pack                 Version   Installed
----------------------------------------------------------------------------------
  STM32G030C6Tx    STMicroelectronics   Keil.STM32G0xx_DFP   1.5.0     False
  STM32G030C8Tx    STMicroelectronics   Keil.STM32G0xx_DFP   1.5.0     False
  STM32G030F6Px    STMicroelectronics   Keil.STM32G0xx_DFP   1.5.0     False
[...]
```

Can you check if a file matching`Keil.STM32G0xx_DFP.*.pdsc` exists and is not corrupted:
- Linux: `$HOME/.local/share/cmsis-pack-manager`
- Mac: `$HOME/Library/Application Support/cmsis-pack-manager` 
- Windows: `'C:\\Users\\$USER\\AppData\\cmsis-pack-manager'`

Is it working with any other searches?

AirChandler, 2025-03-02:
[image]

These are the only STM pdsc's present, the targets included in those are found by pyocd. For some reason its always failing to index packs for other series.

mathias-arm, 2025-03-05:
Can you try `pyocd pack update` a couple of times to see if you get additional files added? It looks like the same issue.
It looks like the limit set in https://github.com/pyocd/cmsis-pack-manager/pull/205 (6 concurrent connections) might still be too high.

timonsku, 2025-08-03:
It looks like I may be running into that issue, I can't do a full update of the pack index and everytime I restart the process it gets stuck sooner than before. I usually can't get beyond 200-300 parts before it gets stuck forever.
On version 0.38.0

SpaceCheetah, 2026-03-11:
I tried to get a bit more info to get the full errors, and they're stuff like this:
```
error sending request for url (https://www.keil.com/pack/ARM.CMSIS-RTX.pdsc)

Caused by:
    0: client error (Connect)
    1: invalid peer certificate: UnknownIssuer
```

Now I'm even more confused. Curl works just fine, and if it's actually a certificate error, I don't see any reason for it to sometimes work on the same computer.