pyocd/pyOCD #1493: ATSAMD21G18A TransferError: No ACK received

Oozlum, 2023-01-17:
**Problem**
I'm trying to use pyOCD with an ATSAMD21G18A connected to an nEDBG CMSIS-DAP, but any command I try results in the following error:

```pyocd.probe.pydapaccess.dap_access_api.DAPAccessIntf.TransferError: No ACK received```

**Possibly useful info**

1. When I start pyocd commander I get ```Connected to ATSAMD21G18A [Lockup]: MCHP3313031800000599```.  I'm not sure what [Lockup] means in this context.
2. I note that the ATSAMD21G17D is provided by Microchip.SAMD21_DFP.3.6.144 but the ATSAMD21G18A is provided by Keil.SAMD21_DFP.1.3.1, which is deprecated.  Perhaps this is a pack issue, but I don't know how to change it.

**Background**
The Curiosity Nano board has a built-in nEDBG adaptor and an ATSAMD21G17D target MCU on-board.  I was able to connect to this chip fine using pyOCD, however I now want to use the nEDBG adaptor to program an ATSAMD21G18A chip on a different board.  I cut the SWD traces on the development board, as described in the user guide, and connected it to my target processor, but pyOCD refuses to play.

Atmel (Microchip) Studio running on the same machine with no change to the hardware setup works perfectly, so I'm confident it isn't a hardware issue.

I'm using a MacBook Pro (Ventura), running pyOCD natively.  I run Atmel Studio using Parallels.

Oozlum, 2023-01-18:
Ok, I manually hacked the files to remove the Keil PDSC from the list, pyOCD found and installed the Microchip PDSC and everything works now.

There seems to be two related issues:

1. the cmsis-pack-manager cache is keyed by part number (eg. ATSAMD21G18A), so it can't handle more than one PDSC option for a given part.
2. there is no way to exclude PDSC entries marked as 'deprecated' (which would imply a second PDSC entry for the same part).

I'm going to raise this with cmsis-pack-manager, but if it is fixed there then presumably a change will be needed to support it here also.

tdasika, 2023-11-02:
Any update on this.?
I'm trying to use ATSAMV71Q21 board, when I give the command pyocd list, I see that target is n/a, even after installing the pack, 
I tried installing both Microchip specific pack and also Keil.SAM-V_DFP, but doesn't help.
 #   Probe/Board                  Unique ID                  Target
-------------------------------------------------------------------------------
  0   Atmel Corp. EDBG CMSIS-DAP   ATMLxxxxxxxxxxxxxxxx       n/a

Is this EDBG supported by pyocd, or do I need to convert it to st-link.?
PS, I also tried by setting the option 'cmsis_dap.prefer_v1' to True

DaAwesomeP, 2025-11-25:
See #1675 for a possible workaround (install pack via a chip only available in the Microchip pack): `pyocd pack install ATSAMD21J17D`