pyocd/pyOCD #1496: Issues while trying to install pack stm32f303

moritz-geier, 2023-01-31:
I installed pyOCD via Zephyr in a virtual enviroment and want now to download an extra pack from [here](https://www.keil.com/dd2/Pack) but on calling `pyocd pack install stm32f303` I get `0000391 W No matching devices. Please make sure the pack index is up to date. [pack_cmd]` then when i try to update via `pyocd pack update` I get: ```
0000355 I Updating pack index... [pack_cmd]
08:34:23 [INFO] Downloaded http://www.keil.com/pack/index.pidx
08:34:23 [INFO] Found 833 Pdsc entries
08:34:24 [WARN] No Core found!33)
08:34:24 [WARN] No Core found!
... 44 Times the same message ...
08:34:24 [WARN] No Core found!
08:34:24 [WARN] XML error: Expecting </link> found </head>
08:34:24 [WARN] the end of the document has been reached prematurely
08:34:24 [WARN] Device found without a processor ATSAMDA1E14A
08:34:24 [WARN] Device found without a processor ATSAME70J19
08:34:24 [WARN] Device found without a processor ATSAMS70J19
08:34:24 [WARN] Device found without a processor ATSAMV70J19
08:34:24 [WARN] Device found without a processor ATSAMV71J19
08:34:24 [WARN] No Core found!
08:34:24 [WARN] No Core found!
08:34:24 [WARN] No Core found!
08:34:24 [WARN] No Core found!
08:34:24 [WARN] No Core found!
08:34:24 [WARN] No Core found!
08:34:24 [WARN] cannot parse integer from empty string
08:34:24 [WARN] cannot parse integer from empty string
08:34:24 [WARN] cannot parse integer from empty string
08:34:24 [WARN] cannot parse integer from empty string
08:34:24 [WARN] cannot parse integer from empty string
08:34:24 [WARN] cannot parse integer from empty string
08:34:24 [WARN] No Core found!
08:34:24 [WARN] No Core found!
08:34:24 [WARN] No Core found!
08:34:24 [WARN] No Core found!
08:34:24 [WARN] No Core found!
08:34:24 [WARN] No Core found!
08:34:24 [WARN] Support for two parent processors not implemented!
08:34:24 [WARN] No Core found!
08:34:24 [WARN] No Core found!
08:34:24 [WARN] Support for two parent processors not implemented!
08:34:24 [WARN] No Core found!
08:34:24 [WARN] No Core found!
08:34:24 [WARN] No Core found!
08:34:24 [WARN] No Core found!
08:34:24 [WARN] XML error: Expecting </P> found </BODY>
... 62 Times the same message ...
08:34:24 [WARN] XML error: Expecting </P> found </BODY>
08:34:25 [WARN] Support for two parent processors not implemented!
```
I already tried to run `pyocd pack -c` but ended in the same results.

System:
- Ubuntu 22.04
- pyOCD 0.34.3

mathias-arm, 2023-01-31:
Does `~/.local/share/cmsis-pack-manager/Keil.STM32F3xx_DFP.2.2.2.pdsc` (or another version) and is it a valid XML file?

It should start with something like:
```
<?xml version="1.0" encoding="UTF-8"?>

<package schemaVersion="1.7.2" xmlns:xs="http://www.w3.org/2001/XMLSchema-instance" xs:noNamespaceSchemaLocation="PACK.xsd">
```

If it starts with `<!doctype html>` or similar try deleting the file and restart update.

flit, 2023-01-31:
Hi @moritz-geier. Please try upgrading `cmsis-pack-manager` via `pip install -U cmsis-pack-manager>=0.5.2`. We just released a fix for a long standing issue where systems that have a fast net connection could run trigger DDoS protections on the Keil and NXP pack (index) servers due to more than 6 concurrent requests.

You will need to clean and re-update to ensure the index is correctly downloaded: `pyocd pack update --clean`

I'll be releasing a new version of pyocd soon that requires this version of `cmsis-pack-manager`, so everyone will automatically get the fix.

moritz-geier, 2023-01-31:
It worked, thanks!
But while executing `pyocd pack update --clean` I'm getting following error message
```
17:51:12 [ERROR] download of Url { scheme: "http", cannot_be_a_base: false, username: "", password: None, host: Some(Domain("www.mcu.com.cn")), port: None, path: "/Cmsemicon.BAT32G135-A.pdsc", query: None, fragment: None } failed: Response code in invalid range: 404
```

flit, 2023-01-31:
Great!

Yeah, you'll see some download failures when updating the index. Some of the pack vendors just have invalid packs listed in the index, or problems with their server. Not much we can do about that except report them as errors. (Any pack tool will show the same errors.)

moritz-geier, 2023-01-31:
Okay thanks for your help!

closed 2023-01-31 by moritz-geier