pyocd/pyOCD #1499: pyocd failed to recognize STM32H745XI board

XiaZhouZero, 2023-02-17:
Hi everyone! I was attempting to use PyOCD to set up GDB servers for the STM32H745XIH6 board. Initially, I checked the board information by running the command:
```
 $ pyocd list
  #   Probe/Board   Unique ID                  Target
----------------------------------------------------------------
  0   STLINK-V3     0051002E3137511839383538   ✖︎ stm32h745zihx
      DISCO-H745I
```
The output showed that the board was recognized as an "stm32h745zihx" board, even though it should have been recognized as an "stm32h745xihx" board.

Then I checked the [MD5 software pack list](https://www.keil.com/dd2/pack/) and the related pack was `Keil.STM32H7xx_DFP.3.0.0.pack`. 

After that, I installed the related pdsc.
```
 $ pyocd pack install stm32h745
Downloading packs (press Control-C to cancel):
    Keil.STM32H7xx_DFP.3.0.0
```

After this, I checked the installed packs by running the command:
```
 $ pyocd pack find stm32h745
  Part            Vendor               Pack                 Version   Installed
---------------------------------------------------------------------------------
  STM32H745BGTx   STMicroelectronics   Keil.STM32H7xx_DFP   3.0.0     True
  STM32H745BITx   STMicroelectronics   Keil.STM32H7xx_DFP   3.0.0     True
  STM32H745IGKx   STMicroelectronics   Keil.STM32H7xx_DFP   3.0.0     True
  STM32H745IGTx   STMicroelectronics   Keil.STM32H7xx_DFP   3.0.0     True
  STM32H745IIKx   STMicroelectronics   Keil.STM32H7xx_DFP   3.0.0     True
  STM32H745IITx   STMicroelectronics   Keil.STM32H7xx_DFP   3.0.0     True
  STM32H745XGHx   STMicroelectronics   Keil.STM32H7xx_DFP   3.0.0     True
  **STM32H745XIHx   STMicroelectronics   Keil.STM32H7xx_DFP   3.0.0     True**
  STM32H745ZGTx   STMicroelectronics   Keil.STM32H7xx_DFP   3.0.0     True
  STM32H745ZITx   STMicroelectronics   Keil.STM32H7xx_DFP   3.0.0     True
```
The output showed that PyOCD does support "STM32H745XIHx" and "STM32H745ZITx", but not "stm32h745zihx".

So, my question is whether this is a bug of PyOCD, or if there is something wrong with my board. Can anyone help me out? Thank you in advance for your assistance!

mathias-arm, 2023-02-17:
It seems the value is wrong in [pyocd/board/board_ids.p](https://github.com/pyocd/pyOCD/blob/main/pyocd/board/board_ids.py#L183)

You can easily work around that with:
```
$ pyocd gdbserver -t STM32H745XIHx
```

XiaZhouZero, 2023-02-17:
Thank you for your kind reply @mathias-arm !
But I encountered another error after executing this command. It reported a `Memory transfer fault`.

```
 $ pyocd gdbserver -t STM32H745XIHx
0001072 I Target type is stm32h745xihx [board]
0001202 I DP IDR = 0x6ba02477 (v2 rev6) [dap]
0001310 I AHB-AP#0 IDR = 0x84770001 (AHB-AP var0 rev8) [ap]
0001312 I AHB-AP#1 IDR = 0x84770001 (AHB-AP var0 rev8) [ap]
0001313 I APB-AP#2 IDR = 0x54770002 (APB-AP var0 rev5) [ap]
0001315 I AHB-AP#3 IDR = 0x24770011 (AHB-AP var1 rev2) [ap]
0001317 I AHB-AP#0 Class 0x1 ROM table #0 @ 0xe00fe000 (designer=020:ST part=450) [rom_table]
0001319 I [0]<e00ff000:ROM class=1 designer=43b:Arm part=4c7> [rom_table]
0001319 I   AHB-AP#0 Class 0x1 ROM table #1 @ 0xe00ff000 (designer=43b:Arm part=4c7) [rom_table]
0001320 I   [0]<e000e000:SCS v7-M class=14 designer=43b:Arm part=00c> [rom_table]
0001321 I   [1]<e0001000:DWT v7-M class=14 designer=43b:Arm part=002> [rom_table]
0001322 I   [2]<e0002000:FPB v7-M class=14 designer=43b:Arm part=00e> [rom_table]
0001322 I   [3]<e0000000:ITM v7-M class=14 designer=43b:Arm part=001> [rom_table]
0001323 I [1]<e0041000:ETM M7 class=9 designer=43b:Arm part=975 devtype=13 archid=4a13 devid=0:0:0> [rom_table]
0001324 I [2]<e0043000:CTI class=9 designer=43b:Arm part=906 devtype=14 archid=0000 devid=40800:0:0> [rom_table]
0001325 I APB-AP#2 Class 0x1 ROM table #0 @ 0xe00e0000 (designer=020:ST part=450) [rom_table]
0001327 W Invalid coresight component, cidr=0x0 [rom_table]
0001327 I [2]e00e3000: cidr=0, pidr=0, component invalid> [rom_table]
0001328 W Invalid coresight component, cidr=0x0 [rom_table]
0001328 I [3]e00e4000: cidr=0, pidr=0, component invalid> [rom_table]
0001328 I [4]<e00e5000:TSGEN class=15 designer=43b:Arm part=101> [rom_table]
0001329 E Error attempting to probe CoreSight component referenced by ROM table entry #5: Memory transfer fault (read) @ 0xe00f0fd0-0xe00f102f [rom_table]
Traceback (most recent call last):
  File "/home/xiazhou/.local/lib/python3.8/site-packages/pyocd/coresight/rom_table.py", line 385, in _read_table
    self._handle_table_entry(entry, entryNumber)
  File "/home/xiazhou/.local/lib/python3.8/site-packages/pyocd/coresight/rom_table.py", line 440, in _handle_table_entry
    cmpid.read_id_registers()
  File "/home/xiazhou/.local/lib/python3.8/site-packages/pyocd/coresight/rom_table.py", line 124, in read_id_registers
    regs = self.ap.read_memory_block32(self.top_address + self.IDR_READ_START, self.IDR_READ_COUNT)
  File "/home/xiazhou/.local/lib/python3.8/site-packages/pyocd/utility/concurrency.py", line 29, in _locking
    return func(self, *args, **kwargs)
  File "/home/xiazhou/.local/lib/python3.8/site-packages/pyocd/coresight/ap.py", line 1216, in _accelerated_read_memory_block32
    return self._accelerated_memory_interface.read_memory_block32(addr, size,
  File "/home/xiazhou/.local/lib/python3.8/site-packages/pyocd/probe/stlink_probe.py", line 301, in read_memory_block32
    return conversion.byte_list_to_u32le_list(self._link.read_mem32(addr, size * 4, self._apsel, csw))
  File "/home/xiazhou/.local/lib/python3.8/site-packages/pyocd/probe/stlink/stlink.py", line 473, in read_mem32
    return self._read_mem(addr, size, Commands.JTAG_READMEM_32BIT, self.MAXIMUM_TRANSFER_SIZE, apsel, csw)
  File "/home/xiazhou/.local/lib/python3.8/site-packages/pyocd/probe/stlink/stlink.py", line 423, in _read_mem
    raise exc
pyocd.core.exceptions.TransferFaultError: Memory transfer fault (read) @ 0xe00f0fd0-0xe00f102f
0001331 I AHB-AP#3 Class 0x1 ROM table #0 @ 0xe00ff000 (designer=020:ST part=450) [rom_table]
0001332 I [0]<e000e000:SCS v7-M class=14 designer=43b:Arm part=00c> [rom_table]
0001333 I [1]<e0001000:DWT v7-M class=14 designer=43b:Arm part=002> [rom_table]
0001333 I [2]<e0002000:FPB v7-M class=14 designer=43b:Arm part=003> [rom_table]
0001334 I [3]<e0000000:ITM v7-M class=14 designer=43b:Arm part=001> [rom_table]
0001335 I [5]<e0041000:ETM M4 class=9 designer=43b:Arm part=925 devtype=13 archid=0000 devid=0:0:0> [rom_table]
0001336 I [6]<e0043000:CTI class=9 designer=43b:Arm part=906 devtype=14 archid=0000 devid=40800:0:0> [rom_table]
0001338 I CPU core #0 is Cortex-M7 r1p1 [cortex_m]
0001338 I FPU present: FPv5-D16-M [cortex_m]
0001340 I CPU core #1 is Cortex-M4 r0p1 [cortex_m]
0001340 I FPU present: FPv4-SP-D16-M [cortex_m]
0001341 I 4 hardware watchpoints [dwt]
0001343 I 8 hardware breakpoints, 1 literal comparators [fpb]
0001347 I 4 hardware watchpoints [dwt]
0001349 I 6 hardware breakpoints, 4 literal comparators [fpb]
0001354 I Semihost server started on port 4444 (core 0) [server]
0001377 I GDB server started on port 3333 (core 0) [gdbserver]
0001379 I Semihost server started on port 4445 (core 1) [server]
0001379 I GDB server started on port 3334 (core 1) [gdbserver]
```

PR #1501 (2023-03-19): Fix DISCO-H745I target name to stm32h745xihx
Fixes #1499
closed 2023-02-25 by XiaZhouZero