pyocd/pyOCD #1504: GDB Halts Cortex-m4 Failed for STM32H745I-DISCO Board

XiaZhouZero, 2023-02-28:
### Preliminary

Hi there. I am writing a target file for a STM32H745I-DISCO board. The board contains a STM32H745XIH6 MCU, which has Cortex-M7 and Cortex-M4 integrated. I followed the instructions from the documentation [Adding a new built-in target](https://pyocd.io/docs/adding_new_targets.html) to generate a new target file for this board.

First, I use a target file, target_STM32F767xx.py, as my template.

After that, I generate the `FLASH_ALGO`.
```bash
$ python scripts/generate_flash_algo.py ../Keil.STM32H7xx_DFP.3.0.0/CMSIS/Flash/MT25TL01G_STM32H745I-DISCO.FLM
```

Next, I add `FlashRegion` and `RamRegion` to the class `STM32H745XIH6`  inherited from class `CoreSightTarget`. I retrieved the Flash and Ram memory information from the [Reference manual](https://www.st.com/resource/en/reference_manual/rm0399-stm32h745755-and-stm32h747757-advanced-armbased-32bit-mcus-stmicroelectronics.pdf).

Finally, I could successfully setup gdb servers for the STM32H745I-DISCO.

### My Issue

However, when I tried to load the program into STM32H745I-DISCO, I need to halt and load the program for Cortex-M7  first. It will fail if I directly tried to load the program for Cortex-M4. 

**Successful case**

First, I halt and load program for Cortex-M7. Second, I halt and load program for Cortex-M4.

+ Terminal 1
I used pyocd to setup two gdb servers, which listen to port 3333 and 3334, respectively.
```bash
 $ pyocd gdbserver -t stm32h745xih6
0000960 I Target type is stm32h745xih6 [board]
0000964 I DP IDR = 0x6ba02477 (v2 rev6) [dap]
0000966 I AHB-AP#0 IDR = 0x84770001 (AHB-AP var0 rev8) [ap]
0000968 I AHB-AP#1 IDR = 0x84770001 (AHB-AP var0 rev8) [ap]
0000970 I APB-AP#2 IDR = 0x54770002 (APB-AP var0 rev5) [ap]
0000972 I AHB-AP#3 IDR = 0x24770011 (AHB-AP var1 rev2) [ap]
0000974 I AHB-AP#0 Class 0x1 ROM table #0 @ 0xe00fe000 (designer=020:ST part=450) [rom_table]
0000976 I [0]<e00ff000:ROM class=1 designer=43b:Arm part=4c7> [rom_table]
0000976 I   AHB-AP#0 Class 0x1 ROM table #1 @ 0xe00ff000 (designer=43b:Arm part=4c7) [rom_table]
0000977 I   [0]<e000e000:SCS v7-M class=14 designer=43b:Arm part=00c> [rom_table]
0000978 I   [1]<e0001000:DWT v7-M class=14 designer=43b:Arm part=002> [rom_table]
0000978 I   [2]<e0002000:FPB v7-M class=14 designer=43b:Arm part=00e> [rom_table]
0000979 I   [3]<e0000000:ITM v7-M class=14 designer=43b:Arm part=001> [rom_table]
0000980 I [1]<e0041000:ETM M7 class=9 designer=43b:Arm part=975 devtype=13 archid=4a13 devid=0:0:0> [rom_table]
0000981 I [2]<e0043000:CTI class=9 designer=43b:Arm part=906 devtype=14 archid=0000 devid=40800:0:0> [rom_table]
0000982 I APB-AP#2 Class 0x1 ROM table #0 @ 0xe00e0000 (designer=020:ST part=450) [rom_table]
0000984 I [2]<e00e3000:SWO CS-400 class=9 designer=43b:Arm part=914 devtype=11 archid=0000 devid=ea0:0:0> [rom_table]
0000985 I [3]<e00e4000:CSTF class=9 designer=43b:Arm part=908 devtype=12 archid=0000 devid=32:0:0> [rom_table]
0000986 I [4]<e00e5000:TSGEN class=15 designer=43b:Arm part=101> [rom_table]
0000987 I [5]<e00f0000:ROM class=1 designer=020:ST part=001> [rom_table]
0000987 I   APB-AP#2 Class 0x1 ROM table #1 @ 0xe00f0000 (designer=020:ST part=001) [rom_table]
0000988 I   [0]<e00f1000:CTI class=9 designer=43b:Arm part=906 devtype=14 archid=0000 devid=40800:0:0> [rom_table]
0000989 I   [2]<e00f3000:CSTF class=9 designer=43b:Arm part=908 devtype=12 archid=0000 devid=34:0:0> [rom_table]
0000991 I   [3]<e00f4000:ETF class=9 designer=43b:Arm part=961 devtype=32 archid=0000 devid=380:0:0> [rom_table]
0000992 I   [4]<e00f5000:TPIU class=9 designer=43b:Arm part=912 devtype=11 archid=0000 devid=a0:0:0> [rom_table]
0000993 I AHB-AP#3 Class 0x1 ROM table #0 @ 0xe00ff000 (designer=020:ST part=450) [rom_table]
0000995 I [0]<e000e000:SCS v7-M class=14 designer=43b:Arm part=00c> [rom_table]
0000995 I [1]<e0001000:DWT v7-M class=14 designer=43b:Arm part=002> [rom_table]
0000996 I [2]<e0002000:FPB v7-M class=14 designer=43b:Arm part=003> [rom_table]
0000997 I [3]<e0000000:ITM v7-M class=14 designer=43b:Arm part=001> [rom_table]
0000998 I [5]<e0041000:ETM M4 class=9 designer=43b:Arm part=925 devtype=13 archid=0000 devid=0:0:0> [rom_table]
0000999 I [6]<e0043000:CTI class=9 designer=43b:Arm part=906 devtype=14 archid=0000 devid=40800:0:0> [rom_table]
0001001 I CPU core #0 is Cortex-M7 r1p1 [cortex_m]
0001001 I FPU present: FPv5-D16-M [cortex_m]
0001003 I CPU core #1 is Cortex-M4 r0p1 [cortex_m]
0001004 I FPU present: FPv4-SP-D16-M [cortex_m]
0001005 I 4 hardware watchpoints [dwt]
0001007 I 8 hardware breakpoints, 1 literal comparators [fpb]
0001012 I 4 hardware watchpoints [dwt]
0001014 I 6 hardware breakpoints, 4 literal comparators [fpb]
[DEBUG] begin_stack: 0x2000c500
[DEBUG] begin_stack: 0x2000c500
0001020 I Semihost server started on port 4444 (core 0) [server]
[... 148 lines cut ...]
(gdb) monitor reset halt
Resetting target with halt
Successfully halted device on reset
(gdb) load
Loading section ._user_heap_stack, size 0x1000 lma 0x10001240
Loading section .isr_vector, size 0x298 lma 0x8100000
Loading section .text, size 0xe428 lma 0x81002c0
Loading section .rodata, size 0x964 lma 0x810e6e8
Loading section .ARM, size 0x10 lma 0x810f04c
Loading section .data, size 0x9dc lma 0x810f060
Start address 0x081011a0, load size 68112
Could not write register "pc"; remote failure reply 'E01'
(gdb) q
```

It seemed that I cannot halt the Cortex-M4 (core #1) to modify the `pc` register.

I further looked into the code from target_STM32F767xx.py. It overrides the function `create_init_sequence()` of class `CoreSightTarget`. However, the documentation does not mention whether or not we can override member functions of class `CoreSightTarget`.

### My Question

+ Does the error posted above related to member functions of class `CoreSightTarget`?
+ If so, should I override the function `create_init_sequence()` like target_STM32F767xx.py does?

Could anybody help me out? Thanks your assistance in advance!

XiaZhouZero, 2023-03-05:
@0xc0170 @flit Can you please take a look into this issue?

unsanded, 2023-05-09:
Hi, I wrote a target file for STM32H723 and STM32H743 [here](https://github.com/unsanded/pyOCD/tree/feature/stm32h7). 
I know they are both single-core, but they are very similar.
I will make a merge request soon, but i feel like almost all stm32's have more or less the same features, both my files are very similar to each other, and also a lot was copy-pasted from stm32f767. So maybe we should look into a common family file ( i don't know the structure of this project well enough).