pyocd/pyOCD #1513: How to build binary?

USTHzhanglu, 2023-03-16:
i read [How to build binary](https://pyocd.io/docs/how_to_build.html) and try it ,but It seems that pyocd.py is not available now.
Is generating binary no longer supported？thinks

flit, 2023-03-17:
Yeah, sorry that the docs are out of date! It requires a custom `.spec` file for PyInstaller, that hasn't fully worked for a long time. A few days ago, a user found a way to solve the issue, so I'll be updating the docs shortly. And hopefully providing a pre-built binary for the 3 major OSes.

USTHzhanglu, 2023-03-18:
> Yeah, sorry that the docs are out of date! It requires a custom `.spec` file for PyInstaller, that hasn't fully worked for a long time. A few days ago, a user found a way to solve the issue, so I'll be updating the docs shortly. And hopefully providing a pre-built binary for the 3 major OSes.

Oh, that's very good. Looking forward to your update

cederom, 2023-03-18:
One friend told me about https://github.com/exaloop/codon do you know this python-to-binary-compiler project? :-)

flit, 2023-03-19:
@cederom Hi there, maybe I saw it quite a while ago? Seems pretty awesome! Definitely should try it out with pyocd. Thanks! 😄

flit, 2023-03-19:
Hmm, not a good experience so far:

```
❯ codon run pyocd/__main__.py
__main__.py:109:1: error: unexpected dedent
```

And there's nothing wrong with the indentation of that file, certainly not around that line. Oh well, should look at it further at some point.

cederom, 2023-03-19:
just a curiosity @flit I never used that myself before :-) if that requires a major code rewrite or any sort of backward incompatibility then its not worth it :-)

flit, 2023-04-09:
Fyi, please see #1529 for the notes on how to build a binary with PyInstaller.

I'm going to close this issue since that one has a more specific error, plus the notes.

closed 2023-04-09 by flit