pyocd/pyOCD #1515: Semihosting and STD*_FILENO

rgrr, 2023-03-16:
Hello,

when using semihosting and trying to send a message to stdout via `write(STDOUT_FILENO,...)`, output goes to nowhere under pyocd, openocd does the expected output.

This is because "STDOUT_FILENO=1" according to C standards but pyocd expects it to be "2".  The sources (https://github.com/pyocd/pyOCD/blob/main/pyocd/debug/semihost.py#64) shows "STDOUT_FD=2" .

If one uses "2" on the target, the output appears.  In the sources mentioned, it is stated that "The fds must be non-zero according to the ARM semihosting spec".  (I've searched the spec but cannot find it)

Now I'm wondering who is mistaken?  C lib defines "STDOUT_FILENO=1" for ages and this is certainly an incompatibility.

rgrr, 2023-03-16:
Perhaps I have found the place in the spec: https://developer.arm.com/documentation/dui0471/g/Semihosting/SYS-OPEN--0x01-?lang=en

It states

```
Return

On exit, R0 contains:

    a nonzero handle if the call is successful

    -1 if the call is not successful.
```

But writes to stdout do not need any open().  So the spec is wrong as well at https://developer.arm.com/documentation/dui0471/g/Semihosting/SYS-WRITE--0x05-?lang=en when it says:

```
Entry

On entry, R1 contains a pointer to a three-word data block:

word 1

    contains a handle for a file previously opened with SYS_OPEN
```

Because, as said, stdout/stderr don't have to be opened.

For compatibility with the C library correct behavior on pyocd side would be to use the same values for the std* handles as the C lib does.

flit, 2023-03-17:
Newlib definitely does the expected behaviour in regards to opening stdio prior to its use, eg opening a file called ":tt" with various modes. Hopefully it's possible to support both behaviours.

rgrr, 2023-03-17:
Hmmm... I guess the spec of the C library is a little bit older than the Arm semihosting spec.  The values of the special file descriptors for stdio made it even into wikipedia, see https://en.wikipedia.org/wiki/File_descriptor.

Even the comment in https://github.com/pyocd/pyOCD/blob/main/pyocd/debug/semihost.py#105 says, that the values are 0/1/2.  But effectively they are 1/2/3, which is clearly wrong.
OpenOCD behaves as expected: "write(1, buf, sizeof(buf))" sends characters to stdout, pyOCD needs "write(2, buf, sizeof(buf))".

Greatest compatibility has to be achieved with the values 0/1/2 for stdin/stdout/stderr.

PS: I agree that handling of ":tt" is as expected.  But standard file descriptors 0/1/2 actually don't have to be opened

flit, 2023-03-17:
This is all kind of beside the point, but to be clear about what I'm thinking: What you're describing is the POSIX behaviour. But embedded systems are not POSIX compliant, and semihosting is not a standard feature of C. So the C library is responsible for complying with the semihosting specification. And actually, `write()` and `read()` are POSIX APIs, not part of the standard C library. So you could even go so far as to say it's a bug in the C lib because it provides POSIX APIs but doesn't correctly map them to the chosen I/O interface. But again, all beside the point because it doesn't help anyone.

The openocd behaviour is best thought of as an undocumented and uncommon extension. For instance, QEMU's Arm semihosting implementation doesn't appear to support it (just quickly checked the code). Nor do the other Arm semihosting implementations I'm familiar with in several debuggers.

Could you create a pull request to implement this? I'd be happy to merge it in. Otherwise it'll probably take a while to get implemented just because of the backlog.

rgrr, 2023-03-17:
I doubt that it was the point of Arms spec to rewrite POSIX.  But... I'm not Arm and perhaps they wanted it... incompatible.

I will prepare a PR and check the side effects of changing the constants.

flit, 2023-03-17:
Well, knowing a little of the history here… the semihosting spec was created back in the day of the early Arm RealView IDE and toolchain (>20 years ago). It had a custom C library (which is still included with AC5 and AC6) and a lot of semihosting integration. POSIX compatibility was probably far from their minds, as is true with most embedded systems even today. The details of the semihosting interface were published as a spec, got picked up by other tools, and was eventually ported to M-profile.

Anyway, thanks for your work on this. I'm all for being compatible with as many tools as possible. My apologies if I seemed to push back or be overly pedantic; I regularly work a lot with software and hardware specifications (at Arm, no less) so the details of how specifications are interpreted and implemented are a prime concern for me. 😄

rgrr, 2023-03-18:
Don't worry about the tone.  I already thought that I was too insisting ;-)

Your response times are quite refreshing which is very motivating.  Comparing to several other projects were you get the feeling talking to dead walls.

The actual patch is very simple.  But I'm currently struggling with the tests.  "test/unit/test_semihosting.py" seems to be at the wrong place.  It requires a probe to work.  And that's were I'm stuck: my probe/target doesn't seem to be accepted (it currently provides no target auto detect).  Any hints how to setup the probe/target under test manually to get the test run?

flit, 2023-03-19:
Well, my response times vary a lot depending on . Sometimes I'm really bad, and completely miss issues, and get far behind. So I definitely can't be commended about that… 😔 Otoh, I'm almost always quite responsive in the pyocd Slack workspace.

Thanks for the patch! And thanks for the comments about the test environment. It does have lots of issues, though.

flit, 2023-03-19:
PR merged, so closing this issue.

closed 2023-03-19 by flit