pyocd/pyOCD #1518: Semihosting and binary file write

rgrr, 2023-03-19:
I guess there is a problem with writing binary data to files via semihosting.

semihosting::_get_string() converts data to (utf-8) strings regardless if the data is expected to be a string or not.

If the binary data from the target contains ill utf-8 data, pyocd throws exceptions, e.g.

```
...
  File "/home/hardy/.local/lib/python3.11/site-packages/pyocd/debug/semihost.py", line 538, in handle_sys_write
    return self.io_handler.write(fd, data_ptr, length)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/hardy/.local/lib/python3.11/site-packages/pyocd/debug/semihost.py", line 223, in write
    data = self.agent._get_string(ptr, length)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/hardy/.local/lib/python3.11/site-packages/pyocd/debug/semihost.py", line 485, in _get_string
    return bytes(data).decode()
           ^^^^^^^^^^^^^^^^^^^^
UnicodeDecodeError: 'utf-8' codec can't decode byte 0x81 in position 0: invalid start byte
```

I will try to look into this when there is time.

rgrr, 2023-03-20:
After thinking a little bit about it: to my opinion there shouldn't be anything like .decode() / .encode() in pyocd in the semihosting part, because this is actually some kind of transport level which should be transparent.

If the files are opened with correct mode from the target and pyocd, then the operating system should do the rest.  With pyocd interpreting the data stream, behavior becomes intransparent.

As said: my opinion.  What do the developer/maintainer think?

flit, 2023-03-20:
You're mostly right. 😄 

There is a distinction between binary and text files in the semihost API, and that results in different Python file objects being created. So there needs to be an encode/decode when writing to text files, but not for binary files.

The main change needed is that `SemihostAgent._get_string()` method should not decode the string itself. It should return a `bytes` object.

`InternalSemihostIOHandler` already appears to check the file mode and encode/decode as necessary. 
`GDBSyscallIOHandler` should just pass binary to the GDB syscall commands.

Not entirely sure about `ConsoleIOHandler`. Probably needs to encode/decode.

rgrr, 2023-03-21:
I had a look at picolib as well and... "fun" fact:  there is actually no distinction between binary and text mode possible, O_BINARY/O_TEXT are only defined and relevant for Cygwin open() operation :-/.

So the picolib side has to decide if files are always opened in binary or in text mode (SH_OPEN_*) on target side.  To my opinion the binary modes are more generic, so more correct.  Currently it's unfortunately the other way round, so picolib uses the non-binary modes, which leads to problems on the host side, namely pyOCD.

I'm in discussion with the picolib developer(s).

flit, 2023-03-21:
I agree with you, binary mode should be the default if the mode passed into open() isn't honoured.

rgrr, 2023-03-22:
Before I start changing something: do you agree that actually there should be no decode/encode within the semihost part and that the data is transported transparently?
This includes read/write to stdio, because no one knows what the target developer transfers (is it really UTF-8?).

flit, 2023-03-22:
Yes, I agree. That basically means that only `SemihostAgent._get_string()` has to be changed to not decode the data. I could be missing something though!

The `*IOHandler` classes, however, are part of the transport. There are two ways to deal with this:

1. Continue with current implementation, and encode/decode as required by the file type. Except `GDBSyscallIOHandler`, which should transfer the original data to the GDB syscalls. (This is just a recap of the comments above.)

    This means that files opened as binary would be fully transparent (currently broken, as you know), while text files would be encoded/decoded as appropriate.

    Aside: There is no way to specify the encoding for text files in the semihosting `SYS_OPEN` call. So `InternalSemihostIOHandler` should probably open text files using UTF-8 by default (I think that's already the case), and there should be a session option to specify the encoding. Could even have separate "target-side encoding" and "file encoding" options.

2. Just ignore the text/binary mode flag, open all files as binary. Assume that the data from the target is correctly encoded / the target can handle data encoded in the file's coding. This will probably work fine. It's a bigger change to the code, however. And it would make it impossible to perform encoding translations if that were desired (and implemented).

PR #1520 (2023-04-09): semihosting transparent io
Semihosting IO no longer does internal encode() / decode() steps because this fails on binary data, e.g. coverage dumps.

There is an additional test to verify correct behavior for binary data.

Closes #1518
closed 2023-04-09 by flit