pyocd/pyOCD #1521: Semihosting: How about halting debugger on SYS_EXIT?

rgrr, 2023-03-24:
OpenOCD behaves that way, that debugging is halted if the target issues SYS_EXIT (0x18).

Should be a oneliner if one knows the corresponding call.

And another idea which goes into the same direction: how about command line options to start the target in gdbserver mode and exit pyocd if SYS_EXIT is executed?  This would allow very easy in target coverage testing.

Hoohaha, 2023-03-27:
Hi @rgrr ,

We are using gdb command: `continue &` to start execution. `&` can make continue run in background and not blocking the gdb.

If you want to exit debuging, you can send `interrupt`.

We are using this way to get coverage data from target with pyocd.

rgrr, 2023-03-31:
My idea is to omit gdb and do pyocd the job.  After finding out, that pyocd is controllable via API there seem to be multiple ways to get to the goal.  I guess I will expand the API examples section a bit.

Nevertheless, openocd behavior to halt debugging (as if there was a breakpoint) is TMO very helpful if in a debug session.  Pyocd currently shows you a "running" target in gdb although effectively it has stopped at the BKPT.