pyocd/pyOCD #1523: Failures when testing programmers with pyocd and the puya py32

prosper00, 2023-03-31:
**Summary**

&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;I recently bought a stack of puya py32 mcu’s after reading this [article](https://jaycarlson.net/2023/02/04/the-cheapest-flash-microcontroller-you-can-buy-is-actually-an-arm-cortex-m0/)
 
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; I’ve been testing them out using the SDK [here](https://github.com/IOsetting/py32f0-template):
 
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; I noticed that many of my daplinks and other programmers fail to connect
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; to this mcu with pyocd \[and openocd. though I didn’t test it as
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; thoroughly\] when using the pack file provided by the vendor [here](https://www.puyasemi.com/uploadfiles/2022/11/PY-MCU资料-20221117.rar)
 
 
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; I have a stack of different programmers and mcu’s running daplink
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; firmwares that I tested, and only one of them works OK. Most of the
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; daplinks fail with:

> pyocd.core.exceptions.TransferError: Unexpected ACK value (5) returned by probe

 

**Method**

&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;To ensure the probes are working OK, I tested each on an STM32F103C8
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;mcu, as well as with a Puya PY32F002AL15. I figured the STM should be a
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;good control test, as it’s a very well supported and commonplace mature
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;product.

&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;I’m running pycod 0.34.3.dev36 – though I did also test .34.3 release as
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;well with identical results.

&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;My OS is Ubuntu lunar (dev), on linux 5.18.13

&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;I prepared a firmware for each mcu, and programmed it with the following
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;commands:

**STM32:**

```$ pyocd erase --chip -t stm32f103c8; pyocd load -t stm32f103c8 ./miniblink.bin```

**PY32:**

```$ pyocd load ./Build/example-blink.hex -t py32f002x5 --config ./Misc/pyocd.yaml```

&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;(the config file simply loads the [pack file](https://www.puyasemi.com/uploadfiles/2022/11/PY-MCU资料-20221117.rar) from the vendor)

```
$ cat pyocd.yaml
pack:
\- Misc/Puya.PY32F0xx_DFP.1.1.3.pack
```

**Results table**

|                                                                            |                                    |                                                  |
|----------------------------------------------------------------------------|------------------------------------|--------------------------------------------------|
| **Probe**                                                                      | **STM32**                              | **PY32**                                             |
|**DAPLinks**| | |
| [jixin.pro DAPLINK](https://www.aliexpress.com/item/1005004518845326.html) | works                              | works                                            |
| [rp2040 picoprobe-cmsis-v1.0.2](https://github.com/raspberrypi/picoprobe/releases)                                             | works                              | Unexpected ACK value (5)                         |
| [rp2040 DapperMime](https://github.com/majbthrd/DapperMime)                                                          | works                              | Unexpected ACK value (5)                         |
| [DAPLink clear](https://www.aliexpress.com/item/1005003056314140.html)                                                            | works                              | Unexpected ACK value (5)                         |
| [WCHLink](https://www.aliexpress.com/item/1005003059548220.html)                                                                    | Unexpected ACK value (5)           | Unexpected ACK value (5)                         |
| [Wagiminator – CH552-picoDAP](https://github.com/wagiminator/CH552-picoDAP)                                                | Timeout reading from device CH55x  | Timeout reading from device CH55x                |
|**STLinks**| | |
| [STLinkV2 – baite ](https://www.aliexpress.com/item/1005005002268447.html)                                                          | STLink error (9): Get IDCODE error | STLink error (9): Get IDCODE error               |
| [STLinkV2 - metal](https://www.aliexpress.com/item/1005001724949176.html)                                                           | works                              | STLink error (9): Get IDCODE error               |
|**JLinks**| | |
| [JlinkOB – baite](https://www.aliexpress.com/item/1005004902159370.html)                                                            | works                              | Memory transfer fault (Unspecified error.)       |
| [JlinkOB 072 – baite ](https://www.aliexpress.com/item/1005004902256035.html)                                                       | works                              | RecursionError: maximum recursion depth exceeded |

Details: [Programmers.odt](https://github.com/pyocd/pyOCD/files/11123490/Programmers.odt)
Details from each probe are in the comments below

prosper00, 2023-03-31:
**[jixin.pro DAPLINK](https://www.aliexpress.com/item/1005004518845326.html)**

[Mar31 08:36] usb 3-2.2: new full-speed USB device number 89 using xhci_hcd
[  +0.146097] usb 3-2.2: New USB device found, idVendor=c251, idProduct=f001, bcdDevice= 1.00
[  +0.000006] usb 3-2.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  +0.000002] usb 3-2.2: Product: CMSIS-DAP_LU
[  +0.000002] usb 3-2.2: Manufacturer: jixin.pro
[  +0.000001] usb 3-2.2: SerialNumber: LU_2022_8888
[  +0.134068] cdc_acm 3-2.2:1.0: ttyACM1: USB ACM device
[  +0.014176] hid-generic 0003:C251:F001.00D4: hiddev2,hidraw5: USB HID v1.00 Device [jixin.pro CMSIS-DAP_LU] on usb-0000:05:00.3-2.2/input2

`$ pyocd --version`
0.34.3.dev36

`$ pyocd list`
```
  #   Probe/Board              Unique ID      Target  
------------------------------------------------------
  0   jixin.pro CMSIS-DAP_LU   LU_2022_8888   n/a     
```
**STM32**
```
$ pyocd erase --chip -t stm32f103c8; pyocd load -t stm32f103c8 ./miniblink.bin
0000942 I Erasing chip... [eraser]
0001164 I Chip erase complete [eraser]
0000971 I Loading /home/bdroy/STM32-base/libopencm3-examples/examples/stm32/f1/stm32-maple/miniblink/miniblink.bin [load_cmd]
[==================================================] 100%
0001498 I Erased 1024 bytes (1 sector), programmed 1024 bytes (1 page), skipped 0 bytes (0 pages) at 1.90 kB/s [loader]
```
**PY32**
```
$ pyocd load ./Build/example-blink.hex -t py32f002x5 --config ./Misc/pyocd.yaml
0000816 I Loading /home/bdroy/py32-base/Example-LL-GPIO-32f002AL15-2/Build/example-blink.hex [load_cmd]
[==================================================] 100%
0001358 I Erased 0 bytes (0 sectors), programmed 0 bytes (0 pages), skipped 1152 bytes (9 pages) at 2.08 kB/s [loader]
```

prosper00, 2023-03-31:
**[rp2040 picoprobe-cmsis-v1.0.2](https://github.com/raspberrypi/picoprobe/releases)** 

```
[Mar30 21:35] usb 3-2.2: new full-speed USB device number 74 using xhci_hcd
[  +0.152667] usb 3-2.2: New USB device found, idVendor=2e8a, idProduct=000c, bcdDevice= 1.01
[  +0.000006] usb 3-2.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  +0.000002] usb 3-2.2: Product: Picoprobe CMSIS-DAP
[  +0.000002] usb 3-2.2: Manufacturer: Raspberry Pi
[  +0.000001] usb 3-2.2: SerialNumber: DE616C82571F2D33
[  +0.071227] cdc_acm 3-2.2:1.1: ttyACM1: USB ACM device
```
```
$ pyocd list
  #   Probe/Board                        Unique ID          Target  
--------------------------------------------------------------------
  0   Raspberry Pi Picoprobe CMSIS-DAP   DE616C82571F2D33   n/a 
```
**STM32**
```
$ pyocd erase --chip -t stm32f103c8; pyocd load -t stm32f103c8 ./miniblink.bin 
0000942 I Erasing chip... [eraser]
0001164 I Chip erase complete [eraser]
0000971 I Loading /home/bdroy/STM32-base/libopencm3-examples/examples/stm32/f1/stm32-maple/miniblink/miniblink.bin [load_cmd]
[==================================================] 100%
0001498 I Erased 1024 bytes (1 sector), programmed 1024 bytes (1 page), skipped 0 bytes (0 pages) at 1.90 kB/s [loader]
```
**PY32**
```
$ pyocd load ./Build/example-blink.hex -t py32f002x5 --config ./Misc/pyocd.yaml
0000705 C Unexpected ACK value (5) returned by probe [__main__]
Traceback (most recent call last):
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/probe/cmsis_dap_probe.py", line 541, in read_dp_result_callback
    value = result()
            ^^^^^^^^
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 1071, in read_reg_cb
    res = transfer.get_result()
          ^^^^^^^^^^^^^^^^^^^^^
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 157, in get_result
    self.daplink.flush()
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/utility/concurrency.py", line 29, in _locking
    return func(self, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 881, in flush
    self._read_packet()
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/utility/concurrency.py", line 29, in _locking
    return func(self, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 1157, in _read_packet
    decoded_data = cmd.decode_data(raw_data)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 478, in decode_data
    data = self._decode_transfer_block_data(data)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 447, in _decode_transfer_block_data
    self._check_response(data[3])
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 370, in _check_response
    raise DAPAccessIntf.TransferError("Unexpected ACK value (%d) returned by probe" % ack)
pyocd.probe.pydapaccess.dap_access_api.DAPAccessIntf.TransferError: Unexpected ACK value (5) returned by probe

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/__main__.py", line 161, in run
    status = cmd.invoke()
             ^^^^^^^^^^^^
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/subcommands/load_cmd.py", line 98, in invoke
    with session:
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/core/session.py", line 391, in __enter__
    self.open()
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/core/session.py", line 529, in open
[... 3 lines cut ...]
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/core/soc_target.py", line 157, in init
    seq.invoke()
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/utility/sequencer.py", line 213, in invoke
    resultSequence.invoke()
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/utility/sequencer.py", line 208, in invoke
    resultSequence = call()
                     ^^^^^^
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/coresight/dap.py", line 446, in _connect
    connector.connect()
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/coresight/dap.py", line 255, in connect
    self._idr = self.read_idr()
                ^^^^^^^^^^^^^^^
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/coresight/dap.py", line 284, in read_idr
    dpidr = self._dp.read_dp(DP_IDR, now=True)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/coresight/dap.py", line 792, in read_dp
    return read_dp_cb()
           ^^^^^^^^^^^^
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/coresight/dap.py", line 780, in read_dp_cb
    result = result_cb()
             ^^^^^^^^^^^
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/probe/cmsis_dap_probe.py", line 546, in read_dp_result_callback
    raise self._convert_exception(error) from error
pyocd.core.exceptions.TransferError: Unexpected ACK value (5) returned by probe
```

prosper00, 2023-03-31:
**[rp2040 DapperMime](https://github.com/majbthrd/DapperMime)**

```
[  +0.155295] usb 3-2.2: new full-speed USB device number 76 using xhci_hcd
[  +0.146445] usb 3-2.2: New USB device found, idVendor=cafe, idProduct=4005, bcdDevice= 1.01
[  +0.000005] usb 3-2.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  +0.000003] usb 3-2.2: Product: RP2040 CMSIS-DAP
[  +0.000001] usb 3-2.2: Manufacturer: TinyUSB
[  +0.000002] usb 3-2.2: SerialNumber: ED16C62875F1D233
[  +0.112185] hid-generic 0003:CAFE:4005.00A1: hiddev2,hidraw5: USB HID v1.11 Device [TinyUSB RP2040 CMSIS-DAP] on usb-0000:05:00.3-2.2/input0
[  +0.002843] cdc_acm 3-2.2:1.1: ttyACM1: USB ACM device
```
```
$ pyocd list
  #   Probe/Board                Unique ID          Target  
------------------------------------------------------------
  0   TinyUSB RP2040 CMSIS-DAP   ED16C62875F1D233   n/a     
```
**STM32**
```
$ pyocd erase --chip -t stm32f103c8; pyocd load -t stm32f103c8 ./miniblink.bin 
0000942 I Erasing chip... [eraser]
0001164 I Chip erase complete [eraser]
0000971 I Loading /home/bdroy/STM32-base/libopencm3-examples/examples/stm32/f1/stm32-maple/miniblink/miniblink.bin [load_cmd]
[==================================================] 100%
0001498 I Erased 1024 bytes (1 sector), programmed 1024 bytes (1 page), skipped 0 bytes (0 pages) at 1.90 kB/s [loader]
```
**PY32**
```
$ pyocd load ./Build/example-blink.hex -t py32f002x5 --config ./Misc/pyocd.yaml
0000757 C Unexpected ACK value (5) returned by probe [__main__]
Traceback (most recent call last):
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/probe/cmsis_dap_probe.py", line 541, in read_dp_result_callback
    value = result()
            ^^^^^^^^
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 1071, in read_reg_cb
    res = transfer.get_result()
          ^^^^^^^^^^^^^^^^^^^^^
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 157, in get_result
    self.daplink.flush()
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/utility/concurrency.py", line 29, in _locking
    return func(self, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 881, in flush
    self._read_packet()
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/utility/concurrency.py", line 29, in _locking
    return func(self, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 1157, in _read_packet
    decoded_data = cmd.decode_data(raw_data)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 478, in decode_data
    data = self._decode_transfer_block_data(data)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 447, in _decode_transfer_block_data
    self._check_response(data[3])
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 370, in _check_response
    raise DAPAccessIntf.TransferError("Unexpected ACK value (%d) returned by probe" % ack)
pyocd.probe.pydapaccess.dap_access_api.DAPAccessIntf.TransferError: Unexpected ACK value (5) returned by probe

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/__main__.py", line 161, in run
    status = cmd.invoke()
             ^^^^^^^^^^^^
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/subcommands/load_cmd.py", line 98, in invoke
    with session:
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/core/session.py", line 391, in __enter__
    self.open()
[... 4 lines cut ...]
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/core/soc_target.py", line 157, in init
    seq.invoke()
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/utility/sequencer.py", line 213, in invoke
    resultSequence.invoke()
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/utility/sequencer.py", line 208, in invoke
    resultSequence = call()
                     ^^^^^^
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/coresight/dap.py", line 446, in _connect
    connector.connect()
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/coresight/dap.py", line 255, in connect
    self._idr = self.read_idr()
                ^^^^^^^^^^^^^^^
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/coresight/dap.py", line 284, in read_idr
    dpidr = self._dp.read_dp(DP_IDR, now=True)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/coresight/dap.py", line 792, in read_dp
    return read_dp_cb()
           ^^^^^^^^^^^^
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/coresight/dap.py", line 780, in read_dp_cb
    result = result_cb()
             ^^^^^^^^^^^
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/probe/cmsis_dap_probe.py", line 546, in read_dp_result_callback
    raise self._convert_exception(error) from error
pyocd.core.exceptions.TransferError: Unexpected ACK value (5) returned by probe
```

prosper00, 2023-03-31:
**[DAPLink - clear](https://www.aliexpress.com/item/1005003056314140.html)**

```
[ +10.503617] usb 3-1.2: new full-speed USB device number 78 using xhci_hcd
[  +0.146366] usb 3-1.2: New USB device found, idVendor=c251, idProduct=f001, bcdDevice= 1.00
[  +0.000005] usb 3-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  +0.000003] usb 3-1.2: Product: CMSIS_DAP
[  +0.000001] usb 3-1.2: Manufacturer: CMSIS-DAP by ARM
[  +0.000002] usb 3-1.2: SerialNumber: 202108120001
[  +0.124053] cdc_acm 3-1.2:1.0: ttyACM1: USB ACM device
[  +0.014147] hid-generic 0003:C251:F001.00AF: hiddev2,hidraw5: USB HID v1.00 Device [CMSIS-DAP by ARM CMSIS_DAP] on usb-0000:05:00.3-1.2/input2
```
```
$ pyocd list
  #   Probe/Board                  Unique ID      Target  
----------------------------------------------------------
  0   CMSIS-DAP by ARM CMSIS_DAP   202108120001   n/a     
```
**STM32**
```
$ pyocd erase --chip -t stm32f103c8; pyocd load -t stm32f103c8 ./miniblink.bin 
0001030 I Erasing chip... [eraser]
0001270 I Chip erase complete [eraser]
0001030 I Loading /home/bdroy/STM32-base/libopencm3-examples/examples/stm32/f1/stm32-maple/miniblink/miniblink.bin [load_cmd]
[==================================================] 100%
0001612 I Erased 1024 bytes (1 sector), programmed 1024 bytes (1 page), skipped 0 bytes (0 pages) at 1.72 kB/s [loader]
```
**PY32**
```
$ pyocd load ./Build/example-blink.hex -t py32f002x5 --config ./Misc/pyocd.yaml
0000857 C Unexpected ACK value (5) returned by probe [__main__]
Traceback (most recent call last):
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/probe/cmsis_dap_probe.py", line 541, in read_dp_result_callback
    value = result()
            ^^^^^^^^
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 1071, in read_reg_cb
    res = transfer.get_result()
          ^^^^^^^^^^^^^^^^^^^^^
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 157, in get_result
    self.daplink.flush()
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/utility/concurrency.py", line 29, in _locking
    return func(self, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 881, in flush
    self._read_packet()
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/utility/concurrency.py", line 29, in _locking
    return func(self, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 1157, in _read_packet
    decoded_data = cmd.decode_data(raw_data)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 478, in decode_data
    data = self._decode_transfer_block_data(data)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 447, in _decode_transfer_block_data
    self._check_response(data[3])
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 370, in _check_response
    raise DAPAccessIntf.TransferError("Unexpected ACK value (%d) returned by probe" % ack)
pyocd.probe.pydapaccess.dap_access_api.DAPAccessIntf.TransferError: Unexpected ACK value (5) returned by probe

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/__main__.py", line 161, in run
    status = cmd.invoke()
             ^^^^^^^^^^^^
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/subcommands/load_cmd.py", line 98, in invoke
    with session:
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/core/session.py", line 391, in __enter__
    self.open()
[... 4 lines cut ...]
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/core/soc_target.py", line 157, in init
    seq.invoke()
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/utility/sequencer.py", line 213, in invoke
    resultSequence.invoke()
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/utility/sequencer.py", line 208, in invoke
    resultSequence = call()
                     ^^^^^^
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/coresight/dap.py", line 446, in _connect
    connector.connect()
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/coresight/dap.py", line 255, in connect
    self._idr = self.read_idr()
                ^^^^^^^^^^^^^^^
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/coresight/dap.py", line 284, in read_idr
    dpidr = self._dp.read_dp(DP_IDR, now=True)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/coresight/dap.py", line 792, in read_dp
    return read_dp_cb()
           ^^^^^^^^^^^^
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/coresight/dap.py", line 780, in read_dp_cb
    result = result_cb()
             ^^^^^^^^^^^
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/probe/cmsis_dap_probe.py", line 546, in read_dp_result_callback
    raise self._convert_exception(error) from error
pyocd.core.exceptions.TransferError: Unexpected ACK value (5) returned by probe
```

prosper00, 2023-03-31:
**[WCHLink](https://www.aliexpress.com/item/1005003059548220.html)**

```
[Mar30 21:51] usb 3-1.1: new full-speed USB device number 79 using xhci_hcd
[  +0.152978] usb 3-1.1: New USB device found, idVendor=2a86, idProduct=8011, bcdDevice= 1.00
[  +0.000006] usb 3-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  +0.000002] usb 3-1.1: Product: WCH-Link
[  +0.000002] usb 3-1.1: Manufacturer: wch.cn
[  +0.000001] usb 3-1.1: SerialNumber: 0001A0000001
[  +0.125069] cdc_acm 3-1.1:1.0: ttyACM1: USB ACM device
[  +0.017109] hid-generic 0003:2A86:8011.00BF: hiddev2,hidraw5: USB HID v1.00 Device [wch.cn WCH-Link] on usb-0000:05:00.3-1.1/input2
```
```
$ pyocd list
  #   Probe/Board       Unique ID      Target  
-----------------------------------------------
  0   wch.cn WCH-Link   0001A0000001   n/a  
```
**STM32**
```
$ pyocd erase --chip -t stm32f103c8; pyocd load -t stm32f103c8 ./miniblink.bin 
0000972 C Unexpected ACK value (5) returned by probe [__main__]
Traceback (most recent call last):
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/probe/cmsis_dap_probe.py", line 541, in read_dp_result_callback
    value = result()
            ^^^^^^^^
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 1071, in read_reg_cb
    res = transfer.get_result()
          ^^^^^^^^^^^^^^^^^^^^^
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 157, in get_result
    self.daplink.flush()
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/utility/concurrency.py", line 29, in _locking
    return func(self, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 881, in flush
    self._read_packet()
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/utility/concurrency.py", line 29, in _locking
    return func(self, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 1157, in _read_packet
    decoded_data = cmd.decode_data(raw_data)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 478, in decode_data
    data = self._decode_transfer_block_data(data)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 447, in _decode_transfer_block_data
    self._check_response(data[3])
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 370, in _check_response
    raise DAPAccessIntf.TransferError("Unexpected ACK value (%d) returned by probe" % ack)
pyocd.probe.pydapaccess.dap_access_api.DAPAccessIntf.TransferError: Unexpected ACK value (5) returned by probe

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/__main__.py", line 161, in run
    status = cmd.invoke()
             ^^^^^^^^^^^^
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/subcommands/erase_cmd.py", line 92, in invoke
    with session:
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/core/session.py", line 391, in __enter__
    self.open()
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/core/session.py", line 529, in open
    self._board.init()
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/board/board.py", line 139, in init
    self.target.init()
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/core/soc_target.py", line 157, in init
    seq.invoke()
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/utility/sequencer.py", line 213, in invoke
    resultSequence.invoke()
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/utility/sequencer.py", line 208, in invoke
[... 135 lines cut ...]
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/core/soc_target.py", line 157, in init
    seq.invoke()
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/utility/sequencer.py", line 213, in invoke
    resultSequence.invoke()
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/utility/sequencer.py", line 208, in invoke
    resultSequence = call()
                     ^^^^^^
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/coresight/dap.py", line 446, in _connect
    connector.connect()
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/coresight/dap.py", line 255, in connect
    self._idr = self.read_idr()
                ^^^^^^^^^^^^^^^
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/coresight/dap.py", line 284, in read_idr
    dpidr = self._dp.read_dp(DP_IDR, now=True)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
[... cut ...] p
    return read_dp_cb()
           ^^^^^^^^^^^^
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/coresight/dap.py", line 780, in read_dp_cb
    result = result_cb()
             ^^^^^^^^^^^
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/probe/cmsis_dap_probe.py", line 546, in read_dp_result_callback
    raise self._convert_exception(error) from error
pyocd.core.exceptions.TransferError: Unexpected ACK value (5) returned by probe
```

prosper00, 2023-03-31:
**[Wagiminator – CH552-picoDAP](https://github.com/wagiminator/CH552-picoDAP)**

```
[Mar30 22:01] usb 3-2.2: new full-speed USB device number 83 using xhci_hcd
[  +0.143651] usb 3-2.2: New USB device found, idVendor=16c0, idProduct=05df, bcdDevice= 1.00
[  +0.000007] usb 3-2.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  +0.000002] usb 3-2.2: Product: picoDAP CMSIS-DAP
[  +0.000002] usb 3-2.2: Manufacturer: wagiminator
[  +0.000001] usb 3-2.2: SerialNumber: CH55x
[  +0.084198] hid-generic 0003:16C0:05DF.00C8: hiddev2,hidraw5: USB HID v1.10 Device [wagiminator picoDAP CMSIS-DAP] on usb-0000:05:00.3-2.2/input0
```
```
$ pyocd list
  #   Probe/Board                     Unique ID   Target  
----------------------------------------------------------
  0   wagiminator picoDAP CMSIS-DAP   CH55x       n/a     
```
**STM32**
```
$ pyocd erase --chip -t stm32f103c8; pyocd load -t stm32f103c8 ./miniblink.bin 
0001003 I Erasing chip... [eraser]
Exception in thread Thread-2 (rx_task):
Traceback (most recent call last):
  File "/usr/lib/python3.11/threading.py", line 1038, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.11/threading.py", line 975, in run
    self._target(*self._args, **self._kwargs)
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/probe/pydapaccess/interface/pyusb_backend.py", line 151, in rx_task
    read_data = self.ep_in.read(self.ep_in.wMaxPacketSize,
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/usb/core.py", line 423, in read
    return self.device.read(self, size_or_buffer, timeout)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/usb/core.py", line 1029, in read
    ret = fn(
          ^^^
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/usb/backend/libusb1.py", line 864, in intr_read
    return self.__read(self.lib.libusb_interrupt_transfer,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/usb/backend/libusb1.py", line 954, in __read
    _check(retval)
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/usb/backend/libusb1.py", line 602, in _check
    raise USBTimeoutError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBTimeoutError: [Errno 110] Operation timed out
0011150 E Error during board uninit: [session]
Traceback (most recent call last):
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 1157, in _read_packet
    decoded_data = cmd.decode_data(raw_data)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 478, in decode_data
    data = self._decode_transfer_block_data(data)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 444, in _decode_transfer_block_data
    raise DAPAccessIntf.TransferError(f'DAP_TRANSFER_BLOCK response error: response is for command {data[0]:02x}')
pyocd.probe.pydapaccess.dap_access_api.DAPAccessIntf.TransferError: DAP_TRANSFER_BLOCK response error: response is for command 05

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/probe/cmsis_dap_probe.py", line 595, in read_ap_result_callback
    value = result()
            ^^^^^^^^
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 1071, in read_reg_cb
    res = transfer.get_result()
          ^^^^^^^^^^^^^^^^^^^^^
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 154, in get_result
    self.daplink._read_packet()
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/utility/concurrency.py", line 29, in _locking
    return func(self, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
[... 1040 lines cut ...]
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/coresight/cortex_m.py", line 952, in is_halted
    return self.get_state() == Target.State.HALTED
           ^^^^^^^^^^^^^^^^
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/coresight/cortex_m.py", line 918, in get_state
    dhcsr = self.read_memory(CortexM.DHCSR)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/coresight/cortex_m.py", line 481, in read_memory
    result = self.ap.read_memory(addr, transfer_size, now)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/utility/concurrency.py", line 29, in _locking
    return func(self, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/coresight/ap.py", line 1084, in _read_memory
    result = read_mem_cb()
             ^^^^^^^^^^^^^
[... cut ...] 2, in read_mem_cb
    res = result_cb()
          ^^^^^^^^^^^
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/coresight/dap.py", line 892, in read_ap_cb
    result = result_cb()
             ^^^^^^^^^^^
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/probe/cmsis_dap_probe.py", line 600, in read_ap_result_callback
    raise self._convert_exception(error) from error
pyocd.core.exceptions.ProbeError: Timeout reading from device CH55x
```

prosper00, 2023-03-31:
**[STLinkV2 – baite](https://www.aliexpress.com/item/1005005002268447.html)**

```
[Mar30 21:55] usb 3-1.2: new full-speed USB device number 81 using xhci_hcd
[  +0.146989] usb 3-1.2: New USB device found, idVendor=0483, idProduct=3748, bcdDevice= 1.00
[  +0.000005] usb 3-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  +0.000003] usb 3-1.2: Product: STM32 STLink
[  +0.000001] usb 3-1.2: Manufacturer: STMicroelectronics
[  +0.000002] usb 3-1.2: SerialNumber: h
```
```
$ pyocd list
  #   Probe/Board    Unique ID                  Target  
--------------------------------------------------------
  0   STM32 STLink   68005500110000383530524E   n/a    
```
**STM32**
```
$ pyocd erase --chip -t stm32f103c8; pyocd load -t stm32f103c8 ./miniblink.bin 
0000882 C STLink error (9): Get IDCODE error [__main__]
Traceback (most recent call last):
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/__main__.py", line 161, in run
    status = cmd.invoke()
             ^^^^^^^^^^^^
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/subcommands/erase_cmd.py", line 92, in invoke
    with session:
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/core/session.py", line 391, in __enter__
    self.open()
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/core/session.py", line 529, in open
    self._board.init()
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/board/board.py", line 139, in init
    self.target.init()
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/core/soc_target.py", line 157, in init
    seq.invoke()
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/utility/sequencer.py", line 213, in invoke
    resultSequence.invoke()
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/utility/sequencer.py", line 208, in invoke
    resultSequence = call()
                     ^^^^^^
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/coresight/dap.py", line 442, in _connect
    probe_conn.connect(self._protocol)

  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/coresight/dap.py", line 172, in connect
    self._connect_probe(protocol)
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/coresight/dap.py", line 189, in _connect_probe
    self._probe.connect(protocol)
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/probe/stlink_probe.py", line 182, in connect
    self._link.enter_debug(STLink.Protocol.SWD)
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/probe/stlink/stlink.py", line 327, in enter_debug
    self._check_status(response)
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/probe/stlink/stlink.py", line 365, in _check_status
    raise exceptions.ProbeError(error_message)
pyocd.core.exceptions.ProbeError: STLink error (9): Get IDCODE error
0000888 C STLink error (9): Get IDCODE error [__main__]
Traceback (most recent call last):
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/__main__.py", line 161, in run
    status = cmd.invoke()
             ^^^^^^^^^^^^
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/subcommands/load_cmd.py", line 98, in invoke
    with session:
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/core/session.py", line 391, in __enter__
    self.open()
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/core/session.py", line 529, in open
    self._board.init()
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/board/board.py", line 139, in init
    self.target.init()
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/core/soc_target.py", line 157, in init
    seq.invoke()
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/utility/sequencer.py", line 213, in invoke
    resultSequence.invoke()
[... 29 lines cut ...]
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/core/session.py", line 529, in open
    self._board.init()
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/board/board.py", line 139, in init
    self.target.init()
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/core/soc_target.py", line 157, in init
    seq.invoke()
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/utility/sequencer.py", line 213, in invoke
    resultSequence.invoke()
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/utility/sequencer.py", line 208, in invoke
    resultSequence = call()
                     ^^^^^^
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/coresight/dap.py", line 442, in _connect
    probe_conn.connect(self._protocol)
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/coresight/dap.py", line 172, in connect
    self._connect_probe(protocol)
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/coresight/dap.py", line 189, in _connect_probe
    self._probe.connect(protocol)
[... cut ...] /stlink_probe.py", line 182, in connect
    self._link.enter_debug(STLink.Protocol.SWD)
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/probe/stlink/stlink.py", line 327, in enter_debug
    self._check_status(response)
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/probe/stlink/stlink.py", line 365, in _check_status
    raise exceptions.ProbeError(error_message)
pyocd.core.exceptions.ProbeError: STLink error (9): Get IDCODE error
```

prosper00, 2023-03-31:
[STLinkV2 - metal](https://www.aliexpress.com/item/1005001724949176.html)

```
[Mar30 21:58] usb 3-1.2: new full-speed USB device number 82 using xhci_hcd
[  +0.146746] usb 3-1.2: New USB device found, idVendor=0483, idProduct=3748, bcdDevice= 1.00
[  +0.000006] usb 3-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  +0.000002] usb 3-1.2: Product: STM32 STLink
[  +0.000002] usb 3-1.2: Manufacturer: STMicroelectronics
[  +0.000001] usb 3-1.2: SerialNumber: \xc2\x85m\x01\x01\x10\x13GS8LN
```
```
$ pyocd list
  #   Probe/Board    Unique ID                  Target  
--------------------------------------------------------
  0   STM32 STLink   856D010110134753384C4E00   n/a  
```
**STM32**
```
$ pyocd erase --chip -t stm32f103c8; pyocd load -t stm32f103c8 ./miniblink.bin 
0000910 I Erasing chip... [eraser]
0001086 I Chip erase complete [eraser]
0000885 I Loading /home/bdroy/STM32-base/libopencm3-examples/examples/stm32/f1/stm32-maple/miniblink/miniblink.bin [load_cmd]
[==================================================] 100%
0001286 I Erased 1024 bytes (1 sector), programmed 1024 bytes (1 page), skipped 0 bytes (0 pages) at 2.49 kB/s [loader]
```
**PY32**
```
$ pyocd load ./Build/example-blink.hex -t py32f002x5 --config ./Misc/pyocd.yaml
0000670 C STLink error (9): Get IDCODE error [__main__]
Traceback (most recent call last):
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/__main__.py", line 161, in run
    status = cmd.invoke()
             ^^^^^^^^^^^^
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/subcommands/load_cmd.py", line 98, in invoke
    with session:
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/core/session.py", line 391, in __enter__
    self.open()
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/core/session.py", line 529, in open
    self._board.init()
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/board/board.py", line 139, in init
    self.target.init()
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/core/soc_target.py", line 157, in init
    seq.invoke()
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/utility/sequencer.py", line 213, in invoke
    resultSequence.invoke()
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/utility/sequencer.py", line 208, in invoke
    resultSequence = call()
                     ^^^^^^
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/coresight/dap.py", line 442, in _connect
    probe_conn.connect(self._protocol)
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/coresight/dap.py", line 172, in connect
    self._connect_probe(protocol)
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/coresight/dap.py", line 189, in _connect_probe
    self._probe.connect(protocol)
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/probe/stlink_probe.py", line 182, in connect
    self._link.enter_debug(STLink.Protocol.SWD)
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/probe/stlink/stlink.py", line 327, in enter_debug
    self._check_status(response)
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/probe/stlink/stlink.py", line 365, in _check_status
    raise exceptions.ProbeError(error_message)
pyocd.core.exceptions.ProbeError: STLink error (9): Get IDCODE error
```

prosper00, 2023-03-31:
**[JlinkOB – baite](https://www.aliexpress.com/item/1005004902159370.html)**

```
[Mar30 22:06] usb 3-1.2: new full-speed USB device number 84 using xhci_hcd
[  +0.152411] usb 3-1.2: New USB device found, idVendor=1366, idProduct=0101, bcdDevice= 1.00
[  +0.000005] usb 3-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  +0.000003] usb 3-1.2: Product: J-Link
[  +0.000001] usb 3-1.2: Manufacturer: SEGGER
[  +0.000002] usb 3-1.2: SerialNumber: 000000123456
```
```
$ pyocd list
  #   Probe/Board               Unique ID   Target  
----------------------------------------------------
  0   Segger J-Link (unknown)   123456      n/a 
```
**STM32**
```
$ pyocd erase --chip -t stm32f103c8; pyocd load -t stm32f103c8 ./miniblink.bin 
0001397 I Erasing chip... [eraser]
0001636 I Chip erase complete [eraser]
0001375 I Loading /home/bdroy/STM32-base/libopencm3-examples/examples/stm32/f1/stm32-maple/miniblink/miniblink.bin [load_cmd]
[==================================================] 100%
0001998 I Erased 1024 bytes (1 sector), programmed 1024 bytes (1 page), skipped 0 bytes (0 pages) at 1.61 kB/s [loader]
```
**PY32**
```
$ pyocd load ./Build/example-blink.hex -t py32f002x5 --config ./Misc/pyocd.yaml
0001146 C Memory transfer fault (Unspecified error.) [__main__]
Traceback (most recent call last):
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/probe/jlink_probe.py", line 407, in read_dp
    value = self._link.coresight_read(addr // 4, ap=False)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pylink/jlink.py", line 204, in wrapper
    return func(self, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pylink/jlink.py", line 3388, in coresight_read
    raise errors.JLinkException(res)
pylink.errors.JLinkException: Unspecified error.

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/coresight/dap.py", line 766, in read_dp
    result_cb = self.probe.read_dp(addr & DPADDR_MASK, now=False)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/probe/jlink_probe.py", line 409, in read_dp
    raise self._convert_exception(exc) from exc
pyocd.core.exceptions.TransferFaultError: Memory transfer fault (Unspecified error.)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/probe/jlink_probe.py", line 418, in write_dp
    self._link.coresight_write(addr // 4, data, ap=False)
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pylink/jlink.py", line 204, in wrapper
    return func(self, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pylink/jlink.py", line 3415, in coresight_write
    raise errors.JLinkException(res)
pylink.errors.JLinkException: Unspecified error.

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/__main__.py", line 161, in run
    status = cmd.invoke()
             ^^^^^^^^^^^^
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/subcommands/load_cmd.py", line 98, in invoke
    with session:
[... 10 lines cut ...]
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/utility/sequencer.py", line 208, in invoke
    resultSequence = call()
                     ^^^^^^
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/coresight/dap.py", line 446, in _connect
    connector.connect()
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/coresight/dap.py", line 255, in connect
    self._idr = self.read_idr()
                ^^^^^^^^^^^^^^^
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/coresight/dap.py", line 284, in read_idr
    dpidr = self._dp.read_dp(DP_IDR, now=True)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/coresight/dap.py", line 768, in read_dp
    self._handle_error(error, num)
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/coresight/dap.py", line 982, in _handle_error
    self.clear_sticky_err()
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/coresight/dap.py", line 993, in clear_sticky_err
    self.write_reg(DP_ABORT, ABORT_ORUNERRCLR | ABORT_WDERRCLR | ABORT_STKERRCLR | ABORT_STKCMPCLR)
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/coresight/dap.py", line 514, in write_reg
    self.write_dp(addr, data)
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/coresight/dap.py", line 808, in write_dp
    self.probe.write_dp(addr & DPADDR_MASK, data)
  File "/home/bdroy/.local/pipx/venvs/pyocd/lib/python3.11/site-packages/pyocd/probe/jlink_probe.py", line 420, in write_dp
    raise self._convert_exception(exc) from exc
pyocd.core.exceptions.TransferFaultError: Memory transfer fault (Unspecified error.)
```

prosper00, 2023-05-01:
Something I just discovered: programming works better at 3V3 Vcc. At least some of the probes that weren't working for me actually DO work, but only when running the PY32 at 3.3V. And all probes seem to work with more stability and reliability at 3.3V

danyhm, 2023-08-05:
@prosper00 I'm having the exact same problem. which programmers did you use to connect?
the vendor mentions a py-link programer which seems like a modified jlink but I couldn't find it anywhere.

I have a st-linkvs but I can't connect to it using neither stlink nor dap link.

prosper00, 2023-08-05:
> @prosper00 I'm having the exact same problem. which programmers did you use to connect?
> the vendor mentions a py-link programer which seems like a modified jlink but I couldn't find it anywhere.
> 
> I have a st-linkvs but I can't connect to it using neither stlink nor dap link.

any of the jlinks generally work, but only when the target is on 3.3V

danyhm, 2023-08-05:
@prosper00  would a clone Jlink OB work as well? like this one
https://sinkrtech.com/products/sinkr-arm-programmer-compatible-j-link-ob-replace-jjlink-v8-sdw-emulation-debugger-arm-programmer-download-cable-supports-mdk-iar-keilno-jtag-interface

prosper00, 2023-08-05:
probably, yeah

Bingo600, 2023-08-20:
I have used this board with success: Air32F103CBT6 
https://www.aliexpress.com/item/1005004635091721.html

With the air32_daplink_v1.hex , from the below  
https://gitee.com/openLuat/luatos-soc-air32f103/tree/master/AIR_Jlink_Keil

I couldn't get the "new" v2 firmware to work.
https://www.eevblog.com/forum/microcontrollers/air105-air32f103-new-chinese-arm-chips/msg5014411/#msg5014411

 I programmed the Air32 w. a jlink , specifying taget as : STM32F103CB

jonnor, 2024-04-07:
I was attempted to flash a Puya PY32F003 on the official development board (PY32F003F18P EVB) with a DAPLink (no brand on it), and got `Unexpected ACK value (5)`. In my case, this was fixed when I plugged in the USB-C. In addition to having 3.3V connected on the DAPLink connector (which I thought would be enough?).
Just thought I would mention it, in case it can save someone else some time :) It was a rather unintuitive error for this case...

grahamwhaley, 2024-12-18:
Just a couple of thoughts, that might not relate directly to the original case but might help somebody else in the future.

I'm just bringing up a board (it is actually the board out of a vape, which had the SWD connections available...) which has a PY32C624F15 QFP20 on it. I'm using a PiPico as a DAP dongle running the [free-dap open firmware](https://github.com/ataradov/free-dap).

I had no joy connecting to the core until I did a couple of other things:
  - I added in the 10k pullups and 22r inline impedance matching resistors as shown on the [Arm SWD wiring docs](https://developer.arm.com/documentation/dui0499/k/arm-dstream-target-interface-connections/signal-descriptions/serial-wire-debug?lang=en)
  - I programmed down the clockspeed on the pyocd command line with `-Ofrequency=1000000` for instance
  - oddly, the chip does not seem to respond until after I have tried to connect using [edbg](https://github.com/ataradov/edbg) - which itself fails, but then pyocd works... I'm presuming `edbg` does something to the SWD that then enables `pyocd` to connect - I'll have to figure out exactly what that is later if I get around to adding the seemingly rare PY32C624 support to pyocd etc. I'm currently treating the chip as a PY32F002B, which is very close in terms of specifications (flash, ram etc.).

I can now read the flash, ram and options area contents.

sherazmurtaza56, 2026-08-21:
I am facing an issue while trying to program the PY32F002AW15U MCU.Initially, I attempted to use a Clone ST-Link, but it did not work.
I then arranged an official PY-Link and installed the PuyaProgrammer v1.0.0.8 software.Currently:The laptop detects the PY-Link in Device Manager.The PuyaProgrammer software also recognizes the PY-Link device.However, the firmware version is not displayed in the software, and I am unable to proceed further.I then downloaded the PY-Link upGrade tool from the official website to upgrade the PY-Link firmware first, so that it can connect to the MCU. Unfortunately, the software freezes every time during this process.
I am also unsure which firmware version is currently loaded on the PY-Link and what the correct procedure is to upgrade it.Additional Query:
On the PY-Link, there is a push button. When I long-press it, the LED color changes as follows:
Green → Yellow → Pink → Green
However, I could not find any documentation explaining what these LED colors indicate.Could you please guide me on:The correct firmware upgrade procedure for PY-LinkHow to check the current firmware versionThe meaning of the LED color indications

can anyone help me out this problem