pyocd/pyOCD #1532: pyocd flash fails on FRDM-K66F

JanneKiiskila, 2023-04-06:
Flashing of a .bin-file (which fits just fine to memory) fails on FRDM-K66F.

```

(310-pyocd)  14:25 jannek@jannek-P720:/ssd/mbed-cloud-client-example-internal$ pyocd --version
0.34.3
(310-pyocd)  14:26 jannek@jannek-P720:/ssd/mbed-cloud-client-example-internal$ pyocd flash /ssd/mbed-cloud-client-example-internal/BUILD/K66F/GCC_ARM/mbed-cloud-client-example-internal.bin
0000741 I Loading /ssd/mbed-cloud-client-example-internal/BUILD/K66F/GCC_ARM/mbed-cloud-client-example-internal.bin [load_cmd]
0004900 W Timed out waiting for core to halt after reset (state is RESET) [cortex_m]
0004930 C cannot write registers pc, r0, r1, r2, r9, sp, lr because core #0 is not halted [__main__]
Traceback (most recent call last):
  File "/home/jannek/venv/310-pyocd/lib/python3.10/site-packages/pyocd/__main__.py", line 161, in run
    status = cmd.invoke()
  File "/home/jannek/venv/310-pyocd/lib/python3.10/site-packages/pyocd/subcommands/load_cmd.py", line 125, in invoke
    programmer.program(filename,
  File "/home/jannek/venv/310-pyocd/lib/python3.10/site-packages/pyocd/flash/file_programmer.py", line 170, in program
    self._loader.commit()
  File "/home/jannek/venv/310-pyocd/lib/python3.10/site-packages/pyocd/flash/loader.py", line 289, in commit
    perf = builder.program(chip_erase=chipErase,
  File "/home/jannek/venv/310-pyocd/lib/python3.10/site-packages/pyocd/flash/builder.py", line 490, in program
    sector_erase_count, page_program_time = self._compute_sector_erase_pages_and_weight(fast_verify)
  File "/home/jannek/venv/310-pyocd/lib/python3.10/site-packages/pyocd/flash/builder.py", line 666, in _compute_sector_erase_pages_and_weight
    self._analyze_pages_with_crc32(fast_verify)
  File "/home/jannek/venv/310-pyocd/lib/python3.10/site-packages/pyocd/flash/builder.py", line 645, in _analyze_pages_with_crc32
    self._enable_read_access()
  File "/home/jannek/venv/310-pyocd/lib/python3.10/site-packages/pyocd/flash/builder.py", line 259, in _enable_read_access
    self.flash.init(self.flash.Operation.VERIFY)
  File "/home/jannek/venv/310-pyocd/lib/python3.10/site-packages/pyocd/flash/flash.py", line 252, in init
    result = self._call_function_and_wait(self.flash_algo['pc_init'],
  File "/home/jannek/venv/310-pyocd/lib/python3.10/site-packages/pyocd/flash/flash.py", line 653, in _call_function_and_wait
    self._call_function(pc, r0, r1, r2, r3, init)
  File "/home/jannek/venv/310-pyocd/lib/python3.10/site-packages/pyocd/flash/flash.py", line 555, in _call_function
    self.target.write_core_registers_raw(reg_list, data_list)
  File "/home/jannek/venv/310-pyocd/lib/python3.10/site-packages/pyocd/core/soc_target.py", line 235, in write_core_registers_raw
    self.selected_core_or_raise.write_core_registers_raw(reg_list, data_list)
  File "/home/jannek/venv/310-pyocd/lib/python3.10/site-packages/pyocd/coresight/cortex_m.py", line 1182, in write_core_registers_raw
    self._base_write_core_registers_raw(reg_list, data_list)
  File "/home/jannek/venv/310-pyocd/lib/python3.10/site-packages/pyocd/coresight/cortex_m.py", line 1196, in _base_write_core_registers_raw
    raise exceptions.CoreRegisterAccessError(
pyocd.core.exceptions.CoreRegisterAccessError: cannot write registers pc, r0, r1, r2, r9, sp, lr because core #0 is not halted
```

Python version info:
```
(310-pyocd)  jannek@jannek-P720:/ssd/mbed-cloud-client-example-internal$ python --version
Python 3.10.6

(310-pyocd)  jannek@jannek-P720:/ssd/mbed-cloud-client-example-internal$ pip list
Package            Version
------------------ --------
appdirs            1.4.4
capstone           4.0.2
cffi               1.15.1
cmsis-pack-manager 0.5.2
colorama           0.4.6
future             0.18.3
intelhex           2.3.0
intervaltree       3.1.0
libusb-package     1.0.26.1
natsort            8.3.1
pip                23.0.1
prettytable        3.6.0
psutil             5.9.4
pycparser          2.21
pyelftools         0.29
pylink-square      0.14.3
pyocd              0.34.3
pyusb              1.2.1
PyYAML             6.0
setuptools         67.2.0
six                1.16.0
sortedcontainers   2.4.0
typing_extensions  4.5.0
wcwidth            0.2.6
wheel              0.38.4

```

At the same time, I can flash the device just fine via `cp <bin-file> /media/$USER/FRDM-K66F && sync`, so device is not in an abnormal state. Seeems pyocd can't get it to the right state for flashing.

FRDM-K66F has Daplink, details below.

```
# DAPLink Firmware - see https://daplink.io
Build ID: v0257-gc782a5ba (gcc)
Unique ID: 03110b0148824e45000b700bdd89003c8761000097969900
HIC ID: 97969900
Auto Reset: 0
Automation allowed: 1
Overflow detection: 1
Incompatible image detection: 0
Page erasing: 1
Daplink Mode: Interface
Interface Version: 0257
Bootloader Version: 0244
Git SHA: c782a5ba907377658bc28aa8d132a0fa44543687
Local Mods: 0
USB Interfaces: MSD, CDC, HID, WebUSB
Bootloader CRC: 0xa870af25
Interface CRC: 0x7ed91f5c
Remount count: 1
URL: http://www.nxp.com/frdm-k66f
```

riggs, 2023-04-07:
Try adding `-M attach` to your command:
``` 
connection:
...
  -M MODE, --connect MODE
                        Select connect mode from one of (halt, pre-reset, under-reset, attach).
```

flit, 2023-04-08:
This could be related to the device being locked. I think I've noticed an issue recently with the auto-unlocking being troublesome. I've done some work on it, but haven't fixed it yet, and haven't had a chance to get back to it.