pyocd/pyOCD #1545: debug sequences: NXP S32K3xx DFP attach failure

flit, 2023-05-02:
Created from discussion in #1488.

From @dev-apcs:

The snag I have is when using gdb to attach without flashing: the cpu hard-faults.

I tracked this down to vector table corruption - or rather annihilation:

The sample being used moves the vector table to RAM
There's a RAMInitialize sequence in the DFP that, well, clears RAM (under control of the debug variable DoRAMInitialize)
This is all well and good if the core is returned to reset state before running the sequence
With the present implementation, this reset appears not to happen so VTOR is not re-initialised hence the hard fault
Setting DoRAMInitialize to 0 (initially directly in the .pdsc, before I discovered -Opack.debug_sequences.debugvars="DoRAMInitialize=0;") works around the issue.
When I finally RTFM, I notice that some standard reset sequences are not yet supported.

I haven't yet done the analysis to identify the missing sequence but this would appear to be the cause.

I figure that a user script might be the easiest fix for now. Do you have a roadmap for the remaining DFP sequences?

P.S.

I may have spoke too soon. The reset is possibly the responsibility of the client. The CLion run configuration has an option to send a reset command. However, setting this to monitor reset (always) doesn't appear to make a difference.

flit, 2023-05-02:
The way `ResetHardwareDeassert` is used in the S32K3xx pack… that's an annoying repurposing of those sequences. ☹️ I'll take a look at how difficult the `ResetHardware*` sequences to support.

When you say "attach without flash", does that mean attaching to running code without reset (in which case I don't understand the issue…), or running existing code from reset?

dev-apcs, 2023-05-03:
Thanks for breaking this issue out  - something I ought to have done.

My expectation was based on previous experience with CLion and OpenOCD:
- CLion run configuration has an option to download (flash in this case) to the target `Always`, `Update only` or `Never`
    - [Documented here](https://www.jetbrains.com/help/clion/openocd-support.html#opecocd-debug)
    - There's also an opportunity to specify the type of reset required (`Run`, `Init`, `Halt` or `None`)
- I tend to leave it `Update only` and `Init` (this is my "attach without flash" option)

The configuration for pyOCD is more generic:
- `Embedded GDB Server` configuration with (defaults):
    - `Download executable` set to `If updated`
    - `Advanded GDB Server Options` -> `Reset command` set to:
        - `monitor reset`
        - `After download`
    - [Documented here](https://www.jetbrains.com/help/clion/embedded-gdb-server.html#tools)

Hmm, CLion has moved on (finally).  There's now a `New Embedded GDB Server Run Configuration` wizard with a `pyOCD` option:
- Following the wizard
  - There's a `Reset Strategy` dropdown offering:
    - `Default`
    - `Halt`
    - `Pre-Reset`
    - `Init-Under-Reset`
    - `Attach`
- Selecting `Pre-Reset`
  - Adds `--connect pre-reset` to the GDB (pyOCD) command line arguments
- The `Reset command` is set to:
  - `monitor reset`
  - `After download`

Adding `-Opack.debug_sequences.debugvars="DoRAMInitialize=0;` to the command line arguments gets me to the same point as my manual configuration.

I was surprised to see S32K3xx listed in the supported targets list.  A nice touch, caching the DFPs.

Yes, "attach" was a poor choice of words, and the reason I back-tracked in my previous post.  It suggests connecting to a running target - which is a valid option but again broken without disabling `RAMInitialize`.

flit, 2023-05-05:
Cool! Nice to see CLion has pyOCD support now. 😄 

Glad you have a workaround for now. With that, I'll probably go ahead and release a new version with the debug sequence support, then start working on improvements such as `ResetHardware*` support.

For the RTOS issues:
- What version of FreeRTOS are you using? v10.x (and v8-M, though you don't need it) are supported on the `feature/freertos_10` branch. It hasn't been merged yet because I haven't had a chance to fully test with FreeRTOS v9 for backwards compatibility (even though that branch has been around for a long time…).
- For Zephyr, I've been using it myself a lot, but only up to v2.7. There are other people that use it with 3.x, afaik.  What are the issues you see?

The missing stack trace is probably caused by `LR` being 0xffffffff. This could possibly be caused by incompatibility with the FreeRTOS version.

I just rebased the `feature/freertos_10` branch onto the latest `develop` with debug sequence support. It's worth a shot.

dev-apcs, 2023-05-05:
Yeah, sounds good.

The lwip_s32k344 sample uses FreeRTOS V10.4.6.

I appear to be having environmental issues (or there're issues with CLion).  Launching pyocd as an `Embedded GDB Server` and attaching GDB all in one run configuration, I get the issues described.  Launching pyocd independently (from pycharm) and then launching GDB (CLion `Remote Debug` run configuration), normal service is restored: all threads listed with the ability to switch between threads and up/down the call stack - all through CLion.  Unfortunately, the more advanced RTOS debugging features (`FreeRTOS Objects` and `FreeRTOS Heap`) don't feature in `Remote Debug`.  These advanced features do look to be working in the "broken" case (`Embedded GDB Server`.  I have some more analysis to do, clearly.

Moving to feature/freertos_10 and running server and client separately breaks FreeRTOS integration: the current thread is intact with the call stack navigable but moving between threads, the call stack is gibberish - which looks like perhaps version compatibility.  I'll concentrate on my environment woes before diving deeper into this.  I need to better understand what both code bases are doing to provide FreeRTOS integration.

The Zephyr issues are more fundamental.  I can set breakpoints early in the init cycle (around z_arm_reset) but it never reaches main.  I need to consider all the usual suspects + I didn't fully follow the instructions (well, you never learn anything unless it goes wrong...).  I'm using the Zephyr toolchain rather than NXP's and I'm seeing unsupported pragma warnings during build (around applying section attributes it seems).  I think they've modified GCC to support a different syntax - reminiscent of AUTOSAR which is likely to be the SoC's primary market.  If I have any sense, I'll create another build environment with the NXP's toolchain for comparison purposes...:blush:

flit, 2023-05-09:
Hmm, well, let me know if it appears to be a pyocd issue!

I'd like to revive the `feature/freertos_10` branch soon, and add automated functional tests for different FreeRTOS versions. Sorry it doesn't seem to work, but thanks for trying.

From having lots of experience debugging Zephyr linking issues (on odd toolchain variants for research), if you're getting warnings related to sections then it's probably not going to be functional. Zephyr is highly dependent on sections be assigned properly to get a functional executable. Sometimes it will link fine, and the errors only happen at runtime. If it's not reaching main, then one likely source is the kernel + device init list sections being incorrect.

dev-apcs, 2023-05-10:
Thanks for the support!

I'm off on another job atm but expect to get back to this shortly.

I haven't gotten any further with the freertos problem.  I need to sanity-check my setup before drawing any conclusions.  Presumably you're only interested in testing the the touch-points for the functional tests?  That ought to reduce the effort required.

Zephyr: from memory (a week's a long time):
- I attributed the "fundamental" problems to clock setup
  - Just for information (again, off-topic):
    - The lwip_s32k344 (freertos) sample uses an earlier "HAL" version that the Zephyr HAL
    - The newer version attempts to configure the `MSCM` IP block but appears to fail
- The section placement issue is distinct from the Zephyr placement
  - The "HAL" is actually drawn from the `RTD` (Real-Time Drivers) which appear to be shared with the AUTOSAR `MCAL`
    - AUTOSAR's hardware abstraction is at the `MCU` layer (MCuAbstractionLayer)
    - AUTOSAR provides a fine-grained section placement infrastructure (using `MemMap` header files - a bit clunky, but then it is `C`)
    - This seems mostly to be a convenience thing but can cause problems with critical placements like boot headers (if written in `C`)
    - From a cursory glance, the placement rounds up the `MCAL` code into a number of sections (text, data, bss) but the loader file places them into flash and ram so I'm not expecting this to put up much of a fight (famous last words...)
- What I initially thought was Zephyr support issues (the failure was different with and without Zephyr support enabled in CLion) I now think was just different symptoms of the same problem.

**Anyhow, back on topic**:

The Keil pack `.pdsc` appears to have been written against an earlier silicon version.  Some of the fields have moved, particularly around `debug from first instruction` (some fields appear to have moved from `MDM-AP` to `SDA-AP` for instance).  I'd like to straighten this out but it probably makes more sense to report the issue to the author - which I'll do once I've completed the analysis.  Perhaps the issue could include a nudge towards not re-purposing?

flit, 2023-05-10:
I'll create an internal Arm bug ticket against the Keil pack to hopefully get it updated to the latest silicon revision. And I'll definitely mention the need to not repurpose standard sequences. 😄 

FreeRTOS: Pretty much. Ideally the functional test would get 100% code coverage on the FreeRTOS support, with several FreeRTOS versions and CPU architecture variants. But anything would be better than the current state.

Zephyr: got it, thanks for the explanation

jonlamb-gh, 2024-07-28:
@flit 
Hi there. I'm looking into getting S32K3xx support added to [probe-rs](https://probe.rs/), and came across this thread while searching around for references related to the debug sequences.

> I'll create an internal Arm bug ticket against the Keil pack to hopefully get it updated to the latest silicon revision. And I'll definitely mention the need to not repurpose standard sequences.

Looks like the Keil pack has since been deprecated, did you get anywhere with this?

korken89, 2025-05-23:
@jonlamb-gh did you ever get it working? I'm looking now to add support to probe-rs for the S32K3xx, maybe we should pool our knowledge?