pyocd/pyOCD #1547: Flashing i.MX RT685EVK

IvoBCD, 2023-05-09:
I'm trying to program the on-board flash of the iMX RT685EVK, but it seems pyOCD doesn't like the flash algorithm provided by NXP:

Using the following `pyocd_user.py`:

```python
session.options['debug.log_flm_info'] = True
session.options['allow_no_cores'] = True

def will_connect():
    extFlash = FlashRegion(
               name="flexspi",
               start=0x18000000,
               length=0x800000,
               is_boot_memory=True,
               blocksize=0x1000,
               page_size=0x100,
               flm = "MIMXRT600_FlexSPI_B_SFDP_QSPI.cfx"
            )
    target.memory_map.add_region(extFlash)
```

I get an "Unexpected token Token('IDENT', 'U')" error:

 ```console 
# pyocd flash --pack /opt/sample-dut-fw/NXP.MIMXRT685S_DFP.16.0.0.pack -t mimxrt685s /opt/sample-dut-fw/blinky-mimxrt685_evk_cm33-3.3.0.elf
0007787 W CMSIS-Pack device MIMXRT685S has no identifiable boot memory [cmsis_pack]
0007974 C Unexpected token Token('IDENT', 'U') at line 3, column 34.
Expected one of: 
	* QMARK
	* RPAR
	* COMMA

Context:         Write32(0x40004130, 0x130U);
                                 ^
 [__main__]
 #
 ```
 
 ```console
 $ readelf -l MIMXRT600_FlexSPI_B_SFDP_QSPI.cfx

Elf file type is EXEC (Executable file)
Entry point 0x1002009d
There is 1 program header, starting at offset 52

Program Headers:
  Type           Offset   VirtAddr   PhysAddr   FileSiz MemSiz  Flg Align
  LOAD           0x000054 0x10020000 0x10020000 0x006c0 0x0a6d8 RWE 0x4

 Section to Segment mapping:
  Segment Sections...
   00     .cr_data .text .data .stackcache 
$
```

I get a similar error when using the MIMXRT6XX_EVK_QSPI.FLM included in Segger's toolchain.

flit, 2023-05-09:
Thanks for the bug report. The "Unexpected token" error is a the debug sequence in the pack. I did test with the RT685, so I wonder if something changed or I just missed something. I'll try to reproduce and hopefully have a fix.

flit, 2023-05-09:
Sorry, forgot it's actually the RT595 that I've tested with. I don't even have an RT685. Still, they're quite similar.

In any case, the issue with the debug sequence is NXP's use of `U` suffix on integer literals (in a `ResetFlash` sequence not present in the RT595 pack). This isn't specified as allowable in the Open-CMSIS-Pack specification… it apparently just happens to work in Keil MDK.

I'll add in support for this and let you know when the pull request is ready for you to test.

Assuming that works, you will probably have issues with the .cfx flash algorithm. If I remember right, this is NXP's proprietary and not publicly documented flash algorithm format used by MCUXpresso. The best option is to extract the .flm from the `NXP.MIMXRT685S_DFP.16.0.0.pack` (you can [download from here](https://www.keil.com/dd2/pack/)); just extract it as a zip and the algorithm is under the `arm/` directory. (While Segger claims CMSIS-Pack flash algorithm compatibility, the algorithms they provide are incompatible from a loading and runtime perspective.)

flit, 2023-05-09:
If you have a chance, please try out the fix in #1548. You can install with `python3 -m pip install --pre -U git+https://github.com/flit/pyOCD.git@bugfix/dbgseq_unsigned_int_lit`.

IvoBCD, 2023-05-10:
@flit With that change, I no longer get the parse error.  It does seem like more is needed to actually get it to flash though.

```console
+ cat pyocd_user.py
session.options['debug.log_flm_info'] = True
# session.options['allow_no_cores'] = True

def will_connect():
    extFlash = FlashRegion(
               name="flexspi",
               start=0x18000000,
               length=0x800000,
               is_boot_memory=True,
               blocksize=0x1000,
               page_size=0x100,
               flm = "MIMXRT6XX_EVK_FLEXSPI.FLM",
               #flm = "MIMXRT6XX_EVK_FLEXSPI_S.FLM",
               #flm = "MIMXRT6XX_FLEXSPI_B_SFDP_QSPI.FLM",
               #flm = "MIMXRT6XX_FLEXSPI_B_SFDP_QSPI_S.FLM",
            )
    target.memory_map.add_region(extFlash)
+ pyocd flash --pack NXP.MIMXRT685S_DFP.16.0.0.pack -t mimxrt685s blinky-mimxrt685_evk_cm33-3.3.0.elf
0007829 W CMSIS-Pack device MIMXRT685S has no identifiable boot memory [cmsis_pack]
0010549 W Skipping CoreSight discovery for AHB-AP#0 because it is disabled [ap]
0010599 C No cores were discovered! [__main__]
```

IvoBCD, 2023-05-10:
trace logs:
```
+ cat pyocd_user.py
session.options['debug.log_flm_info'] = True
session.options['pack.debug_sequences.enable'] = True
# session.options['allow_no_cores'] = True

def will_connect():
    extFlash = FlashRegion(
               name="flexspi",
               start=0x18000000,
               length=0x800000,
               is_boot_memory=True,
               blocksize=0x1000,
               page_size=0x100,
               #flm = "MIMXRT600_FlexSPI_B_SFDP_QSPI.cfx",
               flm = "MIMXRT6XX_EVK_FLEXSPI.FLM",
               #flm = "MIMXRT6XX_EVK_FLEXSPI_S.FLM",
               #flm = "MIMXRT6XX_FLEXSPI_B_SFDP_QSPI.FLM",
               #flm = "MIMXRT6XX_FLEXSPI_B_SFDP_QSPI_S.FLM",
            )
    target.memory_map.add_region(extFlash)
+ pyocd flash -L '*.trace=debug' --pack NXP.MIMXRT685S_DFP.16.0.0.pack -t mimxrt685s blinky-mimxrt685_evk_cm33-3.3.0.elf
0004426 D [cmd:0] New _Command [dap_access_cmsis_dap]
0007679 D Not sending notification because no subscribers: event=pack [notification]
0007681 D Not sending notification because no subscribers: event=target_override [notification]
0007686 D Not sending notification because no subscribers: event=debug.log_flm_info [notification]
0007712 D   USB OUT> (2) 00 fe [pyusb_v2_backend]
0007715 D   USB IN < (3) 00 01 02 [pyusb_v2_backend]
0007716 D   USB RD < (3) 00 01 02 [pyusb_v2_backend]
0007716 D   USB OUT> (2) 00 04 [pyusb_v2_backend]
0007718 D   USB IN < (8) 00 06 32 2e 30 2e 30 00 [pyusb_v2_backend]
0007718 D   USB RD < (8) 00 06 32 2e 30 2e 30 00 [pyusb_v2_backend]
0007719 D   USB OUT> (2) 00 ff [pyusb_v2_backend]
0007720 D   USB IN < (4) 00 02 40 00 [pyusb_v2_backend]
0007721 D   USB RD < (4) 00 02 40 00 [pyusb_v2_backend]
0007722 D   USB OUT> (2) 00 f0 [pyusb_v2_backend]
0007723 D   USB IN < (3) 00 01 31 [pyusb_v2_backend]
0007723 D   USB RD < (3) 00 01 31 [pyusb_v2_backend]
0007724 D [cmd:1] New _Command [dap_access_cmsis_dap]
0007726 D [cmd:2] New _Command [dap_access_cmsis_dap]
0007811 W CMSIS-Pack device MIMXRT685S has no identifiable boot memory [cmsis_pack]
0008122 D trace: open [cmsis_dap_probe]
0008146 D [cmd:3] New _Command [dap_access_cmsis_dap]
0008147 D trace: set_clock(freq=1000000) [cmsis_dap_probe]
0008147 D   USB OUT> (5) 11 40 42 0f 00 [pyusb_v2_backend]
0008150 D   USB IN < (2) 11 00 [pyusb_v2_backend]
0008151 D   USB RD < (2) 11 00 [pyusb_v2_backend]
0010113 D trace: connect(DEFAULT) [cmsis_dap_probe]
0010130 D   USB OUT> (2) 02 01 [pyusb_v2_backend]
0010143 D   USB IN < (2) 02 01 [pyusb_v2_backend]
0010144 D   USB RD < (2) 02 01 [pyusb_v2_backend]
0010150 D   USB OUT> (5) 11 40 42 0f 00 [pyusb_v2_backend]
0010167 D   USB IN < (2) 11 00 [pyusb_v2_backend]
0010173 D   USB RD < (2) 11 00 [pyusb_v2_backend]
0010179 D   USB OUT> (6) 04 02 50 00 00 00 [pyusb_v2_backend]
0010196 D   USB IN < (2) 04 00 [pyusb_v2_backend]
0010202 D   USB RD < (2) 04 00 [pyusb_v2_backend]
0010208 D   USB OUT> (2) 13 00 [pyusb_v2_backend]
0010225 D   USB IN < (2) 13 00 [pyusb_v2_backend]
0010231 D   USB RD < (2) 13 00 [pyusb_v2_backend]
0010237 D trace: swj_sequence(length=51, bits=ffffffffffffff) [cmsis_dap_probe]
0010242 D   USB OUT> (9) 12 33 ff ff ff ff ff ff ff [pyusb_v2_backend]
0010262 D   USB IN < (2) 12 00 [pyusb_v2_backend]
0010268 D   USB RD < (2) 12 00 [pyusb_v2_backend]
0010273 D trace: swj_sequence(length=16, bits=e79e) [cmsis_dap_probe]
0010279 D   USB OUT> (4) 12 10 9e e7 [pyusb_v2_backend]
0010296 D   USB IN < (2) 12 00 [pyusb_v2_backend]
0010302 D   USB RD < (2) 12 00 [pyusb_v2_backend]
0010308 D trace: swj_sequence(length=51, bits=ffffffffffffff) [cmsis_dap_probe]
[... 286 lines cut ...]
0012260 D write_dp:000041 (addr=0x00000008) = 0x00000000 [dap]
0012265 D [cmd:19] get_request_space(1, 08:w)[wc=2, rc=0, ba=1->0] -> (sz=1, free=10, delta=-252) [dap_access_cmsis_dap]
0012271 D [cmd:19] add(1, 08:w) -> [wc=3, rc=0, ba=0] [dap_access_cmsis_dap]
0012277 D trace: write_dp(addr=0x00000008, data=0x00000000) [cmsis_dap_probe]
0012282 D trace: read_ap(addr=0x00000000) -> ... [cmsis_dap_probe]
0012288 D [cmd:19] get_request_space(1, 03:r)[wc=3, rc=0, ba=0->0] -> (sz=1, free=15, delta=-251) [dap_access_cmsis_dap]
0012294 D [cmd:19] add(1, 03:r) -> [wc=3, rc=1, ba=0] [dap_access_cmsis_dap]
0012299 D flush: sending cmd:19; reading 0 outstanding [dap_access_cmsis_dap]
0012305 D [cmd:19] _send_packet: sending [dap_access_cmsis_dap]
0012311 D   USB OUT> (19) 05 00 04 00 1e 00 00 00 00 1e 00 00 00 08 00 00 00 00 03 [pyusb_v2_backend]
0012322 D [cmd:20] New _Command [dap_access_cmsis_dap]
0012328 D   USB IN < (7) 05 04 01 12 00 00 03 [pyusb_v2_backend]
0012329 D [cmd:19] _read_packet: reading [dap_access_cmsis_dap]
0012360 D   USB RD < (7) 05 04 01 12 00 00 03 [pyusb_v2_backend]
0012366 D trace: ... read_ap(addr=0x00000000) -> 0x03000012 [cmsis_dap_probe]
0012372 D read_ap:000040 (addr=0x00000000) -> 0x03000012 [dap]
0012377 W Skipping CoreSight discovery for AHB-AP#0 because it is disabled [ap]
0012383 D trace: disconnect [cmsis_dap_probe]
0012389 D   USB OUT> (1) 03 [pyusb_v2_backend]
0012411 D   USB IN < (2) 03 00 [pyusb_v2_backend]
0012417 D   USB RD < (2) 03 00 [pyusb_v2_backend]
0012423 D trace: close [cmsis_dap_probe]
0012460 D [cmd:21] New _Command [dap_access_cmsis_dap]
0012466 C No cores were discovered! [__main__]
```

flit, 2023-05-10:
Thanks for testing.

This next issue is probably due to the way the the i.MX RT (and LPC55xx) families require a special sequence to unlock debug access, which is handled by a debug sequence in the CMSIS pack. It doesn't appear this sequence is being executed, which explains why the Access Port #0 for the CM33 core is disabled.

Looking at the pack, this might be a result of a missing feature in pyocd's debug sequence support. The `DebugPortStart` sequence is used by the RT685 pack to unlock debug, but this sequence is assigned to the `cm33` core (cores can be given names in packs). Unfortunately, pyocd doesn't support core-specific `DebugPort*` sequences yet…

I'll work on adding this support into pyocd.

Short term, you could edit the `NXP.MIMXRT685S_DFP.pdsc` file inside the pack. In version 16.0.0 of the pack, line 104 of this file looks like:

```
          <sequence name="DebugPortStart" Pname="cm33">
```

Just remove the `Pname="cm33"` part. (There are other sequences also assigned to `cm33`, but they shouldn't matter. It's only `DebugPortStart` that is a problem.)

You don't need to re-zip the pack; pyocd can use an expanded pack directory with the `--pack` argument.

closed 2023-05-10 by flit