pyocd/pyOCD #1554: target mimxrt685s (NXP i.MX RT685): SocTarget has no selected core

IvoBCD, 2023-05-11:
Still trying to program the on-board flash of the NXP i.MX RT685EVK through pyOCD; I get an "SoCTarget has no selected core" error.

(NXP.MIMXRT685S_DFP.pdsc modified, removing `Pname="cm33"`)

```console
+ set -e
+ grep DebugPortStart mypack/NXP.MIMXRT685S_DFP.pdsc
          <sequence name="DebugPortStart">
+ cat pyocd_user.py
session.options['debug.log_flm_info'] = True
session.options['allow_no_cores'] = True

def will_connect():
    extFlash = FlashRegion(
               name="flexspi",
               start=0x18000000,
               length=0x800000,
               is_boot_memory=True,
               blocksize=0x1000,
               page_size=0x100,
               flm = "MIMXRT6XX_EVK_FLEXSPI_S.FLM",
            )
    target.memory_map.add_region(extFlash)
+ pyocd flash -v --pack mypack -e chip -t mimxrt685s blinky-mimxrt685_evk_cm33-3.3.0.elf
0010728 W CMSIS-Pack device MIMXRT685S has no identifiable boot memory [cmsis_pack]
0010978 I Target type is mimxrt685s [board]
0014782 I DP IDR = 0x6ba02477 (v2 rev6) [dap]
0014857 I debugvar 'BootTime' = 0x2710 (10000) [pack_target]
0014863 I debugvar 'Dbg_CR' = 0x0 (0) [pack_target]
0014869 I debugvar 'SWO_Pin' = 0x0 (0) [pack_target]
0016292 I AHB-AP#0 IDR = 0x84770001 (AHB-AP var0 rev8) [discovery]
0016388 I AP#2 IDR = 0x002a0000 (AP var0 rev0) [discovery]
0016808 W Skipping CoreSight discovery for AHB-AP#0 because it is disabled [ap]
0016814 E No cores were discovered! [coresight_target]
0016825 I Creating flash algo for region flexspi from: /home/root/nxp/MIMXRT6XX_EVK_FLEXSPI_S.FLM [flm_region_builder]
0017028 I Loading /home/root/nxp/blinky-mimxrt685_evk_cm33-3.3.0.elf [load_cmd]
[                                                  ]   0%0017455 C Error: 'SoCTarget has no selected core' [__main__]
```

with more logging detail:
```
0017231 D flash algo: [stack=0x1f850; 0x1f850 b] [b2=0x1f950,+0x1f950] [b1=0x1f850,+0x1f850] [code=0x1fa5c,+0x1fa5c,0x5a4 b] (ram=0x00000000, 0x20000 b) [flash_algo]
0017237 D Running task notify [sequencer]
0017255 I Loading /home/root/nxp/blinky-mimxrt685_evk_cm33-3.3.0.elf [load_cmd]
0017302 D Skipping segment LMA:0x18005598, VMA:0x18005598, size 8 [file_programmer]
0017308 D Writing segment LMA:0x18000000, VMA:0x18000000, size 4912 [file_programmer]
0017325 D Writing segment LMA:0x18001330, VMA:0x30180000, size 256 [file_programmer]
0017332 D Writing segment LMA:0x18001430, VMA:0x18001430, size 18364 [file_programmer]
0017349 D Writing segment LMA:0x18005bec, VMA:0x30180100, size 2386 [file_programmer]
0017365 D Writing segment LMA:0x1800653e, VMA:0x1800653e, size 4 [file_programmer]
0017372 D Skipping segment LMA:0x30180a58, VMA:0x30180a58, size 0 [file_programmer]
0017413 D Using double buffer chip erase program [builder]
0017418 D 0 of 102 pages have erased data [builder]
[                                                  ]   0%0017435 D uninit session <pyocd.core.session.Session object at 0x7fbba39810> [session]
0017441 D uninit board <pyocd.board.board.Board object at 0x7fb9c134d0> [board]
0017831 C Error: 'SoCTarget has no selected core' [__main__]
Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/pyocd/__main__.py", line 161, in run
    status = cmd.invoke()
             ^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/pyocd/subcommands/load_cmd.py", line 130, in invoke
    programmer.program(filename,
  File "/usr/lib/python3.11/site-packages/pyocd/flash/file_programmer.py", line 175, in program
    self._loader.commit()
  File "/usr/lib/python3.11/site-packages/pyocd/flash/loader.py", line 295, in commit
    perf = builder.program(chip_erase=chipErase,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
```

flit, 2023-05-13:
This error is because pyocd hasn't been able to successfully find an accessible CPU core. It needs a CPU to run the flash algorithm, and when it tries to access the default core (abstracted by the `SoCTarget` class's notion of a currently selected core) it, naturally, fails.

I'll keep this issue open for the time being, but at this point it looks like a side effect of other debug sequence issues. For some reason, debug is not being properly unlocked, thus the "Skipping CoreSight discovery for AHB-AP#0 because it is disabled" warning.

IvoBCD, 2023-05-15:
I've had some USB host stack issues affecting my tests, with those sorted out and updating to latest PyOCD, the core does seem to be detected, and programming fails with `target was not halted as expected after calling flash algorithm routine (IPSR=3)`.

```console
# pyocd flash --pack ./NXP.MIMXRT685S_DFP.16.0.0 -t mimxrt685s blinky-mimxrt685_evk_cm33-3.3.0.elf
0007439 W CMSIS-Pack device MIMXRT685S has no identifiable boot memory [cmsis_pack]
0011302 I Loading /home/root/mimxrt685evk/blinky-mimxrt685_evk_cm33-3.3.0.elf [load_cmd]
0012667 C target was not halted as expected after calling flash algorithm routine (IPSR=3) [__main__] 
```

Detailed log:
```console
# pyocd --version
0.35.0.dev+gda303b04
+ pyocd list
  #   Probe/Board                                     Unique ID   Target  
--------------------------------------------------------------------------
  0   NXP Semiconductors LPC-LINK2 CMSIS-DAP V5.361   NRAQBQHR    n/a     
# grep DebugPortStart ./NXP.MIMXRT685S_DFP.16.0.0/NXP.MIMXRT685S_DFP.pdsc
	  <sequence name="DebugPortStart"> <!-- [removed for PyOCD] Pname="cm33" -->
# cat pyocd_user.py
session.options['debug.log_flm_info'] = True
session.options['allow_no_cores'] = True
def will_connect():
    extFlash = FlashRegion(
               name="flexspi",
               start=0x18000000,
               length=0x800000,
               is_boot_memory=True,
               blocksize=0x1000,
               page_size=0x100,
               flm = "MIMXRT6XX_EVK_FLEXSPI_S.FLM",
            )
    target.memory_map.add_region(extFlash)
# pyocd flash -v -vv -Lpyocd.debug.sequences.sequences.trace=debug --pack ./NXP.MIMXRT685S_DFP.16.0.0 -t mimxrt685s blinky-mimxrt685_evk_cm33-3.3.0.elf
0004286 D setting log level debug for ['pyocd.debug.sequences.sequences.trace'] [__main__]
0004430 D Project directory: /home/root/mimxrt685evk [session]
0007379 D Project directory: /home/root/mimxrt685evk [session]
0007381 D Loading user script: /home/root/mimxrt685evk/pyocd_user.py [session]
0007423 D Detaching Kernel Driver of Interface 0 from USB device (VID=1fc9 PID=0090). [pyusb_backend]
0007438 D CMSIS-DAP v1 probe NRAQBQHR: protocol version 1.1.0 [dap_access_cmsis_dap]
0007446 D closing interface [pyusb_backend]
0007520 D MIMXRT685S_DFP DFP (MIMXRT685S): not loading non-default flash algorithm 'arm/MIMXRT6XX_FLEXSPI_B_SFDP_QSPI.FLM' [cmsis_pack]
0007521 D MIMXRT685S_DFP DFP (MIMXRT685S): not loading non-default flash algorithm 'arm/MIMXRT6XX_FLEXSPI_B_SFDP_QSPI_S.FLM' [cmsis_pack]
0007522 W CMSIS-Pack device MIMXRT685S has no identifiable boot memory [cmsis_pack]
0007775 I Target type is mimxrt685s [board]
0007814 D Detaching Kernel Driver of Interface 0 from USB device (VID=1fc9 PID=0090). [pyusb_backend]
0007828 D Running task load_svd [sequencer]
0009791 D Running task pre_connect [sequencer]
0009808 D Running task dp_init [sequencer]
0009814 D Running task lock_probe [sequencer]
0009820 D Running task get_probe_capabilities [sequencer]
0009826 D Running task connect [sequencer]
0009941 D Default wire protocol selected; using SWD [dap]
0009962 D Sending deprecated SWJ sequence to select SWD [swj]
0010046 I DP IDR = 0x6ba02477 (v2 rev6) [dap]
0010051 D Running task clear_sticky_err [sequencer]
0010057 D Running task power_up_debug [sequencer]
0010063 D Running debug sequence 'DebugPortStart' [pack_target]
0010073 D (line 3): decl SWO_Pin = 0x0 [sequences]
0010085 D (line 4): decl Dbg_CR = 0x0 [sequences]
0010090 D (line 5): decl BootTime = 0x2710 [sequences]
0010096 I debugvar 'BootTime' = 0x2710 (10000) [pack_target]
0010102 I debugvar 'Dbg_CR' = 0x0 (0) [pack_target]
0010107 I debugvar 'SWO_Pin' = 0x0 (0) [pack_target]
0010116 D (line 2): decl SW_DP_ABORT = 0x0 [sequences]
0010121 D (line 3): decl DP_CTRL_STAT = 0x4 [sequences]
0010127 D (line 4): decl DP_SELECT = 0x8 [sequences]
0010133 D (line 5): decl powered_down = 0x0 [sequences]
0010139 D (line 7): fn writedp (DP_SELECT{0x8}, 0x0) ... [sequences]
0010160 D (line 7): fn writedp () returned 0x0 [sequences]
[... 263 lines cut ...]
    programmer.program(filename,
  File "/usr/lib/python3.11/site-packages/pyocd/flash/file_programmer.py", line 175, in program
    self._loader.commit()
  File "/usr/lib/python3.11/site-packages/pyocd/flash/loader.py", line 295, in commit
    perf = builder.program(chip_erase=chipErase,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/pyocd/flash/builder.py", line 492, in program
    sector_erase_count, page_program_time = self._compute_sector_erase_pages_and_weight(fast_verify)
                                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/pyocd/flash/builder.py", line 673, in _compute_sector_erase_pages_and_weight
    self._analyze_pages_with_partial_read()
  File "/usr/lib/python3.11/site-packages/pyocd/flash/builder.py", line 608, in _analyze_pages_with_partial_read
    self._enable_read_access()
  File "/usr/lib/python3.11/site-packages/pyocd/flash/builder.py", line 262, in _enable_read_access
    self.flash.init(self.flash.Operation.ERASE)
  File "/usr/lib/python3.11/site-packages/pyocd/flash/flash.py", line 253, in init
    result = self._call_function_and_wait(self.flash_algo['pc_init'],
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/pyocd/flash/flash.py", line 676, in _call_function_and_wait
    return self.wait_for_completion(timeout=timeout)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/pyocd/flash/flash.py", line 663, in wait_for_completion
    raise exceptions.FlashFailure("target was not halted as expected after calling "
pyocd.core.exceptions.FlashFailure: target was not halted as expected after calling flash algorithm routine (IPSR=3)
```

IvoBCD, 2023-05-15:
Same outcome, but different debug logs, when using the unmodified pack file:

```console
# pyocd flash -v -vv -Lpyocd.debug.sequences.sequences.trace=debug --pack ./NXP.MIMXRT685S_DFP.16.0.0.pack -t mimxrt685s blinky-mimxrt685_evk_cm33-3.3.0.elf
0004299 D setting log level debug for ['pyocd.debug.sequences.sequences.trace'] [__main__]
0004443 D Project directory: /home/root/mimxrt685evk [session]
0007363 D Project directory: /home/root/mimxrt685evk [session]
0007366 D Loading user script: /home/root/mimxrt685evk/pyocd_user.py [session]
0007407 D Detaching Kernel Driver of Interface 0 from USB device (VID=1fc9 PID=0090). [pyusb_backend]
0007424 D CMSIS-DAP v1 probe NRAQBQHR: protocol version 1.1.0 [dap_access_cmsis_dap]
0007431 D closing interface [pyusb_backend]
0007545 D MIMXRT685S_DFP DFP (MIMXRT685S): not loading non-default flash algorithm 'arm/MIMXRT6XX_FLEXSPI_B_SFDP_QSPI.FLM' [cmsis_pack]
0007546 D MIMXRT685S_DFP DFP (MIMXRT685S): not loading non-default flash algorithm 'arm/MIMXRT6XX_FLEXSPI_B_SFDP_QSPI_S.FLM' [cmsis_pack]
0007546 W CMSIS-Pack device MIMXRT685S has no identifiable boot memory [cmsis_pack]
0007851 I Target type is mimxrt685s [board]
0007891 D Detaching Kernel Driver of Interface 0 from USB device (VID=1fc9 PID=0090). [pyusb_backend]
0007904 D Running task load_svd [sequencer]
0009847 D Running task pre_connect [sequencer]
0009864 D Running task dp_init [sequencer]
0009870 D Running task lock_probe [sequencer]
0009877 D Running task get_probe_capabilities [sequencer]
0009883 D Running task connect [sequencer]
0009975 D Default wire protocol selected; using SWD [dap]
0010010 D Sending deprecated SWJ sequence to select SWD [swj]
0010129 I DP IDR = 0x6ba02477 (v2 rev6) [dap]
0010135 D Running task clear_sticky_err [sequencer]
0010141 D Running task power_up_debug [sequencer]
0010168 D Running task check_version [sequencer]
0010173 D Running task unlock_probe [sequencer]
0010179 D Running task unlock_device [sequencer]
0010185 D Running task create_discoverer [sequencer]
0010191 D Running task discovery [sequencer]
0010196 D Running task find_aps [sequencer]
0010380 D Running task create_aps [sequencer]
0010386 D Running task create_ap.0 [sequencer]
0010454 D AHB-AP#0 default HPROT=3 HNONSEC=0 [ap]
0010474 D AHB-AP#0 implemented HPROT=f HNONSEC=1 [ap]
0010496 I AHB-AP#0 IDR = 0x84770001 (AHB-AP var0 rev8) [discovery]
0010502 D Running task create_ap.2 [sequencer]
0010522 I AP#2 IDR = 0x002a0000 (AP var0 rev0) [discovery]
0010528 D Running task create_ap.3 [sequencer]
0010624 D APB-AP#3 default HPROT=0 HNONSEC=0 [ap]
0010645 D APB-AP#3 implemented HPROT=0 HNONSEC=0 [ap]
0010686 I APB-AP#3 IDR = 0x54770002 (APB-AP var0 rev5) [discovery]
0010692 D Running task find_components [sequencer]
0010698 D Running task init_ap.0 [sequencer]
0010790 I AHB-AP#0 Class 0x1 ROM table #0 @ 0xe00fe000 (designer=015 part=000) [rom_table]
0010841 I [0]<e00ff000:ROM class=1 designer=43b:Arm part=4c9> [rom_table]
0010847 I   AHB-AP#0 Class 0x1 ROM table #1 @ 0xe00ff000 (designer=43b:Arm part=4c9) [rom_table]
0010933 I   [0]<e000e000:SCS M33 class=9 designer=43b:Arm part=d21 devtype=00 archid=2a04 devid=0:0:0> [rom_table]
0010999 I   [1]<e0001000:DWT M33 class=9 designer=43b:Arm part=d21 devtype=00 archid=1a02 devid=0:0:0> [rom_table]
0011035 I   [2]<e0002000:BPU M33 class=9 designer=43b:Arm part=d21 devtype=00 archid=1a03 devid=0:0:0> [rom_table]
0011071 I   [3]<e0000000:ITM M33 class=9 designer=43b:Arm part=d21 devtype=43 archid=1a01 devid=0:0:0> [rom_table]
0011076 D   [4]<fff41002 not present> [rom_table]
0011112 I   [5]<e0041000:ETM M33 class=9 designer=43b:Arm part=d21 devtype=13 archid=4a13 devid=0:0:0> [rom_table]
0011148 I   [6]<e0042000:CTI M33 class=9 designer=43b:Arm part=d21 devtype=14 archid=1a14 devid=40800:0:0> [rom_table]
0011153 D   [7]<fff44002 not present> [rom_table]
0011204 I [1]<e0040000:TPIU M33 class=9 designer=43b:Arm part=d21 devtype=11 archid=0000 devid=ca1:0:0> [rom_table]
0011210 D [2]<1ff02002 not present> [rom_table]
0011215 D [3]<1ff02002 not present> [rom_table]
0011221 D Running task create_cores [sequencer]
0011227 D Creating SCS component [discovery]
0011257 D selected core #0 [soc_target]
0011309 I CPU core #0 is Cortex-M33 r0p3 (security ext present) [cortex_m_v8m]
0011359 I FPU present: FPv5-SP-D16-M [cortex_m]
0011422 D Running task configure_core_reset [sequencer]
0011428 D updated DFP core #0 reset types: {<ResetType.HW: 1>, <ResetType.SW_EMULATED: 5>, <ResetType.SW_SYSTEM: 3>, <ResetType.SW: 2>} [pack_target]
0011433 I Setting core #0 (cm33) default reset sequence to ResetSystem [pack_target]
0011449 D Running task create_components [sequencer]
0011455 D Creating DWT component [discovery]
[... 188 lines cut ...]
    programmer.program(filename,
  File "/usr/lib/python3.11/site-packages/pyocd/flash/file_programmer.py", line 175, in program
    self._loader.commit()
  File "/usr/lib/python3.11/site-packages/pyocd/flash/loader.py", line 295, in commit
    perf = builder.program(chip_erase=chipErase,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/pyocd/flash/builder.py", line 492, in program
    sector_erase_count, page_program_time = self._compute_sector_erase_pages_and_weight(fast_verify)
                                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/pyocd/flash/builder.py", line 673, in _compute_sector_erase_pages_and_weight
[... cut ...]
  File "/usr/lib/python3.11/site-packages/pyocd/flash/flash.py", line 676, in _call_function_and_wait
    return self.wait_for_completion(timeout=timeout)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/pyocd/flash/flash.py", line 663, in wait_for_completion
    raise exceptions.FlashFailure("target was not halted as expected after calling "
pyocd.core.exceptions.FlashFailure: target was not halted as expected after calling flash algorithm routine (IPSR=3)
```

IvoBCD, 2023-05-15:
What does work, is using PyOCD (without CMSIS pack) with NXP's SPSDK, to put the i.MX RT685 in serial download mode:

```console
# nxpdebugmbox -i pyocd -v -vv ispmode -m 0
  #   Interface   Id         Description                                    
----------------------------------------------------------------------------
  0   PyOCD       NRAQBQHR   NXP Semiconductors LPC-LINK2 CMSIS-DAP V5.361  
DEBUG:spsdk.debuggers.debug_probe_pyocd:The SPSDK PyOCD Interface has been initialized (10701ms since start, debug_probe_pyocd.py:66)
DEBUG:pyocd.core.session:Project directory: /home/root
DEBUG:pyocd.core.session:Project directory: /home/root
DEBUG:pyocd.probe.pydapaccess.interface.pyusb_backend:Detaching Kernel Driver of Interface 0 from USB device (VID=1fc9 PID=0090).
DEBUG:pyocd.probe.pydapaccess.dap_access_cmsis_dap:CMSIS-DAP v1 probe NRAQBQHR: protocol version 1.1.0
DEBUG:pyocd.probe.pydapaccess.interface.pyusb_backend:closing interface
DEBUG:pyocd.coresight.coresight_target:Using default Cortex-M memory map (no memory map supplied)
DEBUG:pyocd.probe.pydapaccess.interface.pyusb_backend:Detaching Kernel Driver of Interface 0 from USB device (VID=1fc9 PID=0090).
DEBUG:pyocd.probe.swj:Sending deprecated SWJ sequence to select SWD
DEBUG:spsdk.debuggers.debug_probe_pyocd:DPIDR(idr=1805657207, partno=186, version=2, revision=6, mindp=False) (10924ms since start, debug_probe_pyocd.py:119)
DEBUG:spsdk.debuggers.debug_probe_pyocd:Coresight read DP, address: 00000004, data: F0000F40 (10928ms since start, debug_probe_pyocd.py:190)
DEBUG:spsdk.debuggers.debug_probe:Checked Sticky Errors: 0xf0000f40 (10928ms since start, debug_probe.py:283)
DEBUG:spsdk.debuggers.debug_probe:Power up the debug connection (10929ms since start, debug_probe.py:361)
DEBUG:spsdk.debuggers.debug_probe:Power Control the debug connection:
System power: True
Debug power: True (10929ms since start, debug_probe.py:334)
DEBUG:spsdk.debuggers.debug_probe_pyocd:Coresight write DP, address: 00000004, data: 50000F00 (10930ms since start, debug_probe_pyocd.py:220)
DEBUG:spsdk.debuggers.debug_probe_pyocd:Coresight read DP, address: 00000004, data: F0000F40 (10934ms since start, debug_probe_pyocd.py:190)
INFO:spsdk.debuggers.debug_probe_pyocd:PyOCD connected via LPC-LINK2 CMSIS-DAP V5.361 probe.
DEBUG:spsdk.dat.debug_mailbox:Reset mode: True (10935ms since start, debug_mailbox.py:66)
DEBUG:spsdk.debuggers.debug_probe_pyocd:Coresight write DP, address: 00000008, data: 000000F0 (10936ms since start, debug_probe_pyocd.py:220)
DEBUG:spsdk.debuggers.debug_probe:Selected AP: 0, Bank: 0xf (10936ms since start, debug_probe.py:382)
DEBUG:spsdk.debuggers.debug_probe_pyocd:Coresight read AP, address: 0000000C, data: 84770001 (10940ms since start, debug_probe_pyocd.py:190)
DEBUG:spsdk.debuggers.debug_probe_pyocd:Coresight write DP, address: 00000008, data: 020000F0 (10941ms since start, debug_probe_pyocd.py:220)
DEBUG:spsdk.debuggers.debug_probe:Selected AP: 2, Bank: 0xf (10941ms since start, debug_probe.py:382)
DEBUG:spsdk.debuggers.debug_probe_pyocd:Coresight read AP, address: 0000000C, data: 002A0000 (10945ms since start, debug_probe_pyocd.py:190)
DEBUG:spsdk.dat.debug_mailbox:Found debug mailbox access port at AP2, IDR: 0x002A0000 (10945ms since start, debug_mailbox.py:190)
DEBUG:spsdk.debuggers.debug_probe_pyocd:Coresight write DP, address: 00000008, data: 02000000 (10946ms since start, debug_probe_pyocd.py:220)
DEBUG:spsdk.debuggers.debug_probe:Selected AP: 2, Bank: 0x0 (10946ms since start, debug_probe.py:382)
DEBUG:spsdk.debuggers.debug_probe_pyocd:Coresight write AP, address: 00000000, data: 00000021 (10947ms since start, debug_probe_pyocd.py:220)
DEBUG:spsdk.debuggers.debug_probe:Trying to re-initialize debug connection (10951ms since start, debug_probe.py:319)
DEBUG:spsdk.debuggers.debug_probe:Power down the debug connection (10952ms since start, debug_probe.py:367)
DEBUG:spsdk.debuggers.debug_probe:Power Control the debug connection:
System power: False
Debug power: True (10952ms since start, debug_probe.py:334)
DEBUG:spsdk.debuggers.debug_probe_pyocd:Coresight write DP, address: 00000004, data: 10000F00 (10953ms since start, debug_probe_pyocd.py:220)
DEBUG:spsdk.debuggers.debug_probe_pyocd:Coresight read DP, address: 00000004, data: 30000F00 (10956ms since start, debug_probe_pyocd.py:190)
DEBUG:spsdk.debuggers.debug_probe:Power Control the debug connection:
System power: False
Debug power: False (10957ms since start, debug_probe.py:334)
DEBUG:spsdk.debuggers.debug_probe_pyocd:Coresight write DP, address: 00000004, data: 00000F00 (10957ms since start, debug_probe_pyocd.py:220)
DEBUG:spsdk.debuggers.debug_probe:Power up the debug connection (10958ms since start, debug_probe.py:361)
DEBUG:spsdk.debuggers.debug_probe:Power Control the debug connection:
System power: True
Debug power: True (10958ms since start, debug_probe.py:334)
DEBUG:spsdk.debuggers.debug_probe_pyocd:Coresight write DP, address: 00000004, data: 50000F00 (10959ms since start, debug_probe_pyocd.py:220)
DEBUG:spsdk.debuggers.debug_probe_pyocd:Coresight read DP, address: 00000004, data: F0000F00 (10963ms since start, debug_probe_pyocd.py:190)
DEBUG:spsdk.debuggers.debug_probe_pyocd:Coresight read DP, address: 00000004, data: F0000F00 (10966ms since start, debug_probe_pyocd.py:190)
DEBUG:spsdk.debuggers.debug_probe:Checked Sticky Errors: 0xf0000f00 (10966ms since start, debug_probe.py:283)
DEBUG:spsdk.debuggers.debug_probe:Debug interface: Read OK fail detected:

[... cut ...]
DEBUG:spsdk.debuggers.debug_probe_pyocd:Coresight write AP, address: 00000004, data: 00000000 (11112ms since start, debug_probe_pyocd.py:220)
DEBUG:spsdk.debuggers.debug_probe_pyocd:Coresight read AP, address: 00000000, data: 00000000 (11116ms since start, debug_probe_pyocd.py:190)
DEBUG:spsdk.debuggers.debug_probe_pyocd:Coresight read AP, address: 00000008, data: 00000000 (11119ms since start, debug_probe_pyocd.py:190)
DEBUG:spsdk.dat.debug_mailbox:-> spin_read:  0x0000_0000 (11120ms since start, dm_commands.py:72)
DEBUG:pyocd.probe.pydapaccess.interface.pyusb_backend:closing interface
Entering into ISP mode succeeded
```

flit, 2023-05-16:
Some improvement, I guess. 🫤

In the "target was not halted as expected after calling flash algorithm routine (IPSR=3)" error, the IPSR=3 means that the core is in the HardFault handler. In other words, the flash algorithm crashed when it was called to init for . And unfortunately, NXP doesn't provide the source to the flash algorithms in their DFPs, so it's not possible to debug.

IvoBCD, 2023-05-17:
@flit Thanks for looking into this; looks like I'll have to look into using MCUBoot or similar.

flit, 2023-05-17:
@IvoBCD I've asked NXP for the source to the flash algorithm, seems it might be possible to get it.

There is also source to the .cfx algorithms included with the MCUXpresso IDE, located at `<MCUXpresso IDE>/ide/Examples/Flashdrivers/NXP/iMXRT`. While the .cfx doesn't appear to be compatible, it does have some of the same APIs as seen in the `iMXRT6xxB_FlexSPI_SFDP/flashdriver/FlashPrg.c` source file. I'm asking about how this is related to the CMSIS-Pack style .FLM algorithms used by pyocd.

Also… it turns out that I do actually have an MIMXRT685-EVK! (I have so many boards that it's sometimes hard to remember what I do and don't have.) Sorry I forgot about this and made you do all the testing yourself.

sreenivr, 2024-06-05:
I am new to pyocd. I am trying to erase and program flash on NXP i.MX RT685 EVK using pyocd. Discussion on this thread helped me to overcome initial challenges and configure the pyocd_user.py script. Thanks !

I ran into the same issue described here.
**_0006956 C target was not halted as expected after calling flash algorithm routine (IPSR=3) [__main__]_**

Has anyone found a fix for this issue ?