pyocd/pyOCD #1556: Unable to force cmsis-dap v1 when flashing

afeinman-snap, 2023-05-14:
As a part of a different issue I am forced to fall back to cmsis-dap v1 when flashing on Windows. However the override (`-Ocmsis.dap.prefer_v1=True`) does not work with Load (or gdbserver) command.

Looking into the code, this is caused by the prefer_v1 being checked early in [dap_access_cmsis_dap._get_interfaces ](https://github.com/pyocd/pyOCD/blob/da303b046a8652674b84f27d5b754885209fafc2/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py#L77) called from  [choose_probe](https://github.com/pyocd/pyOCD/blob/da303b046a8652674b84f27d5b754885209fafc2/pyocd/core/helpers.py#L129). However the command-specific options such as -O are not parsed until the session is constructed at the end of [session_with_chosen_probe](https://github.com/pyocd/pyOCD/blob/da303b046a8652674b84f27d5b754885209fafc2/pyocd/core/helpers.py#LL208C9-L208C34). As a result the option is ignored and cmsis-dap v2 over libusb is used unconditionally

afeinman-snap, 2023-05-16:
Yes, I have a custom-build CMSIS-DAP firmware forked off the main project. Everything is fine on Linux/Mac. On Windows I get (using the libusb that comes with libusb_package)
```
(pyocd-test) PS C:\work\pyocd-test> pyocd load -t nrf52840 MainApp_0.hex -vv
0001091 D Project directory: C:\work\pyocd-test [session]
0001351 D Project directory: C:\work\pyocd-test [session]
0001454 C Error: Operation not supported or unimplemented on this platform [__main__]
Traceback (most recent call last):
  File "C:\work\pyocd-test\lib\site-packages\pyocd\__main__.py", line 161, in run
    status = cmd.invoke()
  File "C:\work\pyocd-test\lib\site-packages\pyocd\subcommands\load_cmd.py", line 83, in invoke
    session = ConnectHelper.session_with_chosen_probe(
  File "C:\work\pyocd-test\lib\site-packages\pyocd\core\helpers.py", line 265, in session_with_chosen_probe
    return Session(probe, auto_open=auto_open, options=options, **kwargs)
  File "C:\work\pyocd-test\lib\site-packages\pyocd\core\session.py", line 218, in __init__
    self._board = probe.create_associated_board() or Board(self)
  File "C:\work\pyocd-test\lib\site-packages\pyocd\probe\cmsis_dap_probe.py", line 248, in create_associated_board
    board_info = self.associated_board_info
  File "C:\work\pyocd-test\lib\site-packages\pyocd\probe\cmsis_dap_probe.py", line 219, in associated_board_info
    with _TemporaryOpen(self._link):
  File "C:\work\pyocd-test\lib\site-packages\pyocd\probe\cmsis_dap_probe.py", line 62, in __enter__
    self._device.open()
  File "C:\work\pyocd-test\lib\site-packages\pyocd\utility\concurrency.py", line 29, in _locking
    return func(self, *args, **kwargs)
  File "C:\work\pyocd-test\lib\site-packages\pyocd\probe\pydapaccess\dap_access_cmsis_dap.py", line 746, in open
    self._interface.open()
  File "C:\work\pyocd-test\lib\site-packages\pyocd\probe\pydapaccess\interface\pyusb_v2_backend.py", line 118, in open
    usb.util.claim_interface(dev, interface_number)
  File "C:\work\pyocd-test\lib\site-packages\usb\util.py", line 207, in claim_interface
    device._ctx.managed_claim_interface(device, interface)
  File "C:\work\pyocd-test\lib\site-packages\usb\core.py", line 113, in wrapper
    return f(self, *args, **kwargs)
  File "C:\work\pyocd-test\lib\site-packages\usb\core.py", line 178, in managed_claim_interface
    self.backend.claim_interface(self.handle, i)
  File "C:\work\pyocd-test\lib\site-packages\usb\backend\libusb1.py", line 829, in claim_interface
    _check(self.lib.libusb_claim_interface(dev_handle.handle, intf))
  File "C:\work\pyocd-test\lib\site-packages\usb\backend\libusb1.py", line 600, in _check
    raise NotImplementedError(_strerror(ret))
NotImplementedError: Operation not supported or unimplemented on this platform
```
If I force v1 (hardcoding it), everything works as expected.

flit, 2023-05-16:
> I could swear I searched for it first. Apologies

No worries! It's not always easy to find existing issues.

Sorry there's not an easy answer right now.

From a quick google search, it seems the error might be related to lack of a libusb-compatible driver? [See this](https://libusb-win32-devel.narkive.com/3YJBTsnU/libusb-claim-interface-failed-on-windows7)—although that's old and for Win7 the concept might still apply. Basically, on Windows you either need USB BOS descriptors specifying WinUSB compatibility, or use something like Zadig to install the libusb-compatible kernel driver for your device. This seems vaguely correct, since we're able to use DAPLink and other CMSIS-DAPv2 firmware that have the WInUSB BOS descriptors just fine on Win8 and above.

The reason CMSIS-DAPv1 would work is that it's HID based and uses the standard kernel HID class driver (plus pywinusb to access from pyocd).

afeinman-snap, 2023-05-16:
I figured it out. It's a libusb bug. On my device CMSIS-DAP v2 is interface #10. So windows assigns it a path VID_0D28&PID_0204&MI_0A. And yet libusb has this little gem of highly portable code
https://github.com/libusb/libusb/blob/07441f54244991af55df158b61fd69ca95b39662/libusb/os/windows_winusb.c#L1390-L1391
```C
	mi_str = strstr(device_id, "MI_");
	if ((mi_str != NULL) && isdigit((unsigned char)mi_str[3]) && isdigit((unsigned char)mi_str[4])) {
		interface_number = ((mi_str[3] - '0') * 10) + (mi_str[4] - '0');
```
So the interface simply gets ignored. I have filed a bug with libusb project, but in the meantime since pyocd insists on bringing it's own copy, I might look into rearranging the interface order in USB descriptor so that v1 becomes 10 and v2 becomes 9

elfmimi, 2023-05-16:
nice finding!

flit, 2023-05-17:
Indeed, impressive debugging. 😃 

> pyocd insists on bringing its own copy

Yeah, I know… Originally it didn't, but a somewhat surprising number of people had trouble installing libusb, especially on Windows, that something had to be done. For Linux, there are (imo) good arguments both for and against.

To use your own libusb build, you can either clone the repo and built it yourself or just replace the .dll in the installed libusb-package. There is also a branch (`feature/env_var_override`) that lets you use the `LIBUSBPKG_LIB_PATH` env var to specify either the libusb path or "system" to use the system-provided libusb. Just haven't tested it thoroughly. In addition, I was thinking about creating a variant like `libusb-package-system` that always passes through to the system libusb, but there are always higher priority things to work on.

flit, 2023-05-17:
(Closing as duplicate.)

closed 2023-05-17 by flit

afeinman-snap, 2024-05-31:
I see that libusb got a release in January that has my fix incorporated. How do we get this into PyOCD?