pyocd/pyOCD #1568: pyOCD 0.35.1 flashing fails for target stm32l071cztx (erase works)

easyvolts, 2023-05-31:
pyOCD 0.35.1 flashing fails for target stm32l071cztx (erase works), log output:

pyocd gdbserver --port 50000 --telnet-port 50001 --target stm32l071cztx
0002197 I Target type is stm32l071cztx [board]
0002268 I DP IDR = 0x0bc11477 (v1 MINDP rev0) [dap]
0002272 I debugvar 'DbgMCU_APB1_Fz' = 0x0 (0) [pack_target]
0002272 I debugvar 'DbgMCU_APB2_Fz' = 0x0 (0) [pack_target]
0002272 I debugvar 'DbgMCU_CR' = 0x6 (6) [pack_target]
0002272 I debugvar 'DoOptionByteLoading' = 0x0 (0) [pack_target]
0002323 I AHB-AP#0 IDR = 0x04770031 (AHB-AP var3 rev0) [discovery]
0002336 I AHB-AP#0 Class 0x1 ROM table #0 @ 0xf0000000 (designer=020:ST part=447) [rom_table]
0002344 I [0]<e00ff000:ROM class=1 designer=43b:Arm part=4c0> [rom_table]
0002344 I   AHB-AP#0 Class 0x1 ROM table #1 @ 0xe00ff000 (designer=43b:Arm part=4c0) [rom_table]
0002352 I   [0]<e000e000:SCS v6-M class=14 designer=43b:Arm part=008> [rom_table]
0002356 I   [1]<e0001000:DWT v6-M class=14 designer=43b:Arm part=00a> [rom_table]
0002360 I   [2]<e0002000:BPU v6-M class=14 designer=43b:Arm part=00b> [rom_table]
0002382 I CPU core #0 is Cortex-M0+ r0p1 [cortex_m]
0002385 I Setting core #0 (Cortex-M0+) default reset sequence to ResetSystem [pack_target]
0002391 I 2 hardware watchpoints [dwt]
0002395 I 4 hardware breakpoints, 0 literal comparators [fpb]
0002410 I Semihost server started on port 50001 (core 0) [server]
0002614 I GDB server started on port 50000 (core 0) [gdbserver]
0002620 I Client connected to port 50000! [gdbserver]
0002769 I Attempting to load RTOS plugins [gdbserver]
0003085 E Unhandled exception in handle_message (b'v'): Flash pages (1024 bytes) do not fit evenly into sector (128 bytes) [gdbserver]

flit, 2023-06-03:
Verified that this is caused by the `STM32L0xx_192.FLM` flash algorithm from the `Keil.STM32L0xx_DFP.2.2.0` pack indicating that the page size (0x400) is greater than the sector size (0x80). Pyocd currently doesn't support a configuration like this, as the normal definition of a flash page is a subset of a sector.

I'll report a bug for this algorithm.

Pyocd should probably be more flexible, too, but that's not something that can be done quickly.

hagibr, 2023-11-07:
In version 0.34.3, there is a warning when this happens and it silently changes the page size to be equal sector size:

(target\pack\cmsis_pack.py, inside the function _split_flash_region_by_sector_size(), that was removed with commit f0e62258)

```
            # Limit page size.
            if page_size > sector_size:
                region_page_size = sector_size
                LOG.warning("Page size (%d) is larger than sector size (%d) for flash region %s; "
                            "reducing page size to %d", page_size, sector_size, region.name,
                            region_page_size)
            else:
                region_page_size = page_size
```

Maybe something similar could be done with the current version to "fix" some flash algorithms.