pyocd/pyOCD #1572: `pyocd list` not detecting target with ST-Link/V2 probe

jacoblapenna, 2023-06-13:
`pyocd list` returns:

```
  #   Probe/Board    Unique ID                  Target  
--------------------------------------------------------
  0   STM32 STLink   34FF72063048553717451543   n/a     
```
Even though this should be a probe that supports auto-detection of target. STM32CubeProgrammer detects everything just fine on probe connection. My targets are STM32G071GBU6N and STM32L552VET6. I have installed the correct packs installed:

 ```
 Part             Vendor               Pack                 Version   Installed  
----------------------------------------------------------------------------------
  STM32G071KBUxN   STMicroelectronics   Keil.STM32G0xx_DFP   1.4.0     True  
```     
```
  Part             Vendor               Pack                 Version   Installed  
----------------------------------------------------------------------------------
  STM32L552VETx    STMicroelectronics   Keil.STM32L5xx_DFP   1.4.0     True       
  STM32L552VETxQ   STMicroelectronics   Keil.STM32L5xx_DFP   1.4.0     True      
```
I cannot flash withe generic `cortex_m` target. If I manually set the `Session()` objects options to the correct target via `target_override`, I can flash just fine, however, I need to programmatically detect when my ST-Link/V2 is connected to the L5 or the G0 and flash with the correct firmware. I cannot figure out a way to detect which MCU I am connected to in order to programmatically set the correct target and flash the correct firmware file.

pyocd version = 0.35.1
st-link/v2 firmware = V2J40S7

The documentation suggests I should be able to auto-detect with the `STM32 STLink` Probe/Board:
[image]

hagibr, 2023-08-08:
No, it just identifies the target device when you are using a debug probe embedded on a development kit, like the St-Link embedded on Nucleo Boards or Discovery Boards.
STM32CubeProgrammer must be identifying the device based on what it reads from specific unlocked memory regions and some kind of index table.

flit, 2023-08-08:
@jacoblapenna Sorry, I seem to have totally missed this issue a when you first created it…

Are you using your own boards or STM32 dev boards? As @hagibr says, auto detection only works with on-board probes, where a board-specific probe firmware build tells pyocd that target type.

For custom boards or when using standalone probes, you'll need to have a config file that maps the probe unique ID to target type. Pyocd's [config files](https://pyocd.io/docs/configuration.html) can do this, like this:

```yaml
probes:
  34FF72063048553717451543:
    target_override: STM32G071GBUxN # this is just a guess
```

Since `pyocd.yaml` file in the project directory (cwd by default) will be used automatically, you don't need to do anything in code to use a file like above. But if your script already has its own config file(s), you could use that instead.

Note that commit 12982a8 adds some recent board IDs from ST (and a few others). So if your dev board was in the list of additions it won't be detected automatically. This change will be included in the next release.

flit, 2023-08-09:
Closing as not an issue.

closed 2023-08-09 by flit