pyocd/pyOCD #1576: Unlock fails on nRF91

mekugi, 2023-06-28:
Hi,

Unlocking the nRF91 seems to fail on pyocd 0.35.1 (Windows11, python 3.11.1):

`pyocd.exe commander -vvv --target nrf9160_xxaa -Oauto_unlock
0001833 D CMSIS-DAP v2 probe xxx: protocol version 2.0.0 [dap_access_cmsis_dap]
0002101 D using family class NRF91 for nRF9160_xxAA (matched against nRF9160_xxAA) [pack_target]
0002107 I Target type is nrf9160_xxaa [board]
0002179 D Running task load_svd [sequencer]
0002198 D Running task pre_connect [sequencer]
0002207 D Running task dp_init [sequencer]
0002223 D Running task lock_probe [sequencer]
0002238 D Running task get_probe_capabilities [sequencer]
0002238 D Running task connect [sequencer]
0002242 D Default wire protocol selected; using SWD [dap]
0002243 D Sending deprecated SWJ sequence to select SWD [swj]
0002246 I DP IDR = 0x6ba02477 (v2 rev6) [dap]
0002246 D Running task clear_sticky_err [sequencer]
0002246 D Running task power_up_debug [sequencer]
0002247 D Running task check_version [sequencer]
0002247 D Running task unlock_probe [sequencer]
0002247 D Running task unlock_device [sequencer]
0002247 D Running task create_discoverer [sequencer]
0002247 D Running task discovery [sequencer]
0002247 D Running task find_aps [sequencer]
0002255 D Running task create_aps [sequencer]
0002255 D Running task create_ap.0 [sequencer]
0002258 D AHB-AP#0 default HPROT=3 HNONSEC=0 [ap]
0002259 D AHB-AP#0 implemented HPROT=f HNONSEC=1 [ap]
0002259 I AHB-AP#0 IDR = 0x84770001 (AHB-AP var0 rev8) [discovery]
0002259 D Running task create_ap.1 [sequencer]
0002262 D AHB-AP#1 default HPROT=3 HNONSEC=0 [ap]
0002263 D AHB-AP#1 implemented HPROT=3 HNONSEC=0 [ap]
0002264 I AHB-AP#1 IDR = 0x24770011 (AHB-AP var1 rev2) [discovery]
0002264 D Running task create_ap.2 [sequencer]
0002266 I AP#2 IDR = 0x12880000 (AP var0 rev1) [discovery]
0002266 D Running task create_ap.3 [sequencer]
0002269 D APB-AP#3 default HPROT=0 HNONSEC=0 [ap]
0002270 D APB-AP#3 implemented HPROT=0 HNONSEC=0 [ap]
0002271 I APB-AP#3 IDR = 0x54770002 (APB-AP var0 rev5) [discovery]
0002271 D Running task create_ap.4 [sequencer]
0002273 I AP#4 IDR = 0x12880000 (AP var0 rev1) [discovery]
0002273 D Running task create_ap.5 [sequencer]
0002274 I AP#5 IDR = 0x12880000 (AP var0 rev1) [discovery]
0002274 D Running task create_ap.6 [sequencer]
0002275 I AP#6 IDR = 0x128800a1 (AP var10 rev1) [discovery]
0002275 D Running task check_ctrl_ap_idr [sequencer]
0002275 D Running task check_flash_security [sequencer]
0002276 W nRF9160_xxAA APPROTECT enabled: will try to unlock via mass erase [target_nRF91]
0002683 D Running task lock_probe [sequencer]
0002684 D Running task connect [sequencer]
0002684 D Already connected with SWD [dap]
0002686 D Sending deprecated SWJ sequence to select SWD [swj]
0002689 I DP IDR = 0x6ba02477 (v2 rev6) [dap]
0002689 D Running task clear_sticky_err [sequencer]
0002689 D Running task power_up_debug [sequencer]
0002691 D Running task unlock_probe [sequencer]
0002693 D AHB-AP#0 default HPROT=3 HNONSEC=0 [ap]
0002694 D AHB-AP#0 implemented HPROT=f HNONSEC=1 [ap]
0002695 I AHB-AP#0 IDR = 0x84770001 (AHB-AP var0 rev8) [discovery]
0002699 D APB-AP#3 default HPROT=0 HNONSEC=0 [ap]
0002700 D APB-AP#3 implemented HPROT=0 HNONSEC=0 [ap]
0002700 I APB-AP#3 IDR = 0x54770002 (APB-AP var0 rev5) [discovery]
0002700 D Running task find_components [sequencer]
0002700 D Running task init_ap.0 [sequencer]
0002703 W Skipping CoreSight discovery for AHB-AP#0 because it is disabled [ap]
0002703 D Running task init_ap.1 [sequencer]
0002705 W Skipping CoreSight discovery for AHB-AP#1 because it is disabled [ap]
0002705 D Running task init_ap.3 [sequencer]
0002706 W Skipping CoreSight discovery for APB-AP#3 because it is disabled [ap]
0002706 D Running task create_cores [sequencer]
0002706 D Running task configure_core_reset [sequencer]
0002706 D Running task create_components [sequencer]
0002706 D Running task check_for_cores [sequencer]
0002706 E Error while initing target: No cores were discovered! [commander]
Traceback (most recent call last):
  File "C:\Users\jdoe\AppData\Local\Programs\Python\Python311\Lib\site-packages\pyocd\commands\commander.py", line 251, in _post_connect
    self.session.open(init_board=not self.args.no_init)
  File "C:\Users\jdoe\AppData\Local\Programs\Python\Python311\Lib\site-packages\pyocd\core\session.py", line 540, in open
    self._board.init()
  File "C:\Users\jdoe\AppData\Local\Programs\Python\Python311\Lib\site-packages\pyocd\board\board.py", line 142, in init
    self.target.init()
  File "C:\Users\jdoe\AppData\Local\Programs\Python\Python311\Lib\site-packages\pyocd\coresight\coresight_target.py", line 123, in init
    super().init()
  File "C:\Users\jdoe\AppData\Local\Programs\Python\Python311\Lib\site-packages\pyocd\core\soc_target.py", line 172, in init
    seq.invoke()
  File "C:\Users\jdoe\AppData\Local\Programs\Python\Python311\Lib\site-packages\pyocd\utility\sequencer.py", line 208, in invoke
    resultSequence = call()
                     ^^^^^^
  File "C:\Users\jdoe\AppData\Local\Programs\Python\Python311\Lib\site-packages\pyocd\coresight\coresight_target.py", line 313, in check_for_cores
    raise exceptions.DebugError("No cores were discovered!")
pyocd.core.exceptions.DebugError: No cores were discovered!
0002709 D uninit session <pyocd.core.session.Session object at 0x000002DCF5EED650> [session]`

Using the Python API show the same problem. Could this be a configuration issue in my side?

flit, 2023-06-28:
It's probably not a configuration issue. It looks like everything is in place to perform the unlock. Beyond that, for the time being I'll have to defer to the author of the nRF91 family.
cc @maxd-nordic

maxd-nordic, 2023-06-28:
I'd like two things for debugging:
* what SIP version is printed on the shield? It's the x's in "nRF9160 SICA xxx".
* do you have a working nRST line between probe and SIP? This is required for unlocking.
* do you use a Nordic board? If so, which one?

mekugi, 2023-06-29:
@maxd-nordic: 

- what SIP version is printed on the shield? It's the x's in "nRF9160 SICA xxx".
  - It is a SICA B1
- do you have a working nRST line between probe and SIP? This is required for unlocking.
  - No, I am using a [Raspberry Pi Debug Probe](https://www.raspberrypi.com/products/debug-probe/) which only offers SWCLK and SWDIO wires
- do you use a Nordic board? If so, which one?
  - No, it is a custom board

Unlocking works with Segger's J-Link, but this one indeed has the nRST line. I was trying to find a less costly alternative for the devices on CI.

Thank you for the support!

maxd-nordic, 2023-06-29:
For this version of the nRF91, you can unlock it and then try reconnecting after a power cycle.

I see - maybe pin reset support could be hacked into this by using a stray testpoint and modifying picoprobe. This definitely works with some other custom CMSIS-DAP probes. Other than the missing pin-reset support, this probe is really neat, I didn't know that one before.

closed 2023-06-29 by mekugi

flit, 2023-06-29:
Thanks @maxd-nordic! I can add some docs about this to the [target family usage notes](https://pyocd.io/docs/target_family_notes.html).

I'll also add some notes about the Raspberry Pi Debug Probe.