pyocd/pyOCD #1579: Can not read memory correctly in device.

nlcc96131415, 2023-07-05:
Step1 : In device a globle variable located at 0x20000004 increases by one as following codes

int watch=0;
int main()
{
	......
	
	while (1)
	{
		watch++;	
	        GPIO_BitToggle(GPIOA, GPIO_Pin_1);
	        delay10us(100000);
	}
}

Step2: read value of watch using pyocd python api  as following codes.

import pyocd
import time
from pyocd.core.helpers import ConnectHelper

import logging
logging.basicConfig(level=logging.INFO)

with ConnectHelper.session_with_chosen_probe() as session:

        board = session.board
        target = board.target   
        print("watch is: ", target.read32(0x20000004))
        time.sleep(1)
        print("watch is: ", target.read32(0x20000004))

Issue: target.read32(0x20000004) gets 1 all the time

flit, 2023-07-05:
It's probably because the CPU is halted. The default connect mode causes pyocd to halt the CPU(s). Are you seeing the 
GPIO toggle? Try inserting a `target.resume()` after the first `print()`.

You could also pass `connect_mode='attach'` to `ConnectHelper.session_with_chosen_probe()`. However, this won't ensure the CPU is running, it will simply prevent pyocd from halting upon connect.

flit, 2023-07-05:
That's good to hear. (Although, I don't know how the GPIO could be toggling if the CPU isn't running! 🤔)

The different connect modes are:
- `halt`: the default, halts CPUs after connection
- `pre-reset`: like `halt`, but performs a hardware reset just before connecting
- `under-reset`: holds hardware reset asserted while connecting, halts CPU prior to any code running, most useful for connecting to a device in power down or deep sleep state (if the device supports it, which varies by vendor)
- `attach`: no resets, doesn't halt the CPU

The connect mode to use depends on whether you want to do some setup prior letting the code run, or if you want to use hardware reset in order to start from the firmware from the beginning or recover from sleep mode.

Easiest is probably just using `halt` and then calling `target.reset()` if needed (which does a software reset by default, not hw like the connect mode), then `target.resume()` to ensure the CPU is running.

If you only want to connect and not perturb the running system (like if it's a motor control system), then use `attach` mode.

nlcc96131415, 2023-07-06:
Dear flit,
    I appreciate you for all your helpful help.
Best regards
tom

nlcc96131415, 2023-07-07:
Dear flit,
    I found another issue.  The device seems to reboot when run following codes because I found  "print" alway s starts from 1 no matter how long the device runs.

import pyocd
import time
from pyocd.core.helpers import ConnectHelper

with ConnectHelper.session_with_chosen_probe(connect_mode='attach') as session:
    
    board = session.board
    target = board.target   
    print("watch is: ", target.read32(0x20000004))   
    time.sleep(1)

flit, 2023-07-07:
It's possible the CPU simply isn't running. Remember, `attach` mode simply connects to the target in whatever state it was previously in. It doesn't ensure that it's running. So if the CPU is already halted, for instance if it was left in that state after programming flash, then that's what pyocd will see when you attach.

To see what state the target CPU is in, add a `print("CPU state:", target.get_state())` line.

Pyocd shouldn't be resetting the device.
You can also use the `DHCSR.S_RESET_ST` flag to tell if the CPU has been reset since the last time that register was read. eg:

```py
DHCSR = 0xE000EDF0
S_RESET_ST_MASK = (1 << 24)
was_reset = bool(target.read32(DHCSR) & S_RESET_ST_MASK)
print(f"{was_reset=}")
```

You can also check the debug log to see if pyocd is issuing a reset. Probably the easiest way is to just call `logging.basicConfig()`. For example, insert this before the `with` statement in your code above:

```py
import logging

logging.basicConfig(level=logging.DEBUG)
```

(You could also use `pyocd.utility.color_log.build_color_logger()`).

The debug log will indicate whether pyocd is resetting the target (a line like "reset, core 0, type=x"). If it's not, then something in the firmware must be doing it. If pyocd _is_ resetting, please provide the debug log here.

nlcc96131415, 2023-07-08:
Dear  flit, 
   following is the debug log after inserting logging.basicConfig before the with statement in my code

DEBUG:pyocd.core.session:Project directory: D:\workspace\C#\MyPython1
DEBUG:pyocd.core.session:Project directory: D:\workspace\C#\MyPython1
DEBUG:pyocd.probe.pydapaccess.dap_access_cmsis_dap:CMSIS-DAP v1 probe 69EB67F3AE8925C95E72B7138BBFC6960000000000000: protocol version 1.1.1
DEBUG:pyocd.probe.pydapaccess.interface.hidapi_backend:closing interface
WARNING:pyocd.board.mbed_board:Board ID 69EB is not recognized
WARNING:pyocd.board.board:Generic 'cortex_m' target type is selected by default; is this intentional? You will be able to debug most devices, but not program  flash. To set the target type use the '--target' argument or 'target_override' option. Use 'pyocd list --targets' to see available targets types.
DEBUG:pyocd.coresight.coresight_target:Using default Cortex-M memory map (no memory map supplied)
INFO:pyocd.board.board:Target type is cortex_m
DEBUG:pyocd.utility.sequencer:Running task load_svd
DEBUG:pyocd.utility.sequencer:Running task pre_connect
DEBUG:pyocd.utility.sequencer:Running task dp_init
DEBUG:pyocd.utility.sequencer:Running task lock_probe
DEBUG:pyocd.utility.sequencer:Running task get_probe_capabilities
DEBUG:pyocd.utility.sequencer:Running task connect
DEBUG:pyocd.coresight.dap:Default wire protocol selected; using SWD
DEBUG:pyocd.probe.swj:Sending deprecated SWJ sequence to select SWD
INFO:pyocd.coresight.dap:DP IDR = 0x0bb11477 (v1 MINDP rev0)
DEBUG:pyocd.utility.sequencer:Running task clear_sticky_err
DEBUG:pyocd.utility.sequencer:Running task power_up_debug
DEBUG:pyocd.utility.sequencer:Running task check_version
DEBUG:pyocd.utility.sequencer:Running task unlock_probe
DEBUG:pyocd.utility.sequencer:Running task unlock_device
DEBUG:pyocd.utility.sequencer:Running task create_discoverer
DEBUG:pyocd.utility.sequencer:Running task discovery
DEBUG:pyocd.utility.sequencer:Running task find_aps
DEBUG:pyocd.utility.sequencer:Running task create_aps
DEBUG:pyocd.utility.sequencer:Running task create_ap.0
DEBUG:pyocd.coresight.ap:AHB-AP#0 default HPROT=3 HNONSEC=0
DEBUG:pyocd.coresight.ap:AHB-AP#0 implemented HPROT=3 HNONSEC=0
INFO:pyocd.coresight.discovery:AHB-AP#0 IDR = 0x04770021 (AHB-AP var2 rev0)
DEBUG:pyocd.utility.sequencer:Running task find_components
DEBUG:pyocd.utility.sequencer:Running task init_ap.0
INFO:pyocd.coresight.rom_table:AHB-AP#0 Class 0x1 ROM table #0 @ 0xe00ff000 (designer=43b:Arm part=471)
INFO:pyocd.coresight.rom_table:[0]<e000e000:SCS v6-M class=14 designer=43b:Arm part=008>
INFO:pyocd.coresight.rom_table:[1]<e0001000:DWT v6-M class=14 designer=43b:Arm part=00a>
INFO:pyocd.coresight.rom_table:[2]<e0002000:BPU v6-M class=14 designer=43b:Arm part=00b>
DEBUG:pyocd.utility.sequencer:Running task create_cores
DEBUG:pyocd.coresight.discovery:Creating SCS component
DEBUG:pyocd.core.soc_target:selected core #0
INFO:pyocd.coresight.cortex_m:CPU core #0 is Cortex-M0 r0p0
DEBUG:pyocd.utility.sequencer:Running task create_components
DEBUG:pyocd.coresight.discovery:Creating DWT component
INFO:pyocd.coresight.dwt:2 hardware watchpoints
DEBUG:pyocd.coresight.discovery:Creating BPU component
INFO:pyocd.coresight.fpb:4 hardware breakpoints, 0 literal comparators
DEBUG:pyocd.coresight.fpb:fpb has been disabled
DEBUG:pyocd.utility.sequencer:Running task check_for_cores
DEBUG:pyocd.utility.sequencer:Running task halt_on_connect
DEBUG:pyocd.utility.sequencer:Running task post_connect
DEBUG:pyocd.utility.sequencer:Running task post_connect_hook
DEBUG:pyocd.utility.sequencer:Running task create_flash
DEBUG:pyocd.utility.sequencer:Running task notify
watch is:  1
DEBUG:pyocd.core.session:uninit session <pyocd.core.session.Session object at 0x000001F240206D50>
DEBUG:pyocd.board.board:uninit board <pyocd.board.mbed_board.MbedBoard object at 0x000001F23D2E18D0>
DEBUG:pyocd.coresight.cortex_m:cannot resume core 0: core is RUNNING
DEBUG:pyocd.probe.pydapaccess.interface.hidapi_backend:closing interface
Press any key to continue . . .

flit, 2023-07-08:
Well, pyocd isn't resetting the device. But the CPU does appear to be running, at least at the time when pyocd disconnects. This doesn't mean it's resetting on its own.

Have you debugged to verify the code is actually running as you expect? If not, please add some printf calls so you can see it's behaviour via the serial port, and/or use a debugger so you can step through the code.

Btw, what board are you using? The board ID 69EB isn't recognised, so pyocd is using the default `cortex_m` target.  If you tell me the board and target type, I can add a board ID entry for your board.

Often using the default `cortex_m` target is ok, as long as you don't need to program flash. But certain device families require extra debug logic to work correctly. I don't think that's the case here, but it's still helpful to set the correct target using the `target_override` option or param to the `ConnectHelper` methods. The [target support docs](https://pyocd.io/docs/target_support.html) have more.

nlcc96131415, 2023-07-09:
Dear flit,
     1) I am sure that pyocd does reset the device because the global variable "watch" is always starting from "1" ( via "print" ) no matter when I run the python script.
     2)  How can I get the board and target type? 
     3) I met a new issure when I try to modify my python script as following: the python script is blocked in "memory_value = target.read32(address)" 

import time
from typing import Union
import pyocd
from pyocd.core.helpers import ConnectHelper
target = None

def connect_device():
    global target
    with ConnectHelper.session_with_chosen_probe(connect_mode='attach') as session:
        target = session.board.target
        if target is not None:
            print("Device connected successfully.")
        else:
            print("Failed to connect device.")

def get_memory(address) -> Union[int, None]:       
    global target
    if target is not None:
        try:
            memory_value = target.read32(address)
            print("Memory value:", memory_value)
            return memory_value
        except Exception as e:
            print("Failed to read memory:", str(e))
            return None
    else:
        print("Target is not connected.")
        return None

connect_device()
memory_value = get_memory(0x20000004)
if memory_value is not None:
    print("Memory value:", memory_value)
else:
    print("Failed to read memory.")

flit, 2023-07-10:
Have you tried debugging your code and/or using printf to verify that the code really is running as expected?

The fact that the `watch` variable doesn't change isn't an indication that pyocd is resetting the device. There are other things that can cause a device to reset, such as a watchdog or automatic reset on the core entering lockup.

Because pyocd always outputs a debug log when it performs a reset, and that log message is not present, it's pretty clear that pyocd is not actively resetting the device. (There may be cases where target family support code would reset the device, such as when unlocking a locked device, but that can't be the case here because you are using the default `cortex_m` target.)

I won't rule out something unexpected happening, but we first need to verify the code really works correctly without pyocd playing a role. If it does, then we next need to verify that pyocd's usual interfaces such as the gdbserver work correctly—though it uses the same connect code so it should have the same behaviour.

---

For your second question, what board are you using and what is the MCU's part number?

nlcc96131415, 2023-07-11:
Dear flit
   1) I debugged my code using Keil. The code is running as expected and the led  toggles correctly too. 
   2) The watch variable( via  target.read32(0x20000004) ) changes, only it always changes from 1.

closed 2023-07-16 by nlcc96131415