pyocd/pyOCD #1596: cannot find my WCH-Link

wangwang105, 2023-07-24:
I saw the same problem（ #1121）

in use Ubuntu (Linux w-VirtualBox 5.19.0-46-generic #47~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Wed Jun 21 15:35:31 UTC 2 x86_64 x86_64 x86_64 GNU/Linux)
 When I run pyocd list, it just shows  `No available debug probes are connected.`

i ran `lsusb` on my ubuntu,output:
    Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

    Bus 002 Device 003: ID 1a86:8012 QinHeng Electronics WCH-Link

    Bus 002 Device 002: ID 80ee:0021 VirtualBox USB Tablet

    Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub

i ran `lsusb -v` on my ubuntu,output:

    Bus 002 Device 003: ID 1a86:8012 QinHeng Electronics WCH-Link

    Couldn't open device, some information will be missing

    Device Descriptor:

    bLength                18

    bDescriptorType         1

    bcdUSB               2.00

    bDeviceClass          239 Miscellaneous Device

    bDeviceSubClass         2 

    bDeviceProtocol         1 Interface Association

    bMaxPacketSize0        64

    idVendor           0x1a86 QinHeng Electronics

    idProduct          0x8012 

    bcdDevice            2.09

    iManufacturer           1 wch.cn

    iProduct                2 WCH-Link

    iSerial                 3 BC5DC2A77F04

    bNumConfigurations      1

    Configuration Descriptor:

    bLength                 9

    bDescriptorType         2

    wTotalLength       0x0062

    bNumInterfaces          3

    bConfigurationValue     1

    iConfiguration          0 

    bmAttributes         0x80

      (Bus Powered)

    MaxPower              500mA

    Interface Descriptor:

      bLength                 9

      bDescriptorType         4

      bInterfaceNumber        0

      bAlternateSetting       0

      bNumEndpoints           2

      bInterfaceClass       255 Vendor Specific Class

      bInterfaceSubClass      0 

      bInterfaceProtocol      0 

      iInterface              4 

      Endpoint Descriptor:

        bLength                 7

        bDescriptorType         5

        bEndpointAddress     0x02  EP 2 OUT

        bmAttributes            2

          Transfer Type            Bulk

          Synch Type               None

          Usage Type               Data

        wMaxPacketSize     0x0040  1x 64 bytes

        bInterval               0

      Endpoint Descriptor:

        bLength                 7

        bDescriptorType         5

        bEndpointAddress     0x83  EP 3 IN

        bmAttributes            2

          Transfer Type            Bulk

          Synch Type               None

          Usage Type               Data

        wMaxPacketSize     0x0040  1x 64 bytes

        bInterval               0

    Interface Association:

      bLength                 8

      bDescriptorType        11

      bFirstInterface         1

      bInterfaceCount         2

      bFunctionClass          2 Communications

      bFunctionSubClass       2 Abstract (modem)

      bFunctionProtocol       1 AT-commands (v.25ter)

      iFunction               4 

    Interface Descriptor:

      bLength                 9

      bDescriptorType         4

      bInterfaceNumber        1

      bAlternateSetting       0

      bNumEndpoints           2

      bInterfaceClass        10 CDC Data

      bInterfaceSubClass      0 

      bInterfaceProtocol      0 

      iInterface              0 

      Endpoint Descriptor:

        bLength                 7

        bDescriptorType         5

        bEndpointAddress     0x03  EP 3 OUT

        bmAttributes            2

          Transfer Type            Bulk

          Synch Type               None

          Usage Type               Data

        wMaxPacketSize     0x0040  1x 64 bytes

        bInterval               0

      Endpoint Descriptor:

        bLength                 7

        bDescriptorType         5

        bEndpointAddress     0x81  EP 1 IN

        bmAttributes            2

          Transfer Type            Bulk

          Synch Type               None

          Usage Type               Data

        wMaxPacketSize     0x0040  1x 64 bytes

        bInterval               0

    Interface Descriptor:

      bLength                 9

      bDescriptorType         4

      bInterfaceNumber        2

      bAlternateSetting       0

      bNumEndpoints           1

      bInterfaceClass         2 Communications

      bInterfaceSubClass      2 Abstract (modem)

      bInterfaceProtocol      1 AT-commands (v.25ter)

      iInterface              0 

      CDC Header:

        bcdCDC               1.10

      CDC Call Management:

        bmCapabilities       0x00

        bDataInterface          1

      CDC ACM:

        bmCapabilities       0x02

          line coding and serial state

      CDC Union:

        bMasterInterface        2

        bSlaveInterface         1 

      Endpoint Descriptor:

        bLength                 7

        bDescriptorType         5

        bEndpointAddress     0x84  EP 4 IN

        bmAttributes            3

          Transfer Type            Interrupt

          Synch Type               None

          Usage Type               Data

        wMaxPacketSize     0x0040  1x 64 bytes

        bInterval               1

I hope you can fix it~~

flit, 2023-07-24:
Do you have udev configured correctly to allow access to the device?

wangwang105, 2023-07-27:
In the same environment as above
i ran ‘pyocd --version‘ on my ubuntu,output:
‘0.35.1’
---
i ran 'cat /etc/udev/rules.d/'

output:

49-stlinkv1.rules
49-stlinkv2-1.rules
49-stlinkv2.rules
49-stlinkv3.rules
49-wch-link.rules
60-vboxadd.rules
70-snap.firefox.rules
70-snap.snapd-desktop-integration.rules
70-snap.snapd.rules
70-snap.snap-store.rules

---

There are rules, but there is still no way to recognize wch-link，hope you give me some suggestions~

flit, 2023-07-27:
Even with the udev rule file in place, I still think there is some issue with permissions. That's because `lsusb` failed to open the device ("Couldn't open device, some information will be missing").

If you haven't restarted your system or udev since installing the rules, then that would explain this.

The rules in `49-wch-link.rules` will create a `wch-link*` link under `/dev` somewhere (I'm not a Linux user, so I don't exactly where it would appear). You can also look for 0666 permissions on the WCH-Link device file.

Beyond that, I'm not sure what it could be… you might have to do some debugging. The amount that I can help is limited since I don't have a WCH-Link. I do know other people are using it though since someone contributed a patch for pyocd to be able to see a WCH-Link.

gerryqd, 2023-10-07:
It works well on my PCs, both windows and Ubuntu.
I suffered such kind of thing. But I think you can avoid the problem via below actions,
1. Upgrade the FW to latest one.
2. Avoid use of Virtualbox, use raw Linux directly.

wangwang105, 2023-10-08:
> It works well on my PCs, both windows and Ubuntu. I suffered such kind of thing. But I think you can avoid the problem via below actions,
> 
>     1. Upgrade the FW to latest one.
> 
>     2. Avoid use of Virtualbox, use raw Linux directly.

thanks for you advice. i will try what you suggested.