pyocd/pyOCD #1598: Upgrade capstone 5

patryk4815, 2023-07-27:
Please allow capstone 5 :)

https://github.com/pyocd/pyOCD/blob/401a01cbe193484c3220898a21b7f42c45936bcb/setup.cfg#L53

flit, 2023-07-27:
Can you submit a PR?

flit, 2023-07-30:
Tested with capstone 5. The shared lib in the MacOS wheel gets an error when it is loaded on an AArch64 machine:

```
 Error: dlopen(/VENV_DIR/lib/python3.11/site-packages/capstone/lib/libcapstone.dylib, 0x0006): tried: 
'/VENV_DIR/lib/python3.11/site-packages/capstone/lib/libcapstone.dylib' (mach-o file,
but is an incompatible architecture (have 'x86_64', need 'arm64')), 
'/System/Volumes/Preboot/Cryptexes/OS/VENV_DIR/lib/python3.11/site-packages/capstone/lib/libcapstone.dylib' (no such file),
 '/VENV_DIR/lib/python3.11/site-packages/capstone/lib/libcapstone.dylib' 
(mach-o file, but is an incompatible architecture (have 'x86_64', need 'arm64'))
```

Reported as https://github.com/capstone-engine/capstone/issues/2129

patryk4815, 2023-07-30:
@flit You can build it manualy look here: https://github.com/capstone-engine/capstone/issues/2102#issuecomment-1649019567 Soon wheel will be fixed on capstone side.

flit, 2023-07-31:
Building it manually is nice, but until it installs and runs successfully I cannot update the requirements for pyocd. It wouldn't even run on my own system for testing.

flit, 2023-07-31:
Looks like it will be fixed in v5.0.1. At that time, I'll update pyocd.

anthraxx, 2023-11-25:
@flit Can we get this sorted :cat:

cederom, 2023-11-25:
+1 to move to capstone 5 this would also help some things on BSD.. I patched capstone upstream some time ago and 5 now builds out of the box here with pip also system packages are available.. not sure how it works as I did not use that part :-)

If I can help / test anything on FreeBSD please let me know :-)

```
STARTING PYTHON VIRTUAL ENVIRONMENT AT: /home/XXX/.local/venv3.9embedded
(venv3.9embedded) pip install capstone
Requirement already satisfied: capstone in ./.local/venv3.9embedded/lib/python3.9/site-packages/capstone-4.0.2-py3.9.egg (4.0.2)
(venv3.9embedded) pip uninstall capstone
Found existing installation: capstone 4.0.2
Uninstalling capstone-4.0.2:
  Would remove:
    /usr/home/XXX/.local/venv3.9embedded/lib/python3.9/site-packages/capstone-4.0.2-py3.9.egg
Proceed (Y/n)? y
  Successfully uninstalled capstone-4.0.2

(venv3.9embedded) pip install capstone==5.0.1
Collecting capstone==5.0.1
  Downloading capstone-5.0.1.tar.gz (2.9 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 2.9/2.9 MB 10.7 MB/s eta 0:00:00
  Preparing metadata (setup.py) ... done
Building wheels for collected packages: capstone
  Building wheel for capstone (setup.py) ... done
  Created wheel for capstone: filename=capstone-5.0.1-py3-none-freebsd_13_2_release_p4_amd64.whl size=2745089 sha256=15fa0dccb2e0af48b859db7eeb5e4b28163c4ac6144e2ae7c66e64f157895d88
  Stored in directory: /usr/home/XXX/.cache/pip/wheels/42/00/0d/f50db564c3ebe74cf0ff4fb1ca8c0d4bb6698623941ec6ee33
Successfully built capstone
Installing collected packages: capstone
ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
pyocd 0.34.0 requires capstone<5.0,>=4.0, but you have capstone 5.0.1 which is incompatible.
Successfully installed capstone-5.0.1

(venv3.9embedded) pkg search capstone
capstone-5.0.1_1               Multi-platform, multi-architecture disassembly framework
capstone4-4.0.2_1              Multi-platform, multi-architecture disassembly framework
py39-capstone-5.0.1            Python bindings for the Capstone Disassembly Engine

(venv3.9embedded) uname -a
FreeBSD octagon 13.2-RELEASE-p4 FreeBSD 13.2-RELEASE-p4 GENERIC amd64
```

cederom, 2023-11-25:
> Tested with capstone 5. The shared lib in the MacOS wheel gets an error when it is loaded on an AArch64 machine:
> 
> ```
>  Error: dlopen(/VENV_DIR/lib/python3.11/site-packages/capstone/lib/libcapstone.dylib, 0x0006): tried: 
> '/VENV_DIR/lib/python3.11/site-packages/capstone/lib/libcapstone.dylib' (mach-o file,
> but is an incompatible architecture (have 'x86_64', need 'arm64')), 
> '/System/Volumes/Preboot/Cryptexes/OS/VENV_DIR/lib/python3.11/site-packages/capstone/lib/libcapstone.dylib' (no such file),
>  '/VENV_DIR/lib/python3.11/site-packages/capstone/lib/libcapstone.dylib' 
> (mach-o file, but is an incompatible architecture (have 'x86_64', need 'arm64'))
> ```
> 
> Reported as [capstone-engine/capstone#2129](https://github.com/capstone-engine/capstone/issues/2129)

I think you need to have system package installed in order for pip to work..? Or the capstone python package setup installer is broken :-P I also have MacStudio / M2 AArch64 so I can help testing here too :-)

jjakob, 2024-07-13:
Status on capstone 5 support? Gentoo removed 4 from its repo already