pyocd/pyOCD #1600: C target was not halted as expected after calling flash algorithm routine (IPSR=3) [__main__]

USTHzhanglu, 2023-07-28:
Hi my friends:
    recently i upgrade pyocd from 0.34.1 to 0.35.1. When I flash my program as usual，pyocd show me `C target was not halted as expected after calling flash algorithm routine (IPSR=3) [__main__]`.
    I rollback version to 0.34.3 ,it work. just 0.35.0 and 0.35.1 can`t work.
    this  0.35.1 log:

```powershell
 pyocd flash -vv ms300_top_full.bin
0001594 D Project directory: D:\workspace\git\ms300_firmware\top [session]
0001595 D Loading config from: D:\workspace\git\ms300_firmware\top\pyocd.yaml [session]
0001756 D Project directory: D:\workspace\git\ms300_firmware\top [session]
0001758 D Loading config from: D:\workspace\git\ms300_firmware\top\pyocd.yaml [session]
0001818 D STLink probe 58005200070000574B31534E firmware version: V2J39M7 [stlink]
0001821 D Fetching mounted devices from disk service registry entry [windows]
0001821 D Fetching usb storage devices from USBSTOR service registry entry [windows]
0001821 D No USBSTOR service found, no device can be detected [windows]
0001822 D target_id_usb_id -> mount_point mapping: {}  [windows]
0001828 D STLink probe 58005200070000574B31534E firmware version: V2J39M7 [stlink]
0001829 D Fetching mounted devices from disk service registry entry [windows]
0001829 D Fetching usb storage devices from USBSTOR service registry entry [windows]
0001830 D No USBSTOR service found, no device can be detected [windows]
0001830 D target_id_usb_id -> mount_point mapping: {}  [windows]
0001834 D HC32F460 DFP: skipping non-Keil flash algorithm [cmsis_pack]
0001835 D HC32F460 DFP: skipping non-Keil flash algorithm [cmsis_pack]
0001835 D HC32F460 DFP: skipping non-Keil flash algorithm [cmsis_pack]
0001835 D HC32F460 DFP: skipping non-Keil flash algorithm [cmsis_pack]
0001837 D HC32F460 DFP: skipping non-Keil flash algorithm [cmsis_pack]
0001837 D HC32F460 DFP: skipping non-Keil flash algorithm [cmsis_pack]
0001838 D HC32F460 DFP: skipping non-Keil flash algorithm [cmsis_pack]
0001839 D HC32F460 DFP: skipping non-Keil flash algorithm [cmsis_pack]
0001841 D HC32F460 DFP: skipping non-Keil flash algorithm [cmsis_pack]
0001841 D HC32F460 DFP: skipping non-Keil flash algorithm [cmsis_pack]
0001842 D HC32F460 DFP: skipping non-Keil flash algorithm [cmsis_pack]
0001842 D HC32F460 DFP: skipping non-Keil flash algorithm [cmsis_pack]
0001843 D HC32F460 DFP: skipping non-Keil flash algorithm [cmsis_pack]
0001843 D HC32F460 DFP: skipping non-Keil flash algorithm [cmsis_pack]
0001844 D HC32F460 DFP: skipping non-Keil flash algorithm [cmsis_pack]
0001844 D HC32F460 DFP: skipping non-Keil flash algorithm [cmsis_pack]
0001845 D HC32F460 DFP: skipping non-Keil flash algorithm [cmsis_pack]
0001845 D HC32F460 DFP: skipping non-Keil flash algorithm [cmsis_pack]
0001856 I Target type is hc32f460jeua [board]
0001862 D STLink probe 58005200070000574B31534E firmware version: V2J39M7 [stlink]
0001863 D Running task load_svd [sequencer]
0001909 D Running task pre_connect [sequencer]
0001915 D Running task dp_init [sequencer]
0001921 D Running task lock_probe [sequencer]
0001927 D Running task get_probe_capabilities [sequencer]
0001933 D Running task connect [sequencer]
0001965 D Default wire protocol selected; using SWD [dap]
0001980 I DP IDR = 0x2ba01477 (v1 rev2) [dap]
0001981 D Running task clear_sticky_err [sequencer]
0001981 D Running task power_up_debug [sequencer]
0001982 D Running task check_version [sequencer]
0001982 D Running task unlock_probe [sequencer]
0001983 D Running task unlock_device [sequencer]
0001983 D Running task create_discoverer [sequencer]
0001984 D Running task discovery [sequencer]
0001984 D Running task find_aps [sequencer]
0001986 D Running task create_aps [sequencer]
0001988 D Running task create_ap.0 [sequencer]
0001988 D Using accelerated memory access interface for AHB-AP#0 [ap]
0001989 D AHB-AP#0 default HPROT=3 HNONSEC=0 [ap]
0001990 D AHB-AP#0 implemented HPROT=3 HNONSEC=0 [ap]
0001991 I AHB-AP#0 IDR = 0x24770011 (AHB-AP var1 rev2) [discovery]
0001991 D Running task find_components [sequencer]
0001992 D Running task init_ap.0 [sequencer]
0001994 I AHB-AP#0 Class 0x1 ROM table #0 @ 0xe00ff000 (designer=43b:Arm part=4c4) [rom_table]
0001995 I [0]<e000e000:SCS v7-M class=14 designer=43b:Arm part=00c> [rom_table]
0001996 I [1]<e0001000:DWT v7-M class=14 designer=43b:Arm part=002> [rom_table]
0001997 I [2]<e0002000:FPB v7-M class=14 designer=43b:Arm part=003> [rom_table]
0001998 I [3]<e0000000:ITM v7-M class=14 designer=43b:Arm part=001> [rom_table]
[... 180 lines cut ...]
0002570 D Using double buffer sector erase program [builder]
[========================================          ]  81%0003181 D halting core 0 [cortex_m]
0003187 D resuming core 0 [cortex_m]
0003187 D added=[] removed=[] [manager]
0003188 D bps after flush={} [manager]
0003201 D resuming core 0 [cortex_m]
0003201 D added=[] removed=[] [manager]
0003202 D bps after flush={} [manager]
0003211 D resuming core 0 [cortex_m]
0003211 D added=[] removed=[] [manager]
0003212 D bps after flush={} [manager]
[==================================================] 100%
0003216 D Estimated sector erase programmed page count: 0 [builder]
0003216 D Actual sector erase programmed page count: 0 [builder]
0003217 D set reset catch, core 0 [cortex_m]
0003217 D halting core 0 [cortex_m]
0003219 D reset, core 0, type=SW_SYSRESETREQ [cortex_m]
0003335 D clear reset catch, core 0 [cortex_m]
0003336 I Erased 0 bytes (0 sectors), programmed 0 bytes (0 pages), skipped 89600 bytes (175 pages) at 86.27 kB/s [loader]
0003337 D uninit session <pyocd.core.session.Session object at 0x000002B1C618F970> [session]
0003338 D uninit board <pyocd.board.board.Board object at 0x000002B1C618F820> [board]
0003339 D resuming core 0 [cortex_m]
0003340 D added=[] removed=[] [manager]
0003340 D bps after flush={} [manager]
```

flit, 2023-07-28:
Thanks for the report. Looks like the flash algo is crashing, getting a Hard Fault exception (IPSR == 3).

When you ran v0.34, the flash was not actually programmed because the contents of the flash didn't change. You need to pass `-Osmart_flash=0` to force flash programming. Then we can tell whether the flash algo runs correctly in v0.34.

Note that I don't have an HC32F460, so I can't test or reproduce myself.

flit, 2023-07-28:
Is there a CMSIS-Pack available for the HC32F460? I don't see one on https://www.keil.arm.com/devices/. Without access to the pack, there's no way for me to verify whether there's something in its device description that might cause pyocd trouble.

lennvn, 2023-07-30:
I'll test.

flit, 2023-07-30:
@lennvn Thanks!!

lennvn, 2023-07-31:
No errors occurred during my testing. 
@USTHzhanglu :
1. What is the version of your HC32F460 pack?
2. Waht is your hardware debugger? STLink?
3. Could you show me your operation step by step with full commands?
---------------------------------------------------------------------------------------------
My testing log of pyocd 0.35.1:

Installing collected packages: pyocd
Successfully installed pyocd-0.35.1

wuze@CDLT MINGW64 /g/Test
$ pyocd erase --chip -t hc32f460xe
0001435 I Erasing chip... [eraser]
0001848 I Chip erase complete [eraser]

wuze@CDLT MINGW64 /g/Test
$ pyocd flash -vv -t hc32f460xe gpio_output.bin
0000934 D Project directory: G:\Test [session]
0001206 D Project directory: G:\Test [session]
0001248 D Unrecognised major version of CMSIS-DAP: protocol version 256.0.0 [dap_access_cmsis_dap]
0001248 D CMSIS-DAP v2 probe 070000814d363134ff115048ffff5f04a5a5a5a597969921: protocol version 2.0.0 [dap_access_cmsis_dap]
0001256 I Target type is hc32f460xe [board]
0001299 D Running task load_svd [sequencer]
0001301 D Running task pre_connect [sequencer]
0001302 D Running task dp_init [sequencer]
0001303 D Running task lock_probe [sequencer]
0001304 D Running task get_probe_capabilities [sequencer]
0001306 D Running task connect [sequencer]
0001323 D Default wire protocol selected; using SWD [dap]
0001327 D Sending deprecated SWJ sequence to select SWD [swj]
0001342 I DP IDR = 0x2ba01477 (v1 rev2) [dap]
0001343 D Running task clear_sticky_err [sequencer]
0001389 D Running task power_up_debug [sequencer]
0001397 D Running task check_version [sequencer]
0001398 D Running task unlock_probe [sequencer]
0001401 D Running task unlock_device [sequencer]
0001404 D Running task create_discoverer [sequencer]
0001409 D Running task discovery [sequencer]
0001415 D Running task find_aps [sequencer]
0001482 D Running task create_aps [sequencer]
0001489 D Running task create_ap.0 [sequencer]
0001543 D AHB-AP#0 default HPROT=3 HNONSEC=0 [ap]
0001545 D AHB-AP#0 implemented HPROT=3 HNONSEC=0 [ap]
0001547 I AHB-AP#0 IDR = 0x24770011 (AHB-AP var1 rev2) [discovery]
0001547 D Running task find_components [sequencer]
0001547 D Running task init_ap.0 [sequencer]
0001556 I AHB-AP#0 Class 0x1 ROM table #0 @ 0xe00ff000 (designer=43b:Arm part=4c4) [rom_table]
0001561 I [0]<e000e000:SCS v7-M class=14 designer=43b:Arm part=00c> [rom_table]
0001564 I [1]<e0001000:DWT v7-M class=14 designer=43b:Arm part=002> [rom_table]
0001568 I [2]<e0002000:FPB v7-M class=14 designer=43b:Arm part=003> [rom_table]
0001573 I [3]<e0000000:ITM v7-M class=14 designer=43b:Arm part=001> [rom_table]
0001578 I [4]<e0040000:TPIU M4 class=9 designer=43b:Arm part=9a1 devtype=11 archid=0000 devid=ca0:0:0> [rom_table]
0001578 D [5]<fff42002 not present> [rom_table]
0001578 D Running task create_cores [sequencer]
0001578 D Creating SCS component [discovery]
0001579 D selected core #0 [soc_target]
0001585 I CPU core #0 is Cortex-M4 r0p1 [cortex_m]
0001593 I FPU present: FPv4-SP-D16-M [cortex_m]
0001596 D Running task create_components [sequencer]
0001596 D Creating DWT component [discovery]
0001599 I 4 hardware watchpoints [dwt]
0001601 D Creating FPB component [discovery]
0001605 I 6 hardware breakpoints, 4 literal comparators [fpb]
0001605 D fpb has been disabled [fpb]
0001607 D Creating ITM component [discovery]
0001612 D Creating TPIU component [discovery]
0001614 D Running task check_for_cores [sequencer]
0001614 D Running task halt_on_connect [sequencer]
0001614 D halting core 0 [cortex_m]
[... 41 lines cut ...]
[==========================                        ]  53%0002064 D resuming core 0 [cortex_m]
0002064 D added=[] removed=[] [manager]
0002064 D bps after flush={} [manager]
[==================================                ]  68%0002095 D resuming core 0 [cortex_m]
0002095 D added=[] removed=[] [manager]
0002095 D bps after flush={} [manager]
[==========================================        ]  84%0002125 D resuming core 0 [cortex_m]
0002125 D added=[] removed=[] [manager]
0002125 D bps after flush={} [manager]
[==================================================] 100%
0002153 D resuming core 0 [cortex_m]
0002153 D added=[] removed=[] [manager]
0002154 D bps after flush={} [manager]
0002164 D Estimated sector erase programmed page count: 6 [builder]
0002164 D Actual sector erase programmed page count: 6 [builder]
0002165 D set reset catch, core 0 [cortex_m]
0002165 D halting core 0 [cortex_m]
0002171 D reset, core 0, type=SW_SYSTEM [cortex_m]
0002300 D clear reset catch, core 0 [cortex_m]
0002302 I Erased 8192 bytes (1 sector), programmed 3072 bytes (6 pages), skipped 0 bytes (0 pages) at 4.41 kB/s [loader]
0002302 D uninit session <pyocd.core.session.Session object at 0x000001C7B93C8250> [session]
0002303 D uninit board <pyocd.board.board.Board object at 0x000001C7B93C82B0> [board]
0002305 D resuming core 0 [cortex_m]
0002305 D added=[] removed=[] [manager]
0002305 D bps after flush={} [manager]

USTHzhanglu, 2023-07-31:
> No errors occurred during my testing. @USTHzhanglu :
> 
> 1. What is the version of your HC32F460 pack?
> 2. Waht is your hardware debugger? STLink?
> 3. Could you show me your operation step by step with full commands?
> 
> My testing log of pyocd 0.35.1:
> 
> ```
> Installing collected packages: pyocd
> Successfully installed pyocd-0.35.1
> 
> wu@CDLT MINGW64 /g/Test
> $ pyocd erase --chip -t hc32f460xe
> 0001435 I Erasing chip... [eraser]
> 0001848 I Chip erase complete [eraser]
> 
> wu@CDLT MINGW64 /g/Test
> $ pyocd flash -vv -t hc32f460xe gpio_output.bin
> 0000934 D Project directory: G:\Test [session]
> 0001206 D Project directory: G:\Test [session]
> 0001248 D Unrecognised major version of CMSIS-DAP: protocol version 256.0.0 [dap_access_cmsis_dap]
> 0001248 D CMSIS-DAP v2 probe 070000814d363134ff115048ffff5f04a5a5a5a597969921: protocol version 2.0.0 [dap_access_cmsis_dap]
> 0001256 I Target type is hc32f460xe [board]
> 0001299 D Running task load_svd [sequencer]
> 0001301 D Running task pre_connect [sequencer]
> 0001302 D Running task dp_init [sequencer]
> 0001303 D Running task lock_probe [sequencer]
> 0001304 D Running task get_probe_capabilities [sequencer]
> 0001306 D Running task connect [sequencer]
> 0001323 D Default wire protocol selected; using SWD [dap]
> 0001327 D Sending deprecated SWJ sequence to select SWD [swj]
> 0001342 I DP IDR = 0x2ba01477 (v1 rev2) [dap]
> 0001343 D Running task clear_sticky_err [sequencer]
> 0001389 D Running task power_up_debug [sequencer]
> 0001397 D Running task check_version [sequencer]
> 0001398 D Running task unlock_probe [sequencer]
> 0001401 D Running task unlock_device [sequencer]
> 0001404 D Running task create_discoverer [sequencer]
> 0001409 D Running task discovery [sequencer]
> 0001415 D Running task find_aps [sequencer]
> 0001482 D Running task create_aps [sequencer]
> 0001489 D Running task create_ap.0 [sequencer]
> 0001543 D AHB-AP#0 default HPROT=3 HNONSEC=0 [ap]
> 0001545 D AHB-AP#0 implemented HPROT=3 HNONSEC=0 [ap]
> 0001547 I AHB-AP#0 IDR = 0x24770011 (AHB-AP var1 rev2) [discovery]
> 0001547 D Running task find_components [sequencer]
> 0001547 D Running task init_ap.0 [sequencer]
> 0001556 I AHB-AP#0 Class 0x1 ROM table #0 @ 0xe00ff000 (designer=43b:Arm part=4c4) [rom_table]
> 0001561 I [0]<e000e000:SCS v7-M class=14 designer=43b:Arm part=00c> [rom_table]
> 0001564 I [1]<e0001000:DWT v7-M class=14 designer=43b:Arm part=002> [rom_table]
> 0001568 I [2]<e0002000:FPB v7-M class=14 designer=43b:Arm part=003> [rom_table]
> 0001573 I [3]<e0000000:ITM v7-M class=14 designer=43b:Arm part=001> [rom_table]
> 0001578 I [4]<e0040000:TPIU M4 class=9 designer=43b:Arm part=9a1 devtype=11 archid=0000 devid=ca0:0:0> [rom_table]
> 0001578 D [5]<fff42002 not present> [rom_table]
> 0001578 D Running task create_cores [sequencer]
> 0001578 D Creating SCS component [discovery]
> 0001579 D selected core #0 [soc_target]
> 0001585 I CPU core #0 is Cortex-M4 r0p1 [cortex_m]
> 0001593 I FPU present: FPv4-SP-D16-M [cortex_m]
> 0001596 D Running task create_components [sequencer]
> 0001596 D Creating DWT component [discovery]
> 0001599 I 4 hardware watchpoints [dwt]
> 0001601 D Creating FPB component [discovery]
> 0001605 I 6 hardware breakpoints, 4 literal comparators [fpb]
> 0001605 D fpb has been disabled [fpb]
> 0001607 D Creating ITM component [discovery]
> 0001612 D Creating TPIU component [discovery]
> 0001614 D Running task check_for_cores [sequencer]
> 0001614 D Running task halt_on_connect [sequencer]
[... 190 lines cut ...]
> 0002095 D bps after flush={} [manager]
> [==========================================        ]  84%0002125 D resuming core 0 [cortex_m]
> 0002125 D added=[] removed=[] [manager]
> 0002125 D bps after flush={} [manager]
> [==================================================] 100%
> 0002153 D resuming core 0 [cortex_m]
> 0002153 D added=[] removed=[] [manager]
> 0002154 D bps after flush={} [manager]
> 0002164 D Estimated sector erase programmed page count: 6 [builder]
> 0002164 D Actual sector erase programmed page count: 6 [builder]
> 0002165 D set reset catch, core 0 [cortex_m]
> 0002165 D halting core 0 [cortex_m]
> 0002171 D reset, core 0, type=SW_SYSTEM [cortex_m]
> 0002300 D clear reset catch, core 0 [cortex_m]
> 0002302 I Erased 8192 bytes (1 sector), programmed 3072 bytes (6 pages), skipped 0 bytes (0 pages) at 4.41 kB/s [loader]
> 0002302 D uninit session <pyocd.core.session.Session object at 0x000001C7B93C8250> [session]
> 0002303 D uninit board <pyocd.board.board.Board object at 0x000001C7B93C82B0> [board]
> 0002305 D resuming core 0 [cortex_m]
> 0002305 D added=[] removed=[] [manager]
> 0002305 D bps after flush={} [manager]
> ```

rev1.0.8,[HC32F460_IDE_Rev1.0.8.zip](https://www.hdsc.com.cn/cn/Index/downloadFile/modelid/66/id/58/key/2)
I use stm32v2 and daplink
but i find if i use builtin "hc32f460xe", it work.

USTHzhanglu, 2023-07-31:
@lennvn like this 
```
pyocd flash -vv .\ms300_top_full.bin -t hc32f460jeua --pack .\HDSC.HC32F460.1.0.8.pack
0001634 D Project directory: D:\workspace\git\ms300_firmware\top [session]
0001635 D Loading config from: D:\workspace\git\ms300_firmware\top\pyocd.yaml [session]
0001804 D Project directory: D:\workspace\git\ms300_firmware\top [session]
0001806 D Loading config from: D:\workspace\git\ms300_firmware\top\pyocd.yaml [session]
0001832 D CMSIS-DAP v2 probe 0001A0000000: protocol version 2.0.0 [dap_access_cmsis_dap]
0001838 D HC32F460 DFP: skipping non-Keil flash algorithm [cmsis_pack]
0001839 D HC32F460 DFP: skipping non-Keil flash algorithm [cmsis_pack]
0001839 D HC32F460 DFP: skipping non-Keil flash algorithm [cmsis_pack]
0001839 D HC32F460 DFP: skipping non-Keil flash algorithm [cmsis_pack]
0001840 D HC32F460 DFP: skipping non-Keil flash algorithm [cmsis_pack]
0001841 D HC32F460 DFP: skipping non-Keil flash algorithm [cmsis_pack]
0001841 D HC32F460 DFP: skipping non-Keil flash algorithm [cmsis_pack]
0001842 D HC32F460 DFP: skipping non-Keil flash algorithm [cmsis_pack]
0001843 D HC32F460 DFP: skipping non-Keil flash algorithm [cmsis_pack]
0001845 D HC32F460 DFP: skipping non-Keil flash algorithm [cmsis_pack]
0001846 D HC32F460 DFP: skipping non-Keil flash algorithm [cmsis_pack]
0001846 D HC32F460 DFP: skipping non-Keil flash algorithm [cmsis_pack]
0001846 D HC32F460 DFP: skipping non-Keil flash algorithm [cmsis_pack]
0001847 D HC32F460 DFP: skipping non-Keil flash algorithm [cmsis_pack]
0001847 D HC32F460 DFP: skipping non-Keil flash algorithm [cmsis_pack]
0001848 D HC32F460 DFP: skipping non-Keil flash algorithm [cmsis_pack]
0001848 D HC32F460 DFP: skipping non-Keil flash algorithm [cmsis_pack]
0001849 D HC32F460 DFP: skipping non-Keil flash algorithm [cmsis_pack]
0001876 I Target type is hc32f460jeua [board]
0001902 D Running task load_svd [sequencer]
0001947 D Running task pre_connect [sequencer]
0001955 D Running task dp_init [sequencer]
0001961 D Running task lock_probe [sequencer]
0001966 D Running task get_probe_capabilities [sequencer]
0001972 D Running task connect [sequencer]
0002020 D Default wire protocol selected; using SWD [dap]
0002022 D Sending deprecated SWJ sequence to select SWD [swj]
0002030 I DP IDR = 0x2ba01477 (v1 rev2) [dap]
0002030 D Running task clear_sticky_err [sequencer]
0002030 D Running task power_up_debug [sequencer]
0002036 D Running task check_version [sequencer]
0002036 D Running task unlock_probe [sequencer]
0002036 D Running task unlock_device [sequencer]
0002037 D Running task create_discoverer [sequencer]
0002037 D Running task discovery [sequencer]
0002038 D Running task find_aps [sequencer]
0002046 D Running task create_aps [sequencer]
0002046 D Running task create_ap.0 [sequencer]
0002052 D AHB-AP#0 default HPROT=3 HNONSEC=0 [ap]
0002054 D AHB-AP#0 implemented HPROT=3 HNONSEC=0 [ap]
0002056 I AHB-AP#0 IDR = 0x24770011 (AHB-AP var1 rev2) [discovery]
0002056 D Running task find_components [sequencer]
0002056 D Running task init_ap.0 [sequencer]
0002066 I AHB-AP#0 Class 0x1 ROM table #0 @ 0xe00ff000 (designer=43b:Arm part=4c4) [rom_table]
0002070 I [0]<e000e000:SCS v7-M class=14 designer=43b:Arm part=00c> [rom_table]
0002074 I [1]<e0001000:DWT v7-M class=14 designer=43b:Arm part=002> [rom_table]
0002078 I [2]<e0002000:FPB v7-M class=14 designer=43b:Arm part=003> [rom_table]
0002082 I [3]<e0000000:ITM v7-M class=14 designer=43b:Arm part=001> [rom_table]
0002088 I [4]<e0040000:TPIU M4 class=9 designer=43b:Arm part=9a1 devtype=11 archid=0000 devid=ca0:0:0> [rom_table]
0002088 D [5]<fff42002 not present> [rom_table]
0002088 D Running task create_cores [sequencer]
0002088 D Creating SCS component [discovery]
0002090 D selected core #0 [soc_target]
0002094 I CPU core #0 is Cortex-M4 r0p1 [cortex_m]
0002098 I FPU present: FPv4-SP-D16-M [cortex_m]
0002100 D Running task configure_core_reset [sequencer]
0002101 D updated DFP core #0 reset types: {<ResetType.SW: 2>, <ResetType.SW_EMULATED: 5>, <ResetType.HW: 1>, <ResetType.SW_CORE: 4>, <ResetType.SW_SYSTEM: 3>} [pack_target]
0002101 I Setting core #0 (Cortex-M4) default reset sequence to ResetSystem [pack_target]
0002102 D Running task create_components [sequencer]
0002105 D Creating DWT component [discovery]
0002108 I 4 hardware watchpoints [dwt]
0002108 D Creating FPB component [discovery]
[... 42 lines cut ...]
Traceback (most recent call last):
  File "C:\Users\langen\AppData\Local\Programs\Python\Python310\lib\site-packages\pyocd\__main__.py", line 161, in run
    status = cmd.invoke()
  File "C:\Users\langen\AppData\Local\Programs\Python\Python310\lib\site-packages\pyocd\subcommands\load_cmd.py", line 130, in invoke
    programmer.program(filename,
  File "C:\Users\langen\AppData\Local\Programs\Python\Python310\lib\site-packages\pyocd\flash\file_programmer.py", line 175, in program
    self._loader.commit()
  File "C:\Users\langen\AppData\Local\Programs\Python\Python310\lib\site-packages\pyocd\flash\loader.py", line 295, in commit
    perf = builder.program(chip_erase=chipErase,
  File "C:\Users\langen\AppData\Local\Programs\Python\Python310\lib\site-packages\pyocd\flash\builder.py", line 492, in program
    sector_erase_count, page_program_time = self._compute_sector_erase_pages_and_weight(fast_verify)
  File "C:\Users\langen\AppData\Local\Programs\Python\Python310\lib\site-packages\pyocd\flash\builder.py", line 673, in _compute_sector_erase_pages_and_weight
    self._analyze_pages_with_partial_read()
[... cut ...]
  File "C:\Users\langen\AppData\Local\Programs\Python\Python310\lib\site-packages\pyocd\flash\flash.py", line 676, in _call_function_and_wait
    return self.wait_for_completion(timeout=timeout)
  File "C:\Users\langen\AppData\Local\Programs\Python\Python310\lib\site-packages\pyocd\flash\flash.py", line 663, in wait_for_completion
    raise exceptions.FlashFailure("target was not halted as expected after calling "
pyocd.core.exceptions.FlashFailure: target was not halted as expected after calling flash algorithm routine (IPSR=0)
```

lennvn, 2023-07-31:
@USTHzhanglu The problem seems to be caused by using the wrong target name. The valid target names of hc32f460  in pyocd are:
`hc32f460xc
hc32f460xe`
hc32f460jeua should use "hc32f460xe".

USTHzhanglu, 2023-07-31:
> @USTHzhanglu The problem seems to be caused by using the wrong target name. The valid target names of hc32f460 in pyocd are "hc32f460xc" and "hc32f460xe", and hc32f460jeua should use "hc32f460xe".

not all. i use this target name and pack @0.34.x over a year. now @0.35.x always can use hc32f460xe, so thinks

lennvn, 2023-08-02:
@USTHzhanglu :
pyocd 0.35.1: I tested your command and reproduced the issue. Changing the target name from 'hc32f460jeua' to 'hc32f460xe', it works.
@flit :
1. Is there any difference between the two versions(0.34.1 vs 0.35.1) when recognizing the hc32f460 target?
2. Is there anything I can do for solving this issue?

flit, 2023-08-03:
@lennvn 

Is this understanding of the situation correct?
- The reported bug is with an flash algorithm in a `HDSC.HC32F460.1.0.8.pack` CMSIS-Pack.
- PyOCD's builtin `hc32f460xe` target works correctly.

> Is there any difference between the two versions(0.34.1 vs 0.35.1) when recognizing the hc32f460 target?

As far as CoreSight discovery and target communications, there are no substantial differences.

There are quite a few differences in the CMSIS-Pack parsing, especially regarding memory regions. In addition, pyOCD supports CMSIS-Pack debug sequences now. The docs are [here](https://pyocd.io/docs/open_cmsis_pack_support.html).

Also, the memory layout of CMSIS-Pack based flash algorithms has changed. (Not that this does not apply to builtin flash algorithms, which have a memory layout defined by the algo dict.)

From the earlier comment…

The flash algorithm layout in RAM for CMSIS-Pack-based targets changed in v0.35 to make better use of available memory.
The snippets below from the debug log above show the old and new flash algo layout.

pre v0.35:

```
[code=0x538] [b1=0x540,0x740] [b2=0x740,0x940] [stack=0x2940; 0x2000 b] (ram=0x1fff8000, 0x2f000 b)
[code=0x3fc] [b1=0x400,0x7fc] [b2=0x800,0xbfc] [stack=0x2bfc; 0x2000 b] (ram=0x1fff8000, 0x2f000 b)
```

post v0.35:

```
[stack=0x200266c0; 0x2e6c0 b] [b2=0x200268c0,+0x2e8c0] [b1=0x200266c0,+0x2e6c0] [code=0x20026ac8,+0x2eac8,0x538 b] (ram=0x1fff8000, 0x2f000 b)
[stack=0x20026400; 0x2e400 b] [b2=0x20026800,+0x2e800] [b1=0x20026400,+0x2e400] [code=0x20026c04,+0x2ec04,0x3fc b] (ram=0x1fff8000, 0x2f000 b)
```

The algo code and buffers are placed at the end of the specified memory region, with the stack taking up all available memory below the buffers. This allows the stack to consume as much memory as is available, which is better since there is no specification of the required stack size for flash algos.

> Is there anything I can do for solving this issue?

I'm not yet sure. It depends on whether there is a bug in the flash algorithm contained in `HDSC.HC32F460.1.0.8.pack`, or if it's a pyOCD bug.

To answer this question, we'll probably need to debug the flash algorithm crash to understand why it's failing. (Unfortunately, debugging flash algorithms is very difficult. You can't use a normal debugger, because the debugger is running the flash algorithm already.)

Could you check the "post v0.35" flash algorithm memory layout shown above, and the flash algorithm code, to see whether there might be an issue with the new memory layout?

Is the `HDSC.HC32F460.1.0.8.pack` CMSIS-Pack publicly available? It doesn't appear in the [public CMSIS-Pack list](https://www.keil.arm.com/devices/?q=hdsc&page=1).

Also, because I don't have an HC32F460 development board, I can't reproduce or debug the problem directly…

lennvn, 2023-08-08:
### sdfadf

lennvn, 2023-08-08:
> dafasdf

lennvn, 2023-08-08:
`sdfasdf`

flit, 2023-08-08:
@lennvn 

Pyocd looks at two places in packs for flash algorithm information:

1. The device description in the `.pdsc` XML file. This contains details of memory regions and a references to the flash algorithms. See [the algorithm element docs](https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/pdsc_family_pg.html#element_algorithm). In short, the `<algorithm>` element specifies the `.flm` file to use for a range of flash memory, and can define the range of RAM where the algorithm should run from.

2. The `.flm` flash algorithm referenced from the `<algorithm>` element. See [FLM docs](https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/flashAlgorithm.html). `.flm` files are ELF executables built for position independent code and data. In addition to the actual algorithm code and data, there is a small struct that defines the flash memory address range plus sector and page sizes.

To place the algorithm in memory, pyocd first selects which RAM to use. If the `<algorithm>` element has `RAMstart` and `RAMsize` elements, that RAM will be used. Otherwise, all of the lowest-address RAM region is used.

These blocks are placed in the selected range of RAM:

1. Stack
2. 1 or 2 page programming buffers (will use 2 if there's enough memory, for faster programming)
3. Code + Data

The Code + Data is placed at the _end_ of the selected range of RAM, with the other sections placed in sequence below that.

It looks like the HC32F460 and HC32F4A0 have different RAM regions. For the HC32F460, pyocd is using RAM from 0x1fff_8000..0x20027000. For the HC32F4A0, the RAM used is 0x1ffe0000..0x1ffe4000. This is because of how the RAM regions are defined in the `.pdsc` file.

Given those RAM regions, and different sizes of code and data in the F460 and F4A0 flash algorithms, you'll get the different flash algorithm memory layouts shown above.

Note that you can run the pyocd `show map` command to print the memory map built from the `.pack`. (eg, `pyocd cmd -c show map`)

----

Questions:

- Could you copy the output from the `show map` command for the F460 and F4A0?
- Could you email me the two `.pack` file(s)? (Or attach to a comment here.)
- Are there any RAM regions that are _invalid_ for use by the flash algorithms? For example, RAM that code cannot be run from.
- Similar to the previous question: Is there any reason code could not be executed at the RAM addresses shown in the two memory layouts?

I can help review if the flash algorithm memory layout is correct.

USTHzhanglu, 2023-08-09:
@flit 
this is show map log:
```
pyocd cmd -c show map --pack HDSC.HC32F460.1.0.8.pack
 Region   Type   Start       End         Size        Access  Sector      Page
 flash    Flash  0x00000000  0x0007ffff  0x00080000  rx      0x00002000  0x00000200
 flash_1  Flash  0x03000c00  0x03000ffb  0x000003fc  rx      0x000003fc  0x000003fc
 ram      Ram    0x1fff8000  0x20026fff  0x0002f000  rwx     -           -
 ram_1    Ram    0x200f0000  0x200f0fff  0x00001000  rwx     -           -

````
you can click it [HC32F460_IDE_Rev1.0.8.zip](https://www.hdsc.com.cn/cn/Index/downloadFile/modelid/66/id/58/key/2) to get pack
[HC32F4A0_IDE_Rev1.0.5.zip](https://www.hdsc.com.cn/cn/Index/downloadFile/modelid/66/id/62/key/0)

lennvn, 2023-08-09:
@flit Thanks so much for your answer.
@USTHzhanglu Thanks for your reply.

@flit 

> Could you copy the output from the show map command for the F460 and F4A0?
F460:
```
$ pyocd cmd -c show map -t hc32f460jeua --pack HDSC.HC32F460.1.0.10.pack
 Region  Type   Start       End         Size        Access  Sector      Page
 IROM1   Flash  0x00000000  0x0007ffff  0x00080000  rx      0x00002000  0x00000200
 IROM2   Flash  0x03000c00  0x03000ffb  0x000003fc  rx      0x000003fc  0x000003fc
 IRAM1   Ram    0x1fff8000  0x20026fff  0x0002f000  rwx     -           -
 IRAM2   Ram    0x200f0000  0x200f0fff  0x00001000  rwx     -           -
```
F4A0:
```
$ pyocd cmd -c show map -t hc32f4a0sitb --pack HDSC.HC32F4A0.1.0.7.pack
 Region  Type   Start       End         Size        Access  Sector      Page
 IROM1   Flash  0x00000000  0x001fffff  0x00200000  rx      0x00002000  0x00000800
 IROM2   Flash  0x03000000  0x030017ff  0x00001800  rx      0x00001800  0x00001800
 IRAM1   Ram    0x1ffe0000  0x2005ffff  0x00080000  rwx     -           -
```

flit, 2023-08-15:
@lennvn @USTHzhanglu Thanks for the data! I'm starting to review everything to look for possible issues.

lennvn, 2023-09-08:
@flit May I ask if there is any progress, thanks

lennvn, 2023-09-28:
@flit hello?

AlfYuen, 2025-02-01:
I'm having same issue, not working for HC32F460 since 0.35.0. Any progress? Or is there any workarounds for newer versions?

USTHzhanglu, 2025-02-01:
> Or is there any workarounds for newer versions?

use builtin "hc32f460xe"

USTHzhanglu, 2026-08-27:
**Summary: root cause found — vendor pack is missing the standard `RAMstart`/`RAMsize` attributes; the cleanest fix is patching the pack, not pyocd**

After digging into this old issue again, I believe I've found the actual root cause. In short: **this is not a pyocd bug**. pyOCD's v0.35 layout change exposed a latent problem in the vendor pack, and the proper fix is a small manual patch to the `.pack` file.

**1. The flash algorithm is position-independent**

I disassembled `FlashARM/HC32F460_512K.FLM` (from `HDSC.HC32F460.1.0.8.pack`) and examined every PC-relative literal load. The literal pool contains only peripheral registers (`0x40054000` EFM, `0x40010400` RCU, ...), the unlock sequence `0xffff0123`, and the error code `0xffff3210` (the one we see as `result code 0xffff3210`). **There is not a single absolute RAM address** — the code is fully position-independent, so the memory layout change alone cannot break it by "stale absolute pointers".

**2. Only the "high RAM" placement fails**

| Configuration | Algo code location | Result |
|---|---|---|
| pyOCD builtin `hc32f460xe` | `0x20000000` (low SRAM) | works |
| pre-0.35 pack layout | `0x1fff8000` (low SRAMH) | worked for a year |
| post-0.35 pack layout | `0x20026ac8` (top of SRAM) | HardFault (IPSR=3) |

The `Init` routine even *returns* error code `0xffff3210` — it detects the environment is wrong and fails cleanly before the crash. Since pyOCD 0.35 changed the CMSIS-Pack algo layout to place code at the **top** of the chosen RAM region, and the HC32F460 pack does not constrain that region, the algo ends up at `0x20026xxx` — the top of SRAM bank 2, where execution is unreliable on this part.

**3. The vendor pack violates the CMSIS-Pack spec (by omission)**

Per the Open-CMSIS-Pack spec, the `<algorithm>` element has optional `RAMstart`/`RAMsize` attributes to define where the flash algo may run:

> *RAMstart* — "Base address for the RAM where the Flash programming algorithm will be executed from."
> *RAMsize* — "Maximum size of RAM available for the execution of the Flash programming algorithm."

`HDSC.HC32F460.1.0.8.pack` declares neither. So pyOCD (and Keil MDK) falls back to the whole default RAM (`IRAM1 = 0x1FFF8000 + 0x2F000`), and pyOCD 0.35+ places the code at the top of that range → `0x20026ac8`.

**4. Best fix: patch the pack (no code changes needed)**

Add the two attributes to every `<algorithm>` element in the `.pdsc`, then re-zip the pack:

```xml
<algorithm name="FlashARM/HC32F460_512K.FLM" RAMstart="0x1FFF8000" RAMsize="0x2000" .../>
<algorithm name="FlashARM/HC32F460_otp.FLM"   RAMstart="0x1FFF8000" RAMsize="0x2000" .../>
```

This constrains the algo to the low 8 KB of SRAMH. With pyOCD's top-down layout the code then lands at `0x1fff9ac8` (buffers `0x1fff98c0`/`0x1fff96c0`, stack 5.7 KB) — **all inside the address range that the pre-0.35 layout actually exercised for a year** (`0x1fff8000`–`0x1fffa940`), far away from the broken `0x20026xxx` region.

This is a spec-compliant fix (the attributes are standard), works with both pyOCD and Keil MDK, requires no pyOCD changes, and does not affect debugging — the memory map is unchanged; only the flash-algo scratch area is constrained.

**Suggested follow-ups**

- It would be great if HDSC (now XHSC) fixes their pack by adding `RAMstart`/`RAMsize`.
- pyOCD could optionally detect this class of problem: if a pack algo fails inside its allocated RAM while an alternative low-RAM placement is available, log a hint about `RAMstart`/`RAMsize`. (Or simply document this workaround on the CMSIS-Pack support page.)

closed 2026-08-27 by USTHzhanglu

lennvn, 2026-09-04:
@USTHzhanglu Thank you so much for identifying the root cause and providing a solution. Initially, I couldn't reproduce the issue, so we put it on hold. I didn't expect you to keep pushing forward—your persistence is truly appreciated. Regarding these two attributes, we've added them across our other products, but this particular product is older and was overlooked. We never anticipated this would be the source of the problem. I sincerely apologize for causing you trouble.