pyocd/pyOCD #1601: gdb debugging gives gdb internal-error: int finish_step_over(execution_control_state*): Assertion `ecs->event_thread->control.trap_expected' failed.

zougloub, 2023-08-01:
Hi,

I have used the same GDB macro in OpenOCD and pyOCD and in pyOCD it fails with an internal gdb error.
I don't know how reproducible my scenario is, but I have a target device that is repetitively calling a function and I had a breakpoint that was printing arguments and continuing execution.

```gdb
break some_function
commands
silent
if argument == value
  printf "some message\n"
   print other_argument
   print yet_another_argument
end
continue
end
```

After some cycles (~ 20) gdb would fail with:
```
internal-error: int finish_step_over(execution_control_state*): Assertion `ecs->event_thread->control.trap_expected' failed.
```

There are open bugs in GDB for this messages (eg. [this one](https://sourceware.org/bugzilla/show_bug.cgi?id=22063)) but since the problem is not happening with OpenOCD I'm opening an issue here.

flit, 2023-08-03:
Thanks for the report. It's highly likely that the step commands have differences between pyOCD and OpenOCD that gdb is somehow unhappy with, whether or not it's technically a bug.

Do you have a project that you can provide that reproduces it? There's a good chance it's specific to the code being debugged. In any case, I'll try to reproduce it as soon as I get a chance.

flit, 2023-08-09:
A couple other questions for you.

What MCU and CPU?
Are you using an RTOS and pyOCD's RTOS support?