pyocd/pyOCD #1602: pyocd install custom cmsis pack

WGR7, 2023-08-02:
Hi,

I'm need to install a custom pack for Puya PY32 MCUs. I've saw some tricks unzipping the DFP and copying some pdsc files to a specific folder, but I'm getting a failure when trying to add the pack to cmsis-pack-manager running "pack-manager add-pack abcd.dfp" command. The pack-manager was not found on command line.

So, Is there a method to install a custom pack DFP calling pyocd on command line?

SO: Ubuntu 22

IvoBCD, 2023-10-10:
@WGR7 You can invoke pyocd with the ``-pack`` argument, as in:

```
pyocd flash --pack "Keil.LPC1100_DFP.1.4.0.pack" -t "LPC11C24FBD48/301" blinky.hex
```

elfmimi, 2023-12-28:
Unfortunately pyocd and cmsis-pack-manager do not support easy way to install individual .pack s.

But there is a way to accomplish that part manually.
Let's see haw it can be done step-by-step.

1. Find and download specific .pack you want to use.
search by device name: https://www.keil.arm.com/devices/
or
search by pack name: https://www.keil.arm.com/packs/
　
say I wanted to use NuMicro_DFP 1.3.20, it can be downloaded from this page.
https://www.keil.arm.com/packs/numicro_dfp-nuvoton/versions/

2. extract .pdsc out of .pack
.pack file is actually a .zip file. You can modify the file extension if you want , and then extract it content.
find a file with .pdsc extension. for example it may look like Nuvoton.NuMicro_DFP.pdsc .

3. check the location of cmsis-pack-manager's pack repository.
`python -c "import cmsis_pack_manager.pack_manager as pack_manager; pack_manager.main()" print-cache-dir`
I'm using windows, so this will show a path like this.
C:\Users\ _username_ \AppData\Local\cmsis-pack-manager\cmsis-pack-manager

4. clear the cmsys-pack-manager's pack repository to avoid any complication.
`python -m pyocd pack clean`

5. create an empty directory at the location where pack repository will be placed.
`mkdir $env:LOCALAPPDATA\cmsis-pack-manager\cmsis-pack-manager`

6. create pack index by installing the .pdsc file
`python -c "import cmsis_pack_manager.pack_manager as pack_manager; pack_manager.main()" add-packs path_to_where_you_put\Nuvoton.NuMicro_DFP.pdsc`

7. manually place the .pack file to where it is expected to be.
`mkdir $env:LOCALAPPDATA\cmsis-pack-manager\cmsis-pack-manager\Nuvoton\NuMicro_DFP`
`copy path_to_where_you_put\Nuvoton.NuMicro_DFP.1.3.20.pack $env:LOCALAPPDATA\cmsis-pack-manager\cmsis-pack-manager\Nuvoton\NuMicro_DFP\1.3.20.pack`

8. check that pyocd is properly recognizing the pack you installed.
`python -m pyocd pack show`
should show
```
      Pack                  Version
    ---------------------------------
      Nuvoton.NuMicro_DFP   1.3.20
```
9. check that pyocd properly lists devices from the pack you installed.
`python -m pyocd list --targets`

bryceschober, 2025-06-03:
> 7\. manually place the .pack file to where it is expected to be.

@elfmimi Where would I know where it is expected to be?