pyocd/pyOCD #1616: Supporting cores with private flash partitions

maxd-nordic, 2023-08-25:
Hi! I'm looking into supporting the nRF5340 in pyocd.
Although it seems easy enough to have multiple flasher algos for different memory partitions, I couldn't find a way to specify on which core the algo needs to run. We would usually have a merged HEX file containing all the partitions that is supplied to the flasher tool. It would be great if someone could help me out here. :)

flit, 2023-08-29:
Unfortunately… pyocd currently can only run the flash algo on the main core. ☹️ It's definitely a serious deficiency for certain devices!

flit, 2023-08-29:
It looks like it wouldn't be too hard to implement. The `pyocd.flash.flash.Flash` class that's responsible for running the flash algo already takes a `Target` class used to access the core. If the target instance for a core (rather than SoC) is passed in, the algo would be run on that core.

The main changes would be
- Tracking the core the algo needs to on. Using a `FlashRegion` attribute with the core name (or number) would be fine. (Default of None needs to be added to `FlashRegion.DEFAULT_ATTRS` dict.)
- Update `CoreSightTarget.create_flash()` to look up the core and pass it to the `Flash` class constructor, or use the primary core if not specified.
- Update `pyocd.target.pack.cmsis_pack.CmsisPackDevice` to set the flash algo core name attribute.

maxd-nordic, 2023-09-01:
Thanks for the insight! I'm currently tinkering with this on my develop branch.
Do you have an idea why
* the second core is not automatically generated
* only the first flash region is actually saved as a flash regions, while the others are ROM regions?

maxd-nordic, 2024-01-01:
Got some more progress, here's a PR: https://github.com/pyocd/pyOCD/pull/1656

closed 2024-01-01 by maxd-nordic