pyocd/pyOCD #1617: how to disable load svd  file

FangYuFan, 2023-08-28:
I'm trying to debug my code with VSCode. Since I don't have a svd file,error occured when I start debugging(Exception in thread load-svd). I have already add "svdFile":null in launch.json but it doesn't work. what can I do to prevent pyocd from loading the  svd file?

flit, 2023-08-29:
Pyocd should only attempt to load an SVD file if one is provided.

Fyi, the cortex-debug `svdFile` key is not passed to pyocd. It's only used with the cortex-debug register viewer UI.

Some questions:
What target type are you using?
If the target is from a CMSIS-Pack, which one and which version?
What are the cortex-debug settings?
What is the pyocd command line? (If you know, this can also be inferred from the cortex-debug settings.)
What version of pyocd?

FangYuFan, 2023-08-31:
> 
thanks for reply.
The target is from a self-made .pack file, and the version of pyocd is 0.35.1
In fact, when I download hex/bin file to MCU, it also encounter the prompt message that the svd file fails to load, even in the case of a successful download. 
The pyocd command line is "pyocd flash --target mpf5200x Uart_Demo.hex" as shown below.
[image]
the pdsc file set svd path as empty.
[image]
If I delete line of [svd=""] in pdsc file, the program cannot been download with the error info shown as follow.
[image]

Could you provide any advice to prevent pyocd from loading svd file.

elfmimi, 2024-01-19:
You can make use of [pyocd user script](https://pyocd.io/docs/user_scripts.html) for this.

You only have to define [will_init_target](https://pyocd.io/docs/user_scripts.html#will_init_target) delegate function.

```python
def will_init_target(target: "SoCTarget", init_sequence: "CallSequence") -> None:
    init_sequence.remove_task('load_svd')
```