pyocd/pyOCD #1622: It takes ~3 seconds for gdb to connect to pyocd (likely windows only)

macromaniac, 2023-09-04:
Hi. When I run 

`pyocd gdbserver --port 50000 --telnet-port 50001 --target rp2040`
then
`arm-none-eabi-gdb.exe .\build\templateRP2040.elf`
`target remote localhost:50000`

It can take up to 3 seconds to connect. When I use `127.0.0.1` instead of `localhost` it is instant.

I looked at wireshark and it seems like my computer is trying to connect to the ipv6 localhost `::1` before giving up and trying ipv4.

[image]

I tried editing the hosts file but apparently doesn't work for `localhost` anymore, I tried editing the local adapter to not use ipv6 and no dice. I'm on  windows 10 (10.0.19045.0 Microsoft Windows NT 10.0.19045.0)

When I edit this line in sockets.py it works.

[image]
however this breaks 127.0.0.1, so it's debatable whether this would be a good change.

I am using cortex debug vscode extension and thus I can't easily edit cortex debug so it specifies 127.0.0.1 and not localhost.

TLDR: AFAICT the default behavior in windows causes it to take 3 seconds to connect for a debug session.

macromaniac, 2023-09-04:
Hosting on both ipv6 and ipv4 localhost can work too, I can potentially pr this, just would need to test it on linux and perhaps on another device first

yyjdelete, 2023-09-19:
Maybe https://github.com/libusb/libusb/pull/1270? If you have a usb devices disabled or not install drivers