pyocd/pyOCD #1629: bug: PyOCD does not support flashing of LPCXpressp55S36  RevD1 (public release)

butok, 2023-10-10:
PyOCD does not support flashing of  the [LPCXpressp55S36  RevD1](https://www.nxp.com/design/software/development-software/mcuxpresso-software-and-tools-/lpcxpresso-boards/lpcxpresso55s36-development-board:LPCXpresso55S36) (public release), that contains the latest LPC55S36 v1 chip.
During flashing it print error message:
`0006181 C target was not halted as expected after calling flash algorithm routine (IPSR=3) [__main__]
FATAL ERROR: command exited with status 1: pyocd flash -e sector -t lpc55s36 'C:\_ddm\ZEPHYR\zephyr_vs\zephyr\build\mcuboot\zephyr\zephyr.bin'`

NOTE:
The **PyOCD supports the old LPCXpressp55S36  RevA** board with an old LPC55S36 v0 chip (not public). 
Most probably the issue is caused by incompatible changes in new chip ROM.

butok, 2023-11-14:
Hi @Hoohaha,
could you look at this LPCXpressp55S36 RevD1 issue.

maxwell652, 2024-12-03:
hi @butok,

I got the same issue with another chip, have you solved this problem?

maxwell652, 2024-12-04:
> > hi @butok,
> > I got the same issue with another chip, have you solved this problem?
> 
> The problem has not been solved.

I solved the issue I encountered. My situation was that I loaded the program algorithm to RAM address 0x20000000, but this address is invalid in my MCU memory design. As a result, when pyOCD called the "pc_init" function of the program algorithm([here](https://github.com/pyocd/pyOCD/blob/5166025ae5da5e093d6cfe2b26cae5e1334476e4/pyocd/flash/flash.py#L253)), it didn't execute as expected. I changed the loading address of the program algorithm, and it worked correctly.

Hope this help for you