pyocd/pyOCD #1635: pyocd rtt, how to send data?

wenhuafeng, 2023-10-13:
$ pyocd rtt -t hc32l072
0001312 I Target type is hc32l072 [board]
0001329 I DP IDR = 0x0bc11477 (v1 MINDP rev0) [dap]
0001395 I AHB-AP#0 IDR = 0x04770031 (AHB-AP var3 rev0) [discovery]
0001401 I AHB-AP#0 Class 0x1 ROM table #0 @ 0xe00ff000 (designer=43b:Arm part=4c0) [rom_table]
0001406 I [0]<e000e000:SCS v6-M class=14 designer=43b:Arm part=008> [rom_table]
0001409 I [1]<e0001000:DWT v6-M class=14 designer=43b:Arm part=00a> [rom_table]
0001412 I [2]<e0002000:BPU v6-M class=14 designer=43b:Arm part=00b> [rom_table]
0001415 I CPU core #0: Cortex-M0+ r0p1, v6.0-M architecture [cortex_m]
0001420 I 2 hardware watchpoints [dwt]
0001422 I 4 hardware breakpoints, 0 literal comparators [fpb]
0001452 I 3 up channels and 3 down channels found [rtt_cmd]
0001452 I Reading from up channel 0 ("Terminal") [rtt_cmd]
0001455 I Writing to down channel 0 ("Terminal") [rtt_cmd]
Info:  [modbus_rtu_slave] HEX_CMD_03
Info:  [modbus_rtu_slave] HEX_CMD_03
Info:  [modbus_rtu_slave] HEX_CMD_03

After executing the command, I can receive RTT log data, but I need to send the data. How can I achieve this? For example, sending the "bms" string.

Althrone, 2024-01-30:
i have the same question,wait a help :-)

RehabAbdAllah, 2024-07-02:
is there any updates on this ?

wenhuafeng, 2025-04-10:
pyocd rtt, how to send data?