pyocd/pyOCD #1637: I encountered an issue while using pyOCD with a Cortex-M33 microcontroller. The error occurs during the session initialization and seems to be related to the AP (Access Port) address.

enzheng128, 2023-10-16:
### Error Log

```
0000542 D Project directory: d:\code [session]
0000573 D Project directory: d:\code [session]
0000608 D CMSIS-DAP v2 probe 000100011a5a5bb6000000000001badfa5a5a5a597969908: firmware version 0258, protocol version 2.1.0 [dap_access_cmsis_dap]
0000609 W Board ID 0001 is not recognized [mbed_board]
0000611 D GeckoPlatform_EFR32BG22_DFP DFP: skipping non-Keil flash algorithm [cmsis_pack]
0000611 D GeckoPlatform_EFR32BG22_DFP DFP: skipping non-Keil flash algorithm [cmsis_pack]
0000611 D GeckoPlatform_EFR32BG22_DFP DFP: skipping non-Keil flash algorithm [cmsis_pack]
0000611 D GeckoPlatform_EFR32BG22_DFP DFP: skipping non-Keil flash algorithm [cmsis_pack]
0000611 D GeckoPlatform_EFR32BG22_DFP DFP: skipping non-Keil flash algorithm [cmsis_pack]
0000611 D GeckoPlatform_EFR32BG22_DFP DFP: skipping non-Keil flash algorithm [cmsis_pack]
0000611 D GeckoPlatform_EFR32BG22_DFP DFP: skipping non-Keil flash algorithm [cmsis_pack]
0000611 D GeckoPlatform_EFR32BG22_DFP DFP: skipping non-Keil flash algorithm [cmsis_pack]
0000611 D GeckoPlatform_EFR32BG22_DFP DFP: skipping non-Keil flash algorithm [cmsis_pack]
0000611 D GeckoPlatform_EFR32BG22_DFP DFP: skipping non-Keil flash algorithm [cmsis_pack]
0000618 I Target type is efr32bg22c224f512gm32 [board]
0000639 D Board: LuatOS Air32 MINI DAPLink [cmsis_dap_probe]
0000639 D Target: LuatOS Air32F103CB [cmsis_dap_probe]
0000639 D Running task load_svd [sequencer]
0000705 D Running task pre_connect [sequencer]
0000720 D Running task dp_init [sequencer]
0000736 D Running task lock_probe [sequencer]
0000752 D Running task get_probe_capabilities [sequencer]
0000767 D Running task connect [sequencer]
0000799 D Default wire protocol selected; using SWD [dap]
0000799 D Sending deprecated SWJ sequence to select SWD [swj]
0000799 I DP IDR = 0x6ba02477 (v2 rev6) [dap]
0000799 D Running task clear_sticky_err [sequencer]
0000799 D Running task power_up_debug [sequencer]
0000799 D Running task check_version [sequencer]
0000799 D Running task unlock_probe [sequencer]
0000799 D Running task unlock_device [sequencer]
0000799 D Running task create_discoverer [sequencer]
0000799 D Running task discovery [sequencer]
0000799 D Running task find_aps [sequencer]
0000799 D Running task create_aps [sequencer]
0000799 D Running task create_ap.0 [sequencer]
0000814 D AHB-AP#0 default HPROT=3 HNONSEC=0 [ap]
0000814 D AHB-AP#0 implemented HPROT=f HNONSEC=1 [ap]
0000814 I AHB-AP#0 IDR = 0x84770001 (AHB-AP var0 rev8) [discovery]
0000814 D Running task create_ap.1 [sequencer]
0000814 D APB-AP#1 default HPROT=0 HNONSEC=0 [ap]
0000814 D APB-AP#1 implemented HPROT=0 HNONSEC=0 [ap]
0000814 I APB-AP#1 IDR = 0x54770002 (APB-AP var0 rev5) [discovery]
0000814 D Running task create_ap.3 [sequencer]
0000814 D AHB-AP#3 default HPROT=3 HNONSEC=0 [ap]
0000814 D AHB-AP#3 implemented HPROT=f HNONSEC=1 [ap]
0000814 I AHB-AP#3 IDR = 0x84770001 (AHB-AP var0 rev8) [discovery]
0000814 D Running task find_components [sequencer]
0000814 D Running task init_ap.0 [sequencer]
0000814 I AHB-AP#0 Class 0x1 ROM table #0 @ 0xe00fe000 (designer=673 part=d41) [rom_table]
0000814 I [0]<e00ff000:ROM class=1 designer=43b:Arm part=4c9> [rom_table]
0000814 I   AHB-AP#0 Class 0x1 ROM table #1 @ 0xe00ff000 (designer=43b:Arm part=4c9) [rom_table]
0000814 I   [0]<e000e000:SCS M33 class=9 designer=43b:Arm part=d21 devtype=00 archid=2a04 devid=0:0:0> [rom_table]
0000814 I   [1]<e0001000:DWT M33 class=9 designer=43b:Arm part=d21 devtype=00 archid=1a02 devid=0:0:0> [rom_table]
0000814 I   [2]<e0002000:BPU M33 class=9 designer=43b:Arm part=d21 devtype=00 archid=1a03 devid=0:0:0> [rom_table]
0000814 I   [3]<e0000000:ITM M33 class=9 designer=43b:Arm part=d21 devtype=43 archid=1a01 devid=0:0:0> [rom_table]
0000814 D   [4]<fff41002 not present> [rom_table]
0000814 I   [5]<e0041000:ETM M33 class=9 designer=43b:Arm part=d21 devtype=13 archid=4a13 devid=0:0:0> [rom_table]
0000814 I   [6]<e0042000:CTI M33 class=9 designer=43b:Arm part=d21 devtype=14 archid=1a14 devid=40800:0:0> [rom_table]
0000814 D   [7]<fff44002 not present> [rom_table]
0000814 I [1]<e0040000:TPIU M33 class=9 designer=43b:Arm part=d21 devtype=11 archid=0000 devid=ca1:0:0> [rom_table]
0000814 I [2]<e00fd000:TSGEN class=15 designer=43b:Arm part=101> [rom_table]
0000814 D Running task init_ap.1 [sequencer]
0000814 I APB-AP#1 Class 0x1 ROM table #0 @ 0x000000 (designer=673 part=db6) [rom_table]
0000814 I [0]<00001000:??? class=15 designer=673 part=051> [rom_table]
0000814 D Running task init_ap.3 [sequencer]
0000814 I AHB-AP#3 Class 0x1 ROM table #0 @ 0xf0000000 (designer=673 part=52d) [rom_table]
0000814 I [0]<e00ff000:ROM class=1 designer=43b:Arm part=4c0> [rom_table]
0000814 I   AHB-AP#3 Class 0x1 ROM table #1 @ 0xe00ff000 (designer=43b:Arm part=4c0) [rom_table]
0000814 I   [0]<e000e000:SCS v6-M class=14 designer=43b:Arm part=008> [rom_table]
0000814 I   [1]<e0001000:DWT v6-M class=14 designer=43b:Arm part=00a> [rom_table]
0000814 I   [2]<e0002000:BPU v6-M class=14 designer=43b:Arm part=00b> [rom_table]
0000814 I [1]<f0001000:CTI M0+ class=9 designer=43b:Arm part=9a6 devtype=14 archid=1a14 devid=1040800:0:0> [rom_table]
0000814 D Running task create_cores [sequencer]
0000814 D Creating SCS component [discovery]
0000814 D selected core #0 [soc_target]
0000830 I CPU core #0: Cortex-M33 r0p4, v8.0-M architecture [cortex_m]
0000830 I   Extensions: [DSP, FPU, FPU_V5, MPU, SEC] [cortex_m]
0000830 I   FPU present: FPv5-SP-D16-M [cortex_m]
0000830 D Creating SCS component [discovery]
0000830 D uninit session <pyocd.core.session.Session object at 0x0000025905828AC0> [session]
0000830 C Error: <APv1Address@0x25909297df0 #3 dp=0> [__main__]
Traceback (most recent call last):
  File "C:\Users\zhengen\AppData\Local\Programs\Python\Python310\lib\site-packages\pyocd\__main__.py", line 161, in run
    status = cmd.invoke()
  File "C:\Users\zhengen\AppData\Local\Programs\Python\Python310\lib\site-packages\pyocd\subcommands\load_cmd.py", line 100, in invoke
    with session:
  File "C:\Users\zhengen\AppData\Local\Programs\Python\Python310\lib\site-packages\pyocd\core\session.py", line 415, in __enter__
    self.open()
  File "C:\Users\zhengen\AppData\Local\Programs\Python\Python310\lib\site-packages\pyocd\core\session.py", line 553, in open
    self._board.init()
  File "C:\Users\zhengen\AppData\Local\Programs\Python\Python310\lib\site-packages\pyocd\board\board.py", line 143, in init
    self.target.init()
  File "C:\Users\zhengen\AppData\Local\Programs\Python\Python310\lib\site-packages\pyocd\coresight\coresight_target.py", line 123, in init
    super().init()
  File "C:\Users\zhengen\AppData\Local\Programs\Python\Python310\lib\site-packages\pyocd\core\soc_target.py", line 174, in init
    seq.invoke()
  File "C:\Users\zhengen\AppData\Local\Programs\Python\Python310\lib\site-packages\pyocd\utility\sequencer.py", line 213, in invoke
    resultSequence.invoke()
  File "C:\Users\zhengen\AppData\Local\Programs\Python\Python310\lib\site-packages\pyocd\utility\sequencer.py", line 208, in invoke
    resultSequence = call()
  File "C:\Users\zhengen\AppData\Local\Programs\Python\Python310\lib\site-packages\pyocd\coresight\discovery.py", line 63, in _create_cores
    self._apply_to_all_components(self._create_component,
  File "C:\Users\zhengen\AppData\Local\Programs\Python\Python310\lib\site-packages\pyocd\coresight\discovery.py", line 74, in _apply_to_all_components
    ap.rom_table.for_each(action, filter)
  File "C:\Users\zhengen\AppData\Local\Programs\Python\Python310\lib\site-packages\pyocd\coresight\rom_table.py", line 330, in for_each
    component.for_each(action, filter)
  File "C:\Users\zhengen\AppData\Local\Programs\Python\Python310\lib\site-packages\pyocd\coresight\rom_table.py", line 338, in for_each
    action(component)
  File "C:\Users\zhengen\AppData\Local\Programs\Python\Python310\lib\site-packages\pyocd\coresight\discovery.py", line 56, in _create_component
    cmp = cmpid.factory(cmpid.ap, cmpid, cmpid.address)
  File "C:\Users\zhengen\AppData\Local\Programs\Python\Python310\lib\site-packages\pyocd\coresight\cortex_m.py", line 222, in factory
    root.add_core(core)
  File "C:\Users\zhengen\AppData\Local\Programs\Python\Python310\lib\site-packages\pyocd\target\pack\pack_target.py", line 511, in _pack_target_add_core
    pname = _self._pack_device.processors_ap_map[cast(CortexM, core).ap.address].name
KeyError: <APv1Address@0x1dc2b903af0 #3 dp=0>
```

### Steps to Reproduce

1. pyocd flash -t efr32bg22c224f512gm32 Sample.hex

### Environment

- Operating System: window11
- pyOCD Version: release

Any help would be appreciated. Thank you!

hagibr, 2023-11-07:
Try editing this file:

C:\Users\zhengen\AppData\Local\Programs\Python\Python310\lib\site-packages\pyocd\target\pack\pack_target.py

At line 511, change this:

```
    def _pack_target_add_core(_self, core: CoreTarget) -> None:
        """@brief Override to set node name of added core to its pname."""
        pname = _self._pack_device.processors_ap_map[cast(CortexM, core).ap.address].name
        core.node_name = pname
        CoreSightTarget.add_core(_self, core)
```

to this:

```
    def _pack_target_add_core(_self, core: CoreTarget) -> None:
        """@brief Override to set node name of added core to its pname."""
        if( cast(CortexM, core).ap.address in _self._pack_device.processors_ap_map ):
            pname = _self._pack_device.processors_ap_map[cast(CortexM, core).ap.address].name
            core.node_name = pname
            CoreSightTarget.add_core(_self, core)
```

I'm investigating a similar problem (#1642) with my EFR32BG24.

enzheng128, 2023-11-17:
> Try editing this file:
>
> C:\Users\zhengen\AppData\Local\Programs\Python\Python310\lib\site-packages\pyocd\target\pack\pack_target.py
>
> At line 511, change this:
>
> ```
>     def [...]

I'm still having issues. I suspect it's a problem with the EFR32 pack.

hagibr, 2023-11-19:
I'm using version 0.34.3, it's working partially. Give a try.

hagibr, 2025-05-23:
@enzheng128, I don't know if you are still having this issue. Today I've learned how to add builtin targets to my pyocd installation and now the last release (0.36.0) is working with my EFR32BG24.
I've followed this guide: [docs/adding_new_targets.md](../blob/main/docs/adding_new_targets.md)

hagibr, 2025-06-25:
Hello, this is my file generated with the script, and modified a bit (```pyocd/target/builtin/target_EFR32BG24.py```):
```
# pyOCD debugger
# Copyright (c) 2018 Arm Limited
# Copyright (c) 2021 Chris Reed
# SPDX-License-Identifier: Apache-2.0
#
# Licensed under the Apache License, Version 2.0 (the "License");
# you may not use this file except in compliance with the License.
# You may obtain a copy of the License at
#
#     http://www.apache.org/licenses/LICENSE-2.0
#
# Unless required by applicable law or agreed to in writing, software
# distributed under the License is distributed on an "AS IS" BASIS,
# WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
# See the License for the specific language governing permissions and
# limitations under the License.

from ...coresight.coresight_target import CoreSightTarget
from ...core.memory_map import (FlashRegion, RamRegion, MemoryMap)
from ...debug.svd.loader import SVDFile

# FLASH_ALGO initially generated by using scripts/generate_flash_algo.py GECKOS2C3.FLM
# GECKOS2C3.FLM was extracted from CMSIS-Pack, and we did some modifications
FLASH_ALGO = {
    'load_address' : 0x20000000,

    # Flash algorithm as a hex string
    'instructions': [
    0xe7fdbe00,
    0x6801480d, 0x117cf401, 0x1fb0f5b1, 0x490bd904, 0xf442680a, 0x600a3280, 0xf6414909, 0x600a3271,
    0x604a2200, 0x49076880, 0x2180ea01, 0x20004a06, 0x1002f849, 0xbf004770, 0x0fe08004, 0x40008068,
    0x4003003c, 0x03fffc00, 0x00000004, 0xf6414905, 0x60083071, 0x22014804, 0x22006002, 0x600a2000,
    0xbf004770, 0x4003003c, 0x4003200c, 0x4d09b5b0, 0xf8592400, 0x42840005, 0x2000bf24, 0x4620bdb0,
    0xf80af000, 0xf5042800, 0xbf1c5400, 0xbdb02001, 0xbf00e7ef, 0x00000004, 0x4602490f, 0x3301ca08,
    0x2900d103, 0x0104f101, 0xb193d1f8, 0x4c0bb5b0, 0x490b2501, 0x60486025, 0x60082002, 0x21002001,
    0xf85ef000, 0x5180f44f, 0x50652800, 0x2001bf18, 0x2000bdb0, 0xbf004770, 0xffffe004, 0x4003100c,
    0x40030010, 0x4df0e92d, 0x1cc84605, 0x4a224692, 0xf0204f22, 0x20010403, 0xe02e6010, 0x5000f505,
    0xeba04388, 0x1e690b05, 0xeba1455c, 0xea6f0000, 0xbf380104, 0x46d046a3, 0x607d4288, 0x4601bf88,
    0xf8d81d4e, 0x2e000000, 0xd00a60b8, 0x21082008, 0x0804f108, 0xf824f000, 0xf1062800, 0xd0f00604,
    0x2004e016, 0x20016038, 0x26012100, 0xf818f000, 0x44dab978, 0xeba4445d, 0xf641040b, 0x2c0071ff,
    0xf44fd1cc, 0x21015080, 0x26004a03, 0xe0005011, 0x46302601, 0x8df0e8bd, 0x4003100c, 0x40030010,
    0x4e10b570, 0x4c0e4a0d, 0xe0091d33, 0x428d4005, 0x2000bf04, 0x3401bd70, 0xf06fbf04, 0xbd700002,
    0x421d6915, 0x6810d0f2, 0xf0204235, 0xf06f0101, 0xbf080001, 0x30fff04f, 0xbd706011, 0x4003000c,
    0xff676980, 0x00010002, 0x00000000, 0x00000000
    ],

    # Relative function addresses
    'pc_init': 0x20000005,
    'pc_unInit': 0x20000051,
    'pc_program_page': 0x200000e9,
    'pc_erase_sector': 0x2000009d,
    #'pc_eraseAll': 0x20000071, # Main modification: the code from CMSIS-Pack doesn't work. By commenting this, pyocd will call pc_erase_sector multiple times instead

    'static_base' : 0x20000000 + 0x00000004 + 0x000001c8,
    'begin_stack' : 0x200071e0,
    'end_stack' : 0x200061e0,
    'begin_data' : 0x20000000 + 0x1000,
    'page_size' : 0x3000,
    'analyzer_supported' : False,
    'analyzer_address' : 0x00000000,
    # Enable double buffering.
    'page_buffers' : [
        0x200001e0,
        0x200031e0
    ],
    'min_program_length' : 0x3000,

    # Relative region addresses and sizes
    'ro_start': 0x4,
    'ro_size': 0x1c8,
    'rw_start': 0x1cc,
    'rw_size': 0x4,
    'zi_start': 0x1d0,
    'zi_size': 0x4,

    # Flash information
    'flash_start': 0x8000000,
    'flash_size': 0x180000,
    'sector_sizes': (
        (0x0, 0x2000),
    )
}

class EFR32BG24(CoreSightTarget):

    VENDOR = "Silicon Labs"

    MEMORY_MAP = MemoryMap(
        FlashRegion(name='Flash', start=0x08000000, length=0x100000, blocksize=0x2000, is_boot_memory=True, algo=FLASH_ALGO),
        RamRegion(name='RAM', start=0x20000000, length=0x20000),
        )

    def __init__(self, session):
        super().__init__(session, self.MEMORY_MAP)

```

I've also added these lines to `pyocd/target/builtin/__init__.py`:
```
...
from . import target_EFR32BG24
...
BUILTIN_TARGETS = {
...
          'efr32bg24': target_EFR32BG24.EFR32BG24,
         }
```

And now I just have to call pyocd with this parameter: `-t EFR32BG24`