pyocd/pyOCD #1642: Silabs EFR32BG24 stopped working

hagibr, 2023-11-01:
It worked fine with both ST-Link V2 and J-Link until version 0.34.3. OK, more or less, with J-Link, it is very slow.
With versions 0.35.0 and later, this error appears when I start the commander:

ST-Link:
0002972 E Transfer error while reading AHB-AP#2 ROM table: Memory transfer fault (read) @ 0xf0000fd0-0xf000102f [ap]
0002995 C Error: <APv1Address@0x9602fb0 #3 dp=0> [__main__]

J-Link:
0004180 W Invalid coresight component, cidr=0x0 [rom_table]
0004459 C Error: <APv1Address@0xa468df0 #3 dp=0>

I can provide more logs, if someone gives me directions on how to do it.

ntn888, 2025-02-22:
Hi did you manage to resolve it? I'm getting the same error with efr32mg21 series with both stlink and dap-link.. Thank you!

```
pyocd load --target EFR32MG21A020F768IM32 /var/home/ajit/work/sl/gecko-sdk/myapps/mk-template/build_output/main.elf
0000926 E Transfer error while reading AHB-AP#2 ROM table: Memory transfer fault (read) @ 0xf0000fd0-0xf000102f [ap]
0000944 I Loading /var/home/ajit/work/sl/gecko-sdk/myapps/mk-template/build_output/main.elf [load_cmd]
0000949 C Error: Sector (8192 bytes) do not fit evenly into page (12288 bytes) [__main__]
```

hagibr, 2025-02-22:
Hi, I've downgraded pyOCD to version 0.34.3 to make it work.

ntn888, 2025-02-22:
Thanks for your response. I did try that according to your other post. But resulted in same error.

hagibr, 2025-05-23:
@ntn888, I don't know if you are still having this issue. Today I've learned how to add builtin targets to my pyocd installation and now the last release (0.36.0) is working with my EFR32BG24.
I've followed this guide: [docs/adding_new_targets.md](https://github.com/pyocd/pyOCD/blob/main/docs/adding_new_targets.md)